framps/raspiBackup #403: raspiBackup auf NFS - RBK0263E

pyro2k9, 2021-11-30:
Bekomme dabei aber die Fehlermeldung "RBK0263E: Dateisystem auf /backup unterstützt keine Linux Dateiattribute."

DIe NFS Freigabe kommt von einer Synology DS218+ NAS.

/etc/fstab auf dem Client (raspi):

192.168.x.x.local:/volume1/Backups_lokal/raspiBackup         /backup         nfs    rw      0       0

HIer der Debug Code:

```
20211130-183236 DBG 3283:          --> getFsType /backup
20211130-183237 DBG 3328:              --> findMountPath /backup
20211130-183237 DBG 3336:                  --- Path: /backup
20211130-183237 DBG 3344:              <-- findMountPath /backup
20211130-183237 DBG 3286:              --- Mountpoint: /backup
20211130-183237 DBG 3290:              --- df -T: pyro-ds.local:/volume1/Backups_lokal/raspiBackup nfs4     3746086272 3084814720 661271552  83% /backup
20211130-183237 DBG 3294:          <-- getFsType nfs4
20211130-183237 DBG 6293:          --- Filesystem: nfs4
20211130-183237 DBG 3190:          --> supportsFileAttributes /backup
20211130-183237 DBG 3204:              --- ----r-xrwx # nobody # nogroup
cp: failed to preserve ownership for '//backup/raspiBackup.fileattributes': Operation not permitted
20211130-183237 DBG 3210:              --- -rwxrwxrwx # 1024 # users
20211130-183237 DBG 3218:          <-- supportsFileAttributes 1
20211130-183237 DBG 3328:          --> findMountPath /backup
20211130-183237 DBG 3336:              --- Path: /backup
20211130-183237 DBG 3344:          <-- findMountPath /backup
20211130-183237 DBG 3591:          --> colorAnnotation 1
20211130-183237 DBG 3615:          <-- colorAnnotation 
??? RBK0266E: Es fehlt die Berechtigung um Linux Dateiattribute auf /backup zu erstellen (Dateisystem: nfs4).
20211130-183237 DBG 2248:          --> exitError 102
20211130-183237 DBG 2255:          <-- exitError 102
```

Der lokale /backup Ordner auf dem Raspi hat  
 drwxrwxrwx   1 1026 users    12 Nov 30 18:32 backup

Der Ordner aus dem NAS
drwxrwxrwx+ 1 root            root                 1078 Nov 29 19:11 Backups_lokal

framps, 2021-11-30:
Stell bitte das gesamte Debuglog hier ein damit ich nachsehen kann :wink:  

Zu dem Hotfix: Der ist in der aktuellen Version drin.

pyro2k9, 2021-12-01:
Hallo, hier der komplette Debuglog. Ich hofffe du findest woran es liegt..

```
20211130-183225 DBG 1937:  --- raspiBackup 0.6.6.1, 2021-11-19/22:29:00 - 5635776
20211130-183225 DBG 1940:  --- +================================================================================================================================================+
20211130-183225 DBG 1941:  --- | ===> A lot of sensitive information is masqueraded in this log file. Nevertheless please check the log carefully before you distribute it <=== |
20211130-183225 DBG 1942:  --- +================================================================================================================================================+
20211130-183225 DBG 1775:  --> trapWithArg TRAP cleanupStartup SIGINT SIGTERM EXIT
20211130-183225 DBG 1780:  <-- trapWithArg 
20211130-183225 DBG 2437:  --> initializeDefaultConfigVariables 
20211130-183225 DBG 2569:  <-- initializeDefaultConfigVariables 
20211130-183225 DBG 2574:  --> copyDefaultConfigVariables 
20211130-183225 DBG 2643:  <-- copyDefaultConfigVariables 
20211130-183225 DBG 3351:  --> readConfigParameters 
20211130-183225 DBG 3366:      --- Read config /usr/local/etc/raspiBackup.conf : 0.1.5
20211130-183225 DBG 3366:      --- VERSION_CONFIG="0.1.5"
20211130-183225 DBG 3366:      --- DEFAULT_BACKUPPATH="/backup"
20211130-183225 DBG 3366:      --- DEFAULT_BACKUPTYPE="rsync"
20211130-183225 DBG 3366:      --- DEFAULT_ZIP_BACKUP="0"
20211130-183225 DBG 3366:      --- DEFAULT_DD_BACKUP_SAVE_USED_PARTITIONS_ONLY=0
20211130-183225 DBG 3366:      --- DEFAULT_KEEPBACKUPS="3"
20211130-183225 DBG 3366:      --- DEFAULT_KEEPBACKUPS_DD=0
20211130-183225 DBG 3366:      --- DEFAULT_KEEPBACKUPS_DDZ=0
20211130-183225 DBG 3366:      --- DEFAULT_KEEPBACKUPS_TAR=0
20211130-183225 DBG 3366:      --- DEFAULT_KEEPBACKUPS_TGZ=0
20211130-183225 DBG 3366:      --- DEFAULT_KEEPBACKUPS_RSYNC=0
20211130-183225 DBG 3366:      --- DEFAULT_SMART_RECYCLE="1"
20211130-183225 DBG 3366:      --- DEFAULT_SMART_RECYCLE_DRYRUN="0"
20211130-183225 DBG 3366:      --- DEFAULT_SMART_RECYCLE_OPTIONS="7 0 4 1"
20211130-183225 DBG 3366:      --- DEFAULT_STOPSERVICES=":"
20211130-183225 DBG 3366:      --- DEFAULT_STARTSERVICES=":"
20211130-183225 DBG 3366:      --- DEFAULT_BEFORE_STOPSERVICES=""
20211130-183225 DBG 3366:      --- DEFAULT_AFTER_STARTSERVICES=""
20211130-183225 DBG 3366:      --- DEFAULT_EMAIL=""
20211130-183225 DBG 3366:      --- DEFAULT_SENDER_EMAIL=""
20211130-183225 DBG 3366:      --- DEFAULT_EMAIL_PARMS=""
20211130-183225 DBG 3366:      --- DEFAULT_MAIL_PROGRAM="mail"
20211130-183225 DBG 3366:      --- DEFAULT_TELEGRAM_TOKEN=""
20211130-183225 DBG 3366:      --- DEFAULT_TELEGRAM_CHATID=""
20211130-183225 DBG 3366:      --- DEFAULT_TELEGRAM_NOTIFICATIONS="F"
20211130-183225 DBG 3366:      --- DEFAULT_NOTIFY_START=0
20211130-183225 DBG 3366:      --- DEFAULT_MAIL_ON_ERROR_ONLY=0
20211130-183225 DBG 3366:      --- DEFAULT_MSG_LEVEL="0"
20211130-183225 DBG 3366:      --- DEFAULT_COLORING="CM"
20211130-183225 DBG 3366:      --- DEFAULT_EMAIL_COLORING="SUBJECT"
20211130-183225 DBG 3366:      --- DEFAULT_RESTORE_DEVICE=""
20211130-183225 DBG 3366:      --- DEFAULT_VERBOSE=0
20211130-183225 DBG 3366:      --- DEFAULT_SKIPLOCALCHECK=0
20211130-183225 DBG 3366:      --- DEFAULT_DD_BLOCKSIZE=1M
20211130-183225 DBG 3366:      --- DEFAULT_DD_PARMS=""
20211130-183225 DBG 3366:      --- DEFAULT_EXCLUDE_LIST=""
20211130-183225 DBG 3366:      --- DEFAULT_NOTIFY_UPDATE=1
20211130-183225 DBG 3366:      --- DEFAULT_EXTENSIONS="mem temp disk"
20211130-183225 DBG 3366:      --- DEFAULT_LANGUAGE="DE"
20211130-183225 DBG 3366:      --- DEFAULT_DEPLOYMENT_HOSTS=""
20211130-183225 DBG 3366:      --- DEFAULT_YES_NO_RESTORE_DEVICE=""
20211130-183225 DBG 3366:      --- DEFAULT_RESIZE_ROOTFS=1
20211130-183225 DBG 3366:      --- DEFAULT_LINK_BOOTPARTITIONFILES=0
20211130-183225 DBG 3366:      --- DEFAULT_TAR_BOOT_PARTITION_ENABLED=0
20211130-183225 DBG 3366:      --- DEFAULT_SKIP_DEPRECATED=""
20211130-183225 DBG 3366:      --- DEFAULT_USE_UUID=1
20211130-183225 DBG 3366:      --- DEFAULT_CHECK_FOR_BAD_BLOCKS=0
20211130-183225 DBG 3366:      --- DEFAULT_TIMESTAMPS=0
20211130-183225 DBG 3366:      --- DEFAULT_SYSTEMSTATUS=0
20211130-183225 DBG 3366:      --- DEFAULT_RESTORE_REMINDER_INTERVAL=6
20211130-183225 DBG 3366:      --- DEFAULT_RESTORE_REMINDER_REPEAT=3
20211130-183225 DBG 3366:      --- DEFAULT_PARTITIONBASED_BACKUP="0"
20211130-183225 DBG 3366:      --- DEFAULT_PARTITIONS_TO_BACKUP="1 2"
20211130-183225 DBG 3366:      --- DEFAULT_DYNAMIC_MOUNT=""
20211130-183225 DBG 3366:      --- UUID=5358b260-d576-4eaa-98d9-8b5ccb6386a2
[... 427 lines cut ...]
20211130-183237 DBG 4380:              <-- cleanupTempFiles 
20211130-183238 DBG 4111:              --- Terminate now with rc 102
20211130-183238 DBG 3591:              --> colorAnnotation 1
20211130-183238 DBG 3615:              <-- colorAnnotation 
??? RBK0005E: Backup fehlerhaft beendet. Siehe vorhergehende Fehlermeldungen.
20211130-183238 DBG 3591:              --> colorAnnotation 1
20211130-183238 DBG 3615:              <-- colorAnnotation 
--- RBK0010I: @HOSTNAME@: raspiBackup V0.6.6.1 (5635776) Tue Nov 30 18:32:38 UTC 2021 beendet mit Returncode 102.
20211130-183238 DBG 3736:              --> sendEMail 
20211130-183238 DBG 3858:              <-- sendEMail 
20211130-183238 DBG 3957:              --- Masquerading some mount options
[... cut ...]
20211130-183239 DBG 1964:                  --- LOG_OUTPUT: 3
20211130-183239 DBG 2004:                  --- DEST_LOGFILE: /home/ubuntu/raspiBackup.log
20211130-183239 DBG 2005:                  --- DEST_MSGFILE: 
20211130-183239 DBG 3591:                  --> colorAnnotation 1
20211130-183239 DBG 3615:                  <-- colorAnnotation 
20211130-183239 DBG 2024:                  --- Updating logfile ownership
20211130-183239 DBG 2029:              <-- logFinish 
20211130-183239 DBG 4190:          <-- cleanup 

```

pyro2k9, 2021-12-01:
Oh man, das scheint es gewesen zu sein. Wie peinlich! Naja, vielleicht hilfts dem ein oder anderem, wenn sie dies hier lesen....

Vielen Dank!

closed 2021-12-01 by pyro2k9