framps/raspiBackup #404: Compressed backup is larger than the source

g6nhu, 2021-12-03:
Hi,
I've been using this for a while and everything works and I've been able to successfully restore an SD card.  I have it running on multiple Pis, backing up to my Synology NAS.

However, I've just taken the time to actually look at my backup files and in some cases, the compressed file is larger than the source.

For example:
```
pi@internal_tracker:~ $ df -h
Filesystem                       Size  Used Avail Use% Mounted on
/dev/root                         15G  1.7G   13G  12% /
devtmpfs                         459M     0  459M   0% /dev
tmpfs                            464M     0  464M   0% /dev/shm
tmpfs                            464M  156M  308M  34% /run
tmpfs                            5.0M  4.0K  5.0M   1% /run/lock
tmpfs                            464M     0  464M   0% /sys/fs/cgroup
/dev/mmcblk0p1                    43M   23M   20M  53% /boot
192.168.1.251:/volume1/pi-clone   22T   15T  6.8T  69% /home/pi/nas-share
tmpfs                             32M   28K   32M   1% /var/cache/fontconfig
tmpfs                             93M     0   93M   0% /run/user/1000
```

The backup file on my NAS from last week is called 'internal_tracker-ddz-backup-20211128-012001.img.gz' is 6.43Gb.

I'm using 0.6.6 and have the backup type configured as Backup with dd and restore on Windows with Compression on.  

I can provide logs if needed for investigation.  I have plenty of space on my NAS but I'm just concerned roughly 2Gb of data on the SD card creates a compressed backup file that's over 6Gb.

Thanks.

framps, 2021-12-04:
> However, I've just taken the time to actually look at my backup files and in some cases, the compressed file is larger than the source.

`/dev/root                         15G  1.7G   13G  12% /`

Your SD card has about 16GB and dd saves the whole 16GB all the time. Because you use dd compression the image will be compressed to something < 16GB. The size of the compressed image depends on the amount and structure of the data contained in 16GB. 

Even if just used about 2GB on your SD card the data which resides on the rest of your SD card (14GB) will be saved and finally the whole 16GB is compressed. If there is a lot of filesystem activity on your system and files are created and deleted the deleted files will not count in the number of used bytes in `df -h` but it's still persistent - but marked as deleted - somewhere on the 16GB of your SD card and will be saved by dd. If you have just files with zeros the compression ration will be high but all other random file contents will decrease the compression rate.

I suggest to check [FAQ16](https://www.linux-tips-and-tricks.de/en/faq/#a16) to reduce the image size to something below 2GB all the time.

> and restore on Windows with Compression on.

I'm not very familiar with Windows. Which tool do you use to restore the compressed dd image? I know of windisk32imager which does not accept a compressed image.

g6nhu, 2021-12-04:
Thanks for the quick and detailed reply :)

OK, yes, I understand, thanks.  Some of the Pis (I actually have 12 of them backing up) do see a fair bit of activity on the SD card which would explain it.  I'll have a look at the options, it might be worth very occasionally just zeroing out the spare space on the cards to keep the backup size down but naturally, this isn't something that should be done very often.

Regarding the backup type, I'll be honest, I followed a guide and that's what it suggested.  I also don't use Windows and restore the compressed dd image using Balenaetcher on either a Mac or a Linux box.  I want to produce an image that I can simply restore straight onto an SD card and slap straight back into the Pi and this works. Would a tar backup do the same thing?

Thanks again, I really appreciate your help.

framps, 2021-12-04:
> Would a tar backup do the same thing?

`DEFAULT_DD_BACKUP_SAVE_USED_PARTITIONS_ONLY` will reduce the size of your dd backup to the partition size and keep your dd backup much smaller that 16GB. tar and rsync backup will backup the used data only and thus you will get a smaller backup than with a dd backup which uses `DEFAULT_DD_BACKUP_SAVE_USED_PARTITIONS_ONLY`.

You wrote you have a plenty of space on your NAS. So you don't have to care about space used by a backup. Nevertheless I suggest to use either tar of rsync if you dont' need a dd backup which can be restored on Windows. 

rsync just copies only updates, uses hardlinks and usually very is fast. A tar backup is a full backup every time which takes longer to create the backup. rsync has the advantage the backup exists as a plain directory and you can easily retrieve any number of files from the backup. If you use tar you have to extract the files via tar commands first and cannot navigate in the directory tree. But a tar backup is a full backup which can be copied easily somewhere else. So there are pros and cos for tar and rsync :wink:

closed 2021-12-23 by github-actions[bot]