framps/raspiBackup #406: Restore with -g misses progress inidation for root partition

framps, 2021-12-05:
Restore of rsync backup creates a progress bar for boot partition but not for root partition.

```
pi@pi4:~ $ sudo raspiBackup -g -d /dev/sda /backup/pi4/pi4-rsync-backup-20211205-101132/
--- RBK0009I: pi4: raspiBackup V0.6.6.1 (5635776) started at Sun 05 Dec 2021 12:33:46 PM CET.
!!! RBK0018W: Target /dev/sda with 447.13 GiB is larger than backup source with 59.47 GiB. root partition will be expanded accordingly to use the whole space.
!!! RBK0065W: Device /dev/sda will be repartitioned and all data will be lost.

--- RBK0066I: Device /dev/sda will be overwritten with the saved boot and root partition.
--- RBK0069I: Bootpartition /dev/sda1 will be formatted and will get the restored Boot partition.
--- RBK0070I: Rootpartition /dev/sda2 will be formatted and will get the restored Root partition.
--- RBK0038I: Are you sure? y/N
y
--- RBK0050I: Restoring backup from /backup/pi4/pi4-rsync-backup-20211205-101132.
--- RBK0004I: Adjusting second partition from 59.22 GiB to 446.87 GiB.
--- RBK0053I: Restoring first partition (boot partition) to /dev/sda1.
256MiB 0:00:23 [10.8MiB/s] [================================>] 100%
--- RBK0055I: Restoring second partition (root partition) to /dev/sda2.
--- RBK0033I: Please wait until cleanup has finished.
--- RBK0076I: Restore finished successfully.
--- RBK0010I: pi4: raspiBackup V0.6.6.1 (5635776) stopped at Sun 05 Dec 2021 12:43:27 PM CET with rc 0.
--- RBK0026I: Debug logfile saved in /home/pi/raspiBackup.log.
```

framps, 2021-12-05:
I just detected -g doesn't work for a backup. In addition the boot partition backup is not enabled for a progress bar.

mcm1957, 2021-12-08:
> 

> I just detected -g doesn't work for a backup. In addition the boot partition backup is not enabled for a progress bar.

Yes 
I noticed this too. But I was not sure wether rsync backup does not support this as the amount of data to be transferred is not known before executing the backup and evaluating the changes. Any - you are right, I did not see any changes when using option -g during a backup.

framps, 2021-12-08:
> Yes I noticed this too. But I was not sure wether rsync backup does not support this as the amount of data to be transferred is not known before executing the backup and evaluating the changes. Any [...]

Please check rsync option `--info=progress2` :wink:

closed 2021-12-15 by framps