framps/raspiBackup #407: Option DEFAULT_EXCLUDE_LIST

mcm1957, 2021-12-06:
Da ich Probleme mit ACLs bei Verwendung meines Synology NAS als Backup Target habe, habe ich versucht der Verzeichnis /media/pi vom Backup auszuschließen. Dazu habe ich in der Config Datei  /usr/local/etc/raspiBackup.conf die Zeile

DEFAULT_EXCLUDE_LIST="/media/pi"

eingefügt bzw. das Verzeichnis eingetragen. Dies hatte aber nicht da gewünschte Ergebnis, da der Wert dieser Variablen DIREKT an den rsync Befehl weitergegeben wird.

Erst die Angabe

DEFAULT_EXCLUDE_LIST="--exclude=/media/pi"

brachte den gewünschten Erfolg.

Da ich nicht sicher bin, ob dies das gewünschte Verhalten ist, oder ich eine Doku übersehen habe erstelle ich hier mal eine Question. 
Aus Kompatibilitätsgründen zu existierenden Installationen würde es wahrscheinlich schon ausreichen  einen entsprechenden Kommentar in der Konfigdatei mit der richtigen Syntax einzutragen.

Auszug aus dem fehlhaften Lauf:

20211206-174005 DBG 2273:                  --> executeCmd Command: rsync --exclude="/backup"                    --exclude="/home/@USER@/raspiBackup.log"                        --exclude="/home/@USER@/raspiBackup.msg"                        --exclude='.gvfs'                       --exclude=/proc/*                       --exclude=/lost+found/*                         --exclude=/sys/*                        --exclude=/dev/*                        --exclude=/boot/*                       --exclude=/tmp/*                        --exclude=/run/*                        --exclude=/@HOSTNAME@-backup.img --exclude=/@HOSTNAME@-backup.tmg --exclude=/@HOSTNAME@-backup.sfdisk --exclude=/@HOSTNAME@-backup.mbr --exclude=/raspiBackup.log --exclude=/raspiBackup.msg                    **/media/pi**                                               --numeric-ids                   -aHAx --delete                                                                  /                       "/backup/@HOSTNAME@/@HOSTNAME@-rsync-backup-20211206-173935"

framps, 2021-12-06:
> ob dies das gewünschte Verhalten ist, oder ich eine Doku übersehen habe

Ja. [Hier](https://www.linux-tips-and-tricks.de/de/raspibackup/#parm_u) ist es dokumentiert. 

Aber da es immer mal wieder uebersehen wird greife ich die Idee das noch mal in der Config zu dokumentieren auf :+1: . 

Ich hatte auch mal ueberlegt Code zu schreiben der die `--exclude` einpflegt. Aber ich bin dann wieder davon abgekommen da man ja auch `--include` und andere rsync Anweisung dort reinschreiben kann. Am flexibelsten ist man eben doch immer noch wenn man direkt die Backuptool Anweisungen nutzen kann.

Vielen Dank fuer den Hinweis.

framps, 2021-12-06:
Added comments in config files. See https://github.com/framps/raspiBackup/commit/26ebda1c28855b40b3dfb7d5deefa8cbb309fe27

framps, 2021-12-06:
> Sorry, maybe I should have posted in english:

Up to you. But it was funny: I saw your issue in German first and thus replied in German but now I see the issue in English :laughing:

framps, 2021-12-06:
> Wollte niemand ausschließen ...

:+1: Englisch ist immer besser da es von vielen Menschen gelesen werden kann. Aber ich will auch niemanden ausschliessen hier im github ein Problem zu melden oder eine Frage zu stellen nur weil er/sie nicht gut Englisch kann. :wink:

framps, 2021-12-13:
@mcm1957 Please close this issue if your're fine now

mcm1957, 2021-12-22:
I'm fine with some documentation at the config file.
I'm not aware that a new version has been released. Is this correct?

As I am not really used to git and do not know how to handle branches / merges / pull requests etc. I avoid to to try to get individual variants unless they are required for a fix. So I did not read the remark inside config file yet. Nevertheless I will close the issue. Thanks for your work.

Martin

closed 2021-12-22 by mcm1957

framps, 2021-12-22:
> Is this correct?

No. The change will be in next release. Next release branch is `development`. You can check [here](https://github.com/framps/raspiBackup/blob/6aada776d0e7feb47e7b1df12875095166b5a8e0/config/raspiBackup_en.conf#L113) the change.