framps/raspiBackup #408: Smart Recycle deleting backups of current day

mcm1957, 2021-12-08:
When I activate smart recycle I have noticed two issues I would like to discuss:

a) The smart recycle part is executed twice according to the log. Is this really desired? It would not be a big ssue as the second run would not do much unless dry run is active.

b) BUT according to smart recycle would delete ALL older backups of the current day. I would have expected ALL backups of the current day would be kept and deleteing of dayly backups starts with the previos day. 

This would be usefull when performing multiple backups during installation and test. I know that I could override options at the commandline. But this would require to disable smart recycle as default as otherwise a forgotten parameter during an intermediate backup would delete backups.

So - if its not due to an error of me - please consider this as an improvementrequest:

-) do not delete multiple backups of the current day - This would be very usefull in my oppinion
-) optionally do not delete multiply backups of the last x days - effectivly starting smart recycle actions with a delay of x days

Example:
current day  -- keep all backups
yesterday     -- keep all backups
-2                 -- keep one backup
...
-7                 -- keep one backup
-8                 -- keep one backup (weekly)
-9                 -- delete all
...
-13               -- delete all
-14               -- keep one backup (weekly)
-15               -- delete all
...

c) If I try to disable classic versionlimit using 
DEFAULT_KEEPBACKUPS = 0
I get an error. I would have expected that 0 disables any deletion. But thats total cosmetic as a value of 10000 would have the same effect in reality.

**My config is**

# how many backups to keep
DEFAULT_KEEPBACKUPS="100"
# how many backups to keep of the specific backup type. If zero DEFAULT_KEEPBACKUPS is used
DEFAULT_KEEPBACKUPS_DD=0
DEFAULT_KEEPBACKUPS_DDZ=0
DEFAULT_KEEPBACKUPS_TAR=0
DEFAULT_KEEPBACKUPS_TGZ=0
DEFAULT_KEEPBACKUPS_RSYNC=0

# Smart recycle
DEFAULT_SMART_RECYCLE="1"
# Smart recycle dryrun
DEFAULT_SMART_RECYCLE_DRYRUN="1"
# Smart recycle parameters (daily, weekly, monthly and yearly)
DEFAULT_SMART_RECYCLE_OPTIONS="7 4 12 1"

**Output from raspiBackup is**

 $ sudo /usr/local/bin/raspiBackupNfsWrapper.sh
--- MCM0000I: mounting 10.17.2.5:/volume3/backup to /backup
--- MCM0000I: starting raspiBackup
--- RBK0009I: pi4: raspiBackup.sh V0.6.6.1 (5635776) started at Wed 08 Dec 2021 08:31:37 AM CET.
--- RBK0151I: Using backuppath /backup with partition type nfs.
!!! RBK0216W: Applying backup strategy in /backup/pi4 only.
--- RBK0218I: Applying smart backup strategy. Daily:7 Weekly:4 Monthly:12 Yearly:1.
--- RBK0217I: 4 backups will be smart recycled. 4 backups will be kept. Please be patient.
!!! RBK0220W: Smart backup strategy would delete /backup/pi4/pi4-rsync-backup-20211206-202155.
!!! RBK0220W: Smart backup strategy would delete /backup/pi4/pi4-rsync-backup-20211207-151240.
!!! RBK0220W: Smart backup strategy would delete /backup/pi4/pi4-rsync-backup-20211208-082059.
!!! RBK0220W: Smart backup strategy would delete /backup/pi4/pi4-rsync-backup-20211208-083111.
!!! RBK0222W: Smart backup strategy would keep /backup/pi4/pi4-rsync-backup-20211206-183706.
!!! RBK0222W: Smart backup strategy would keep /backup/pi4/pi4-rsync-backup-20211206-203543.
!!! RBK0222W: Smart backup strategy would keep /backup/pi4/pi4-rsync-backup-20211207-205432.
!!! RBK0222W: Smart backup strategy would keep /backup/pi4/pi4-rsync-backup-20211208-083136.
--- RBK0033I: Please wait until cleanup has finished.
--- RBK0218I: Applying smart backup strategy. Daily:7 Weekly:4 Monthly:12 Yearly:1.
--- RBK0217I: 4 backups will be smart recycled. 4 backups will be kept. Please be patient.
!!! RBK0220W: Smart backup strategy would delete /backup/pi4/pi4-rsync-backup-20211206-202155.
!!! RBK0220W: Smart backup strategy would delete /backup/pi4/pi4-rsync-backup-20211207-151240.
!!! RBK0220W: Smart backup strategy would delete /backup/pi4/pi4-rsync-backup-20211208-082059.
!!! RBK0220W: Smart backup strategy would delete /backup/pi4/pi4-rsync-backup-20211208-083111.
!!! RBK0222W: Smart backup strategy would keep /backup/pi4/pi4-rsync-backup-20211206-183706.
!!! RBK0222W: Smart backup strategy would keep /backup/pi4/pi4-rsync-backup-20211206-203543.
!!! RBK0222W: Smart backup strategy would keep /backup/pi4/pi4-rsync-backup-20211207-205432.
!!! RBK0222W: Smart backup strategy would keep /backup/pi4/pi4-rsync-backup-20211208-083136.
--- RBK0017I: Backup finished successfully.
--- RBK0010I: pi4: raspiBackup.sh V0.6.6.1 (5635776) stopped at Wed 08 Dec 2021 08:31:47 AM CET with rc 0.
--- RBK0026I: Debug logfile saved in /backup/pi4/pi4-rsync-backup-20211208-083136/raspiBackup.log.
--- MCM0000I: raspiBackup completed

framps, 2021-12-08:
Would you please attach the log?

mcm1957, 2021-12-08:
Heres a log - I hope this is the desired DEBUG log. If not, please advise how to create a DEBUG log.
[raspiBackup.log](https://github.com/framps/raspiBackup/files/7677066/raspiBackup.log)

I've created a second issue as during retest and creating this log, I detected that a dry-run will create an empty backup (directory) with only the log inside. As I would call the empty directory a serious issue I think it can be handled seperated from this issue which is something between prio3 and change request. If you want to handle different please feel free to close the second issue and merge it with this one.

Anyway raspiBackup is a super tool. Restoring works like a charm. Even migrating from sd to ssd worked without major issues (after I noticed that creating new uuid would be a very good idea :-) ). THANKS for your work.

framps, 2021-12-08:
> a) The smart recycle part is executed twice according to the log. Is this really desired?

No. Good catch. I will fix this. As you said it doesn't harm but is unnecessary and confusing if you execute a dryrun.

> b) BUT according to smart recycle would delete ALL older backups of the current day. I would have expected ALL backups of the current day would be kept and deleteing of dayly backups starts with the [...]

I see your point. But this requires you finish your installation and test before the next cron scheduled backup starts. What if you need another day and want to keep all intermediate backups? Then you have to delay recycling again for another day. Or even for a week if it takes very long. I suggest to separate the regular cron backup from the manual initiated backups and use a different backup directory for the manual backups. Just use a different config file for your temporary backups which defines another backup path and use an appropriate keep count. How to use this temporary config file  depends on your installation and test strategy. See [here](https://www.linux-tips-and-tricks.de/en/backup/#parameters) for the config file search precedence. I think a good place for the temporary config file would be your current directory or your home directory. 

> c) If I try to disable classic versionlimit using
> DEFAULT_KEEPBACKUPS = 0
> I get an error. I would have expected that 0 disables any deletion. But thats total cosmetic as a value of 10000 would have the same effect in reality.

Will check on this and come back.

framps, 2021-12-08:
> I will fix this.

See https://github.com/framps/raspiBackup/commit/9c8a37f2f7e851d60293566d96855ed0afcececc

framps, 2021-12-08:
> DEFAULT_KEEPBACKUPS = 0

You get error message `??? RBK0023E: Invalid parameter 0 for -k detected.`. That's because not the way to switch between smart backup and normal backup. If you enable smart backup this will override the normal backup and you don't have to turn the normal backup off.

mcm1957, 2021-12-09:
> > DEFAULT_KEEPBACKUPS = 0
> 
> You get error message `??? RBK0023E: Invalid parameter 0 for -k detected.`. That's not the way to switch between smart backup and normal backup. If you enable smart backup this will override the normal backup and you don't have to turn the normal backup off.

I do not refer to smart recycle in special. I have found at some forum post the info that activating smart recycle will make defeult_keepbackup obsolete. (Note: Maybe a remark in configfile would be good to clearifxy that).

I noted the fact that DEFAULT_KEEPBACKUPS = 0 as I can imagine that someone does not want to make raspiBackup any cleanup operations at all but handle this cleanup manually or by some private script. So setting keepbackup to zero would be the first guess to disabale any cleanup by raspiBackup. But as I notes, setting it to 1000 or higher would create the same effect.

mcm1957, 2021-12-09:
Refering to the smart strategy. I will setup the cron job using a separate environment from the default environment which is used by manual intermidiate backups. This seems to be the safest way and does not require any change.

framps, 2021-12-09:
> Maybe a remark in configfile would be good to clearifxy that)

Will add this comment.

> So setting keepbackup to zero would be the first guess to disabale any cleanup by raspiBackup.

`KEEPBACKUP=0` asks raspiBackup to keep no backup - which doesn't make sense at all. I may create a new option `NOCLEANUP` but as you said - just use a high number and you will be able to do the cleanup on your own.

> I have found at some forum post

Do you remember which forum you read this?

framps, 2021-12-09:
> This seems to be the safest way and does not require any change.

:+1: 

Please close the issue if your're fine now.

mcm1957, 2021-12-22:
My installation states

 RBK0009I: pi4: raspiBackup.sh V0.6.6.1-hotfix ()

So how can I check whether those changes are present at my system? Is there any ident inside the file (I did not find one) to check the exact version of the script?

As I noticed at another post, I'm not familar with git. So please excuse if the question is ruther silly.

framps, 2021-12-22:
> I'm not familar with git

No problem. You should have the fix. But you can verify it: Just check whether the code change https://github.com/framps/raspiBackup/commit/9c8a37f2f7e851d60293566d96855ed0afcececc is in your code and/or whether the recycle is executed twice.

closed 2022-01-07 by github-actions[bot]