framps/raspiBackup #409: Dry-Run creates empty backup

mcm1957, 2021-12-08:
When executing a dry run - maybe only in conjunction with rsync on nfs - an empty (!) backup will be created.

Screen output:pi@pi4:~ $ sudo raspiBackup.sh
--- RBK0009I: pi4: raspiBackup.sh V0.6.6.1 (5635776) started at Wed 08 Dec 2021 04:16:37 PM CET.
--- RBK0151I: Using backuppath /backup with partition type nfs.
!!! RBK0216W: Applying backup strategy in /backup/pi4 only.
--- RBK0218I: Applying smart backup strategy. Daily:7 Weekly:4 Monthly:12 Yearly:1.
--- RBK0217I: 7 backups will be smart recycled. 4 backups will be kept. Please be patient.
!!! RBK0220W: Smart backup strategy would delete /backup/pi4/pi4-rsync-backup-20211206-202155.
!!! RBK0220W: Smart backup strategy would delete /backup/pi4/pi4-rsync-backup-20211207-151240.
!!! RBK0220W: Smart backup strategy would delete /backup/pi4/pi4-rsync-backup-20211208-082059.
!!! RBK0220W: Smart backup strategy would delete /backup/pi4/pi4-rsync-backup-20211208-083111.
!!! RBK0220W: Smart backup strategy would delete /backup/pi4/pi4-rsync-backup-20211208-083136.
!!! RBK0220W: Smart backup strategy would delete /backup/pi4/pi4-rsync-backup-20211208-091434.
!!! RBK0220W: Smart backup strategy would delete /backup/pi4/pi4-rsync-backup-20211208-161058.
!!! RBK0222W: Smart backup strategy would keep /backup/pi4/pi4-rsync-backup-20211206-183706.
!!! RBK0222W: Smart backup strategy would keep /backup/pi4/pi4-rsync-backup-20211206-203543.
!!! RBK0222W: Smart backup strategy would keep /backup/pi4/pi4-rsync-backup-20211207-205432.
!!! RBK0222W: Smart backup strategy would keep /backup/pi4/pi4-rsync-backup-20211208-161636.
--- RBK0033I: Please wait until cleanup has finished.

--- RBK0218I: Applying smart backup strategy. Daily:7 Weekly:4 Monthly:12 Yearly:1.
--- RBK0217I: 7 backups will be smart recycled. 4 backups will be kept. Please be patient.
!!! RBK0220W: Smart backup strategy would delete /backup/pi4/pi4-rsync-backup-20211206-202155.
!!! RBK0220W: Smart backup strategy would delete /backup/pi4/pi4-rsync-backup-20211207-151240.
!!! RBK0220W: Smart backup strategy would delete /backup/pi4/pi4-rsync-backup-20211208-082059.
!!! RBK0220W: Smart backup strategy would delete /backup/pi4/pi4-rsync-backup-20211208-083111.
!!! RBK0220W: Smart backup strategy would delete /backup/pi4/pi4-rsync-backup-20211208-083136.
!!! RBK0220W: Smart backup strategy would delete /backup/pi4/pi4-rsync-backup-20211208-091434.
!!! RBK0220W: Smart backup strategy would delete /backup/pi4/pi4-rsync-backup-20211208-161058.
!!! RBK0222W: Smart backup strategy would keep /backup/pi4/pi4-rsync-backup-20211206-183706.
!!! RBK0222W: Smart backup strategy would keep /backup/pi4/pi4-rsync-backup-20211206-203543.
!!! RBK0222W: Smart backup strategy would keep /backup/pi4/pi4-rsync-backup-20211207-205432.
!!! RBK0222W: Smart backup strategy would keep /backup/pi4/pi4-rsync-backup-20211208-161636.
--- RBK0017I: Backup finished successfully.
--- RBK0010I: pi4: raspiBackup.sh V0.6.6.1 (5635776) stopped at Wed 08 Dec 2021 04:16:47 PM CET with rc 0.
--- RBK0026I: Debug logfile saved in /backup/pi4/pi4-rsync-backup-20211208-161636/raspiBackup.log.

Look at the target directory:
pi@pi4:~ $ less  /backup/pi4/pi4-rsync-backup-20211208-161636/
/backup/pi4/pi4-rsync-backup-20211208-161636/: Permission denied
pi@pi4:~ $

The directory looks like a normal backup. But there is absolutely no data stored. So if someone cleans backups up its likely thet the empty backup is kept but useable backups are destroyed.

The problem seems to occure only if smart recycle is executed. I did not dare to check whether it is only when using dry run mode.

Log attached.
[raspiBackup.log](https://github.com/framps/raspiBackup/files/7677066/raspiBackup.log)

framps, 2021-12-08:
I just created a fix. Please grab raspiBackup.sh from https://github.com/framps/raspiBackup/tree/issue409 an verify.

mcm1957, 2021-12-09:
Hi
SUPER FAST RESPONSE

I will grab the new release. But I currently have troubles retesting as I installed new software at my pi (ioBroker) which uses acls heavily. And acls do not really work with synology and rsync :-(. You are aware of this as I found your post at syno forum. I'm currently investigation this problem and have opened an official synology issue to clearify whether synology supports or blocks acls over nfs. 

I'll give feedback as soon as I was able to retest the fix.
Sorry for the delay.

framps, 2021-12-09:
Take your time.

> I have opened an official synology issue

:+1: Would be great if you share Synologies response in the Syno forum thread I created :wink:

mcm1957, 2021-12-09:
I've installed the hotfix:
20211209-191423 DBG 1937:  --- raspiBackup 0.6.6.1-hotfix, $/$ - 

The issues are fixed. I could not reproduce the problem.

-) applying smart strategy is now executed only once
-) logfile is now located at home driectory, so no empty backup directory is created.

All attach the log for reference
[raspiBackup.log](https://github.com/framps/raspiBackup/files/7687538/raspiBackup.log)

I noticed that the commit remark for raspiBackup notes fixe for issue 408 although it seems to fix 409. The branch refers to 409.
[image]

Issue can be close from my side.