framps/raspiBackup #410: xattr again - Raspian Bullseye

upD8R, 2021-12-10:
Hi there,

I again struggle with rsync error 23 after upgrading one of my systems to Bullseye.
`--- RBK0085I: Backuperstellung vom Typ rsync gestartet. Bitte Geduld.
20211210-155509 DBG 1978:                  --> executeCmd Command: rsync --exclude="/media/nfs/backups"                         --exclude="/home/@USER@/raspiBackup.log"                        --exclude="/home/@USER@/raspiBackup.msg"
                --exclude='.gvfs'                       --exclude=/proc/*                       --exclude=/lost+found/*                         --exclude=/sys/*                        --exclude=/dev/*                        --exclude=/boot/*
                      --exclude=/tmp/*                        --exclude=/run/*                        --exclude=/@HOSTNAME@-backup.img --exclude=/@HOSTNAME@-backup.tmg --exclude=/@HOSTNAME@-backup.sfdisk --exclude=/@HOSTNAME@-backup.mbr --exclud
e=/raspiBackup.log --exclude=/raspiBackup.msg                    /var/log/journal                                                --numeric-ids                   -aHAx --delete                                                                  /
                      "/media/nfs/backups/@HOSTNAME@/@HOSTNAME@-rsync-backup-20211210-155329"
20211210-155509 DBG 1979:                      --- Redirect: & - Skips:
rsync: [generator] set_acl: sys_acl_set_file(journal, ACL_TYPE_ACCESS): Operation not supported (95)
rsync: [generator] set_acl: sys_acl_set_file(journal/e6c89af4771644f1b21d62a026e65b00, ACL_TYPE_ACCESS): Operation not supported (95)
rsync: [receiver] set_acl: sys_acl_set_file(journal/e6c89af4771644f1b21d62a026e65b00/.system.journal.rWAmoa, ACL_TYPE_ACCESS): Operation not supported (95)
rsync: [receiver] set_acl: sys_acl_set_file(journal/e6c89af4771644f1b21d62a026e65b00/.system@0005d1b011c07ac7-eab116ef5bcdc6e4.journal~.knMDx9, ACL_TYPE_ACCESS): Operation not supported (95)
rsync: [receiver] set_acl: sys_acl_set_file(journal/e6c89af4771644f1b21d62a026e65b00/.system@0005d1b011ce1e1f-e8e8b0536ee5b969.journal~.S3Wyh6, ACL_TYPE_ACCESS): Operation not supported (95)
rsync: [receiver] set_acl: sys_acl_set_file(journal/e6c89af4771644f1b21d62a026e65b00/.system@0005d1b141ae6155-1685a591ae720d84.journal~.G91KI9, ACL_TYPE_ACCESS): Operation not supported (95)
rsync: [receiver] set_acl: sys_acl_set_file(journal/e6c89af4771644f1b21d62a026e65b00/.user-1000.journal.gVaB76, ACL_TYPE_ACCESS): Operation not supported (95)
rsync: [generator] set_acl: sys_acl_set_file(var/log/journal, ACL_TYPE_ACCESS): Operation not supported (95)
rsync: [generator] set_acl: sys_acl_set_file(var/log/journal/e6c89af4771644f1b21d62a026e65b00, ACL_TYPE_ACCESS): Operation not supported (95)
rsync: [receiver] set_acl: sys_acl_set_file(var/log/journal/e6c89af4771644f1b21d62a026e65b00/.system.journal.pfEko7, ACL_TYPE_ACCESS): Operation not supported (95)
rsync: [receiver] set_acl: sys_acl_set_file(var/log/journal/e6c89af4771644f1b21d62a026e65b00/.system@0005d1b011c07ac7-eab116ef5bcdc6e4.journal~.pLQpu7, ACL_TYPE_ACCESS): Operation not supported (95)
rsync: [receiver] set_acl: sys_acl_set_file(var/log/journal/e6c89af4771644f1b21d62a026e65b00/.system@0005d1b011ce1e1f-e8e8b0536ee5b969.journal~.C02LB6, ACL_TYPE_ACCESS): Operation not supported (95)
rsync: [receiver] set_acl: sys_acl_set_file(var/log/journal/e6c89af4771644f1b21d62a026e65b00/.system@0005d1b141ae6155-1685a591ae720d84.journal~.cnbh39, ACL_TYPE_ACCESS): Operation not supported (95)
rsync: [receiver] set_acl: sys_acl_set_file(var/log/journal/e6c89af4771644f1b21d62a026e65b00/.user-1000.journal.BSuBja, ACL_TYPE_ACCESS): Operation not supported (95)
rsync error: some files/attrs were not transferred (see previous errors) (code 23) at main.c(1333) [sender=3.2.3]
` 
I tried to exclude /var/log/journal as test but somehow this doesn't work properly. Any idea?

Many thanks!

framps, 2021-12-10:
That's a duplicate. Please read [this issue](https://github.com/framps/raspiBackup/issues/393) which discusses exactly your issue :wink: and close this issue.

upD8R, 2021-12-10:
Sure, thanks.

Just an additional hint: The path of vcgemcmd has also changed, it's now in /usr/bin (it is e.g. used in raspiBackup_temp_pre.sh)

closed 2021-12-10 by upD8R

framps, 2021-12-10:
Thank you very much for the hint. It's also already fixed :wink: See [here](https://github.com/framps/raspiBackup/issues/399)

upD8R, 2021-12-11:
It is verified and works! Thanks for your support!