framps/raspiBackup #411: RBK0203E - Native NVME (m.2) fails to discover boot device.

Leidrin, 2021-12-11:
I converted my raspberry over to Native NVMe boot today using an m.2 carrier board and raspiBackup can no longer locate boot device and will not initiate a backup.

When attempting a backup I see "RBK0203E: Unable to discover boot device..."

I uninstalled raspiBackup and reinstalled it, reconfigured my backup settings, and tried manually telling it to back up the partitions on the NVME (which it does see when manually selecting partitions, but still backup fails).

I did not see anything on the messages page about this so here is my debug log!

[raspiBackup.log](https://github.com/framps/raspiBackup/files/7696387/raspiBackup.log)

framps, 2021-12-11:
According the log
```
20211210-195133 DBG 5658:              /dev/nvme0n1p2 on / type ext4 (rw,noatime)
```

It's because nvme uses a different naming schema for partitions than SD cards (`/dev/mmcblk) or disks (`/dev/sd). 

I don't have a NVMe and thus never faced this issue. According [this thread](https://unix.stackexchange.com/questions/452120/persistent-device-naming-for-nvme-storage-devices)

```
the NVMe naming standard describes:

    nvme0: first registered device's device controller
    nvme0n1: first registered device's first namespace
    nvme0n1p1: first registered device's first namespace's first partition
```

the naming is much more complicated than with SD cards. 

The boot discovery is very sensitive and I have to carefully execute tests in order to support NVMes with raspiBackup. But I don't have neither a spare Raspberry nor a NVMe to execute any tests to add NVMe support :disappointed: 

Sorry about this.

> I did not see anything on the messages page about this so here is my debug log!

I'll update the page NVMe is not supported.

Leidrin, 2021-12-11:
Thanks for getting back to me.

I love this utility so if you are willing to work with me on testing I'd be happy to perform any necessary tests and validation on my end. I can afford to lose the data on the SSD if it goes wrong.

It looks like the NVME naming convention is consistent. I've wiped the pi a couple times, and looked at others who have moved to NVME boot, and they all follow this structure (as you alluded to above):

nvme0n1     259:0    0 238.5G  0 disk 
├─nvme0n1p1 259:1    0   256M  0 part /boot
└─nvme0n1p2 259:2    0 238.2G  0 part /

with 1p1 being boot and 1p2 being root.

This seems to be identical to the SD card layout, just with the partition numbers increased by 1:

mmcblk0     179:0    0 59.5G  0 disk 
├─mmcblk0p1 179:1    0  256M  0 part /boot
└─mmcblk0p2 179:2    0 59.2G  0 part /

Just with "mmcblk0" with "nvme0n1"

Given most carrier boards are only going to support a single m.2, it could be safely assumed that, if you discover a root and boot partition on nvme0n1 it could be safely assumed to be the boot device.

For clarity's sake, this is using the Rpi foundation's instructions for enabling NVMe boot, so anyone following those instructions should have a similarly predictable partition layout.

framps, 2021-12-11:
> I love this utility so if you are willing to work with me on testing I'd be happy to perform any necessary tests and validation on my end. I can afford to lose the data on the SSD if it goes wrong.

Thank you very much you offer to help. But unfortunately support of NVMe is a bigger effort (feature request) and the new hardware is required by development in order to create new code and execute tests.

Please not it's not just NVMe support for backup and restore. raspiBackup is also able to backup and restore between SD card and USB disk. With NVMe support raspiBackup has to be able to backup and restore also from SD card and USB disk to NVMe SSD and vice versa :wink: 

I created issue #412 and will start working on it as soon as the donation bucket of raspiBackup allows to buy an adapter and a NVMe. 

I again appreciate you want to test and help and give feedback but that's too inefficient for me and waste of my spare time. NVMe support will be available much faster if I have the required HW on site.

Said this you either have to switch to another backup utility or wait until NVMe support is available. Given the current willingness to donate I expect this feature to be available somewhere next summer - unfortunately.

Leidrin, 2021-12-11:
Thanks for your candor on this. I totally understand the time commitment and wanting to be able to run tests yourself.

What would you need for an adapter/carrier and and nvme ssd for testing? I'm willing to contribute! :)

framps, 2021-12-13:
Let's close this issue and continue on #412

closed 2021-12-13 by framps