framps/raspiBackup #412: Support NVMes in raspiBackup

framps, 2021-12-11:
As of now SD cards and USB disks are supported. Given the number of people which use a fast NVMe on their Raspberry4/8GB NVMes should be supported.

Leidrin, 2021-12-12:
Thanks for opening this as an issue appreciate your willingness to put this in.

As mentioned #411 I am happy to sponsor an nvme ssd and appropriate carrier or adapter (I use the Waveshare CM4 carrier model A personally) - just let me know an appropriate amount to donate :) I understand this will still be subject to your time and availability, and will not impose any expectation for delivery time.

framps, 2021-12-12:
Which adapter do you use exactly? I'm new to the NVMe area and just learned there are adapters out there which present the NVMe as a normal disk as /dev/sdx . That will work with raspiBackup. So I have to make sure the NVMe adapter I will use for evelopment will use /dev/nvmexnyp1 instead of /dev/sdx

Leidrin, 2021-12-12:
I am using the waveshare carrier board with m.2 slot : https://www.waveshare.com/product/raspberry-pi/boards-kits/compute-module-4-cat/cm4-io-base-a.htm

This one is a direct pcie handoff to the cm4, where (I think) the ones that present as /dev/sdx run through USB (and are a little slower as a result).

framps, 2021-12-12:
I frankly didn't know until now there exists a cm4. I thought I can just buy an adapter and a NVMe and connect them via USB to my existing Raspberry3. 

In order to make sure the NVMe is identified as /dev/nvme by Linux and I can add the required code and execute tests with /dev/nvme we have to be careful. An NVMe which is identified as /dev/sdx will not help me.

Do you know whether there is an adapter out there which identifies a NVMe as /dev/nvme and I can plug in into my existing Reaspberry3 for development? If not I should use exactly the HW you use: waveshare adapter, NVMe and cm4.

Leidrin, 2021-12-12:
I will inquire (and hopefully get a response) from Jeff Geerling, who's blog on NVMe boot got me to make the jump. He's quite the tinkerer and runs a hardware compatibility database for pi add-ons so he may know of something.

Otherwise we can talk about getting a full kit (CM4 + carrier board + nvme) ordered :)

Leidrin, 2021-12-13:
Jeff got back to me (Super nice guy!) and confirmed that anything hooked up to USB would present itself as /dev/sdx or /dev/mmcx rather than /dev/nvmex - that being the case, seems we'll need to get you a full kit!

Let me know what the cost would be for you to order a full CM4 kit (Below). If you feel you need more than the 1gb base ram I'd be fine going to a 2gb unit.

1. Pi CM4 1/2GB - would recommend one with EMMC so you go through the experience of modifying the bootloader and validating the built in emmc does not interfere with your program.
2. CM4 Carrier board with M.2 - any should work but the waveshare one linked above is what I use
3. 2230/2242 sized (30/42mm) m.2 NVMe SSD - I use a little 30mm kioxa I salvaged from a dell laptop, but any 2230/2242 nvme  should work.

If you need any help with parts selection feel free to ping me!

framps, 2021-12-13:
> 1. would recommend one with EMMC so you go through the experience of modifying the bootloader and validating the built in emmc does not interfere with your program.

Would you please give me some additional information/links how to do this?

re 1) I found https://www.welectron.com/Raspberry-Pi-CM4001008-Compute-Module-8-GB-1-GB-RAM for about €30+€5 shipping
re 2) I found https://www.waveshare.com/product/raspberry-pi/boards-kits/compute-module-4-cat/cm4-io-base-a.htm but I frankly don't want to bother with any customs and try to find a seller in Germany but failed until now :cry: I found other adapter for about €100 but that's too expensive.
re 3) I found https://www.reichelt.de/kioxia-bg4-client-ssd-128gb-m-2-2230-kbg40zns128g-p317830.html?utm_source=Preisvergleich_&utm_medium=CPC&utm_campaign=Preisvergleich_&utm_source=psuma&utm_medium=idealo.de&PROVID=2378 for about €50 + €5 shipping

> If you feel you need more than the 1gb base

1gb is enough. 128gb NVMe is also enough. I just need a small system which populates /dev/nvmex :smile:

framps, 2021-12-13:
> feel free to ping me!

If you don't mind let's continue the HW discussion via eMail. Please see [here](https://www.linux-tips-and-tricks.de/de/kontakt/) for my email address.

Leidrin, 2021-12-14:
I will send you an e-mail :) Response will be a little slow from me this week, preparing for holiday time off at work, but we'll figure this out!

framps, 2022-01-23:
Just created [Branch for feature](https://github.com/framps/raspiBackup/tree/NVMe)

framps, 2022-04-20:
@Leidrin Do you still want to have this feature added to raspiBackup? It's a long time I got your last mail: 2/24. 

Every day a power adapter from US to DE on my desk remembers me you created this feature request and and you wanted to send me the required HW kit in order to this feature :wink:

Leidrin, 2022-04-22:
Hi Peter!

I won't bore you with the details, but things have not been great on my side since we last spoke. Apologies for the delay in response.

I asked my partner and it looks like she never sent out the kit I put together for you, we found it still sitting in her car. Things have been chaotic to say the least so I certainly don't blame her, but owe you an apology there.

We are both working through this weekend (part of the craziness) so I dont think we can get it tomorrow/sunday but will get this sent out next weekend (4/30). Thanks for your patience and sorry to have strung you along this long.

-Matt

Leidrin, 2022-05-31:
> Hi Peter!
>
> Here is the traking - I was informed once this goes international you won't be able to track anymore but you can see up until it's put on the boat:
>
> https://tools.usps.com/go/TrackConfirmAction?tLabels=CJ474926234US
>
> This is everything except the power adapter (see attached text file for contents) which I will order directly from a German retailer for you since those are readily available. Going to wait and make sure the package is progressing as expected before ordering the adapter since I imagine you'll get that pretty quickly.
>
> Thanks again Peter and sorry for the wait!
>
> -Matthew
>
>
> Hi Peter!
>
> Package is finally sent! I probably should have just ordered from German merchants since the shipping was pretty high,  but it's finally on the way! I'll send you tracking once it's active.
>
> Your friend,
>
> Matt

Hi Peter,

Hope you are well. Let me know when you get back from your travels and if the kit arrived, and I'll get the power supply ordered locally for you.

Safe travels!

-Matt

framps, 2022-06-04:
Hi Matt,

good timing :+1: because I'm just back home to receive the kit. Until now it didn't arrive but I may have to pick it up at the customs station. Waiting for a customs notification now ...

Leidrin, 2022-06-07:
Great News! Let me know once it's in hand and I'll get the power supply ordered for you.

framps, 2022-06-10:
> once it's in hand

Mission completed :smiley: . I just picked up the kit from the post office (had to pay €31 customs duty :angry: ) 

[image]

> I'll get the power supply ordered for you.

Thank you very much. Please note: I'll be out mid next week and back on 7/5. Please make sure I'm back home when the power supply will be delivered.

One question: I detected an item I don't know what it's useful for: Some HDMI plug which is closed at the end.

[image]

Leidrin, 2022-06-10:
So glad everything finally arrived! I'll send a donation your way to cover the customs duty charge. Between the shipping cost + customs definitely a lesson learned here, if I need to send something to European countries in the future, ill try and buy from a local merchant and do direct-ship.

The HDMI dongle is a headless HDMI adapter. I found that the Pi doesn't activate it's GPU if it's run headless (ssh/vnc only with no monitor connected) which causes some issues in Chromium or other apps which expect hardware acceleration. The Dongle acts as a dummy monitor so the GPU activates. Not necessary to use but a nice to have and I had a spare :)

I have set a calendar reminder to order the adapter on 7/4 so it arrives after you are back. Feel free to send me a private e-mail if you'd like to give me your PayPal to send the €31 or I can donate directly to your github if you take donations here.

framps, 2022-06-10:
> Between the shipping cost + customs definitely a lesson learned here, if I need to send something to European countries in the future, ill try and buy from a local merchant and do direct-ship.

That's definitely a good strategy :+1:  And it will also be much faster. It took approx 4 weeks from US to Germany :disappointed:  Usually it takes a couple of days if you order in Germany. But this requires the parts are available in Europe. Raspberries are quite expensive in Europe as of now. Maybe CM4s are cheaper. I don't know.

> The HDMI dongle is a headless HDMI adapter. I found that the Pi doesn't activate it's GPU if it's run headless (ssh/vnc only with no monitor connected)

I connect a monitor all the time I develop raspiBackup on a tiny image and use ssh. But it's good to know I have to use the HDMI dongle if I run the CM4 headless.

> if you'd like to give me your PayPal

Just use my eMail you used for our private communication :wink: 

I already started to update the NVMe branch and I think I'm done (theoretically) with the backup process: raspiBackup running  on NVMe will detect boot and root are stored on a NVMe and backup both partitions to some external storage.

Now I'm wondering how I can restore the backup: As far as I understand I have to use any Linux box or Raspberry4 to restore the backup. But this requires  to mount the NVMe device from the cm4 on the Linux box. (That's the -d option parameter). Would be great if you can give me instructions how I can achieve this.

framps, 2022-06-10:
> Now I'm wondering how I can restore the backup: As far as I understand I have to use any Linux box or Raspberry4 to restore the backup. But this requires to mount the NVMe device from the cm4 on the [...]

Just detected `/github.com/raspberrypi/usbboot/mass-storage-gadget` will export eMMC and NVMe and they can be mounted on a Linux box. Do you have any other process in mind?

Leidrin, 2022-06-10:
Was just working with the stock rpiboot tool trying to figure out a process for this. My initial imaging was do use rpiboot to mount the emmc or SD card as USB storage, image that, then use the Raspberry's "SD Card Copier" to move the data to the NVME. I hadn't thought of how to direct-image to the NVMe. I'll try that utility you just linked and let you know if it works for me!

Leidrin, 2022-06-10:
Just ran through the process and the mass-storage-gadget (part of rpiboot/usbboot which I had been looking at) is exactly what we needed.

Here are full instructions (Sourced from: https://jamesachambers.com/full-compute-module-4-raspberry-pi-setup-imaging-guide/ combined with the mass-storage-gadget documentation on github):

Disconnect power from the CM4 carrier board, and change the USB boot switch behind the HDMI port to "ON"

Disconnect the CM4 from power and connect a USB cable from your other Pi/Linux computer to the CM4 Carrier's USB-C port

Open terminal on your linux computer and run the following to set up rpiboot/usbboot:

`sudo apt install git libusb-1.0-0-dev build-essential`
`git clone --depth=1 https://github.com/raspberrypi/usbboot`
`cd usbboot`
`make`
`sudo make install`

Then to mount the drives:
`cd mass-storage-gadget`
`sudo ../rpiboot -d .`

Once done the boot and rootfs partitions should mount. The CM4 I sent you also has EMMC on board (mine do not, could not find the ones with EMMC in stock) - I set up the bootloader on your CM4 to use the NVMe first, but it's possible this may mount the EMMC and NVME. I'll be curious to see if only this primary boot device (NVMe) is loaded, or both with your unit.

framps, 2022-06-11:
Coding is now complete :wink: 

I tried to power the cm4 via USB3 from my desktop but cm4 failed during start. Then I tried my Samsung mobile power adapter which provides 2A and was able to boot cm4 :smile: .

I spent some time to get familiar with cm4 and finally started raspiBackup on the system booted from NVMe. Backup was created successfully. Next I connected the cm4 to my Linux desktop according the your instructions above. Partitions for eMMC and NVMe popped up in the UI. Next I umounted the NVMe partitions (that's required by raspiBackup). Next I cleared the NVMe partitions with `sudo dd if=/dev/zero of=/dev/<nmve-device> bs=4Mib count=1000` to clear the NVMe partitions. Finally I  restored the backup on the clean NVMe partition. Next I started cm4 again and it booted the restored image :smile: .

I used an nfs attached backup space in order to reduce the cm4 current and backup type rsync. Please download  [this version](linux-tips-and-tricks.de/raspiBackupNVMe.sh) and execute tests on your environment now.  If you face any issues please describe the issue and attach the debug log so I will be able to check what's wrong and fix the issue.

Would be great if you could also test other backup types like dd and tar and also backup mode partition oriented. When I have an appropriate power supply I will execute additional unit tests and test edge conditions. 

Please note: You have to use option `--unsupportedEnvironment` if you execute the restore on your Laptop running Linux. In addition - if your Laptop doesn't run Buster but an older Linux distro - you have to delete line 5 introduced by a new version of sfdisk in Buster in *.sfdisk file in the backup directory first before you start the restore.

Leidrin, 2022-06-12:
Was able to successfully execute tests with Rsync and TAR backup types. I was able to complete a backup with DD but when attempting to restore (using a second raspberry pi running usbboot) I get:

`./raspiBackupNVMe.sh: line 5261: executeCommandDD: command not found`
`??? RBK0035E: Backupprogram dd failed during restore with RC 127.`

I looked down at 5261 (and the $cmd variable it's driven off of) and looks like it's just calling standard dd, which works on the pi, so not sure what the issue is.

Debug log attached

[raspiBackupNVMe.zip](https://github.com/framps/raspiBackup/files/8886173/raspiBackupNVMe.zip)

framps, 2022-06-12:
> Was able to successfully execute tests with Rsync and TAR backup types.

Great news :+1: 

That's a bug in master (#502). Looks like dd backups are only restored on Windows because until now nobody reported the bug. I just fixed it and published the new version on the link above. You may also just change `executeCommandDD` to `executeDD` in your local version.

I now will check why my regression test didn't catch this issue.

Leidrin, 2022-06-12:
I made the DD command change and the script executed perfectly in DD mode. I also added 2 more rsync version for a total of 3, and restored the latest to be sure versioned worked, everything is as expected!

Let me know if there are any additional tests I can run which would be helpful.

framps, 2022-06-12:
> script executed perfectly in DD mode.

That's what I expected :smile: I frankly have no idea where the typo came from. The regression test detects this issue. 

> to be sure versioned worked, everything is as expected!

There were no changes in raspiBackup logic other than NVMe discovery and handling of NVMe partitions and that's why I don't expect any issues there. We only have to make sure all backup types (dd, tar and rsync) still work correctly on real CM4 HW. I will add NVMe tests in my QEMU regression tests to test there is no regression in NVMe device handling but that's no real HW test.  

> Let me know if there are any additional tests I can run which would be helpful.

Would be great if you could test partition oriented mode also. It's a totally different code path which will be executed. You don't have to create additional partitions on NVMe. Just use the partition oriented mode and save the two existing standard partitions.

1) use option -P with -t tar
2) use option -P with -t rsync

Thank you very much for your test support!

Leidrin, 2022-06-13:
Happy to help!

Both (rsync/tar with -P argument added) completed without issue.

framps, 2022-06-13:
Perfect :+1: 

That's what I'm going to do next:

1) Spend some weeks in Bavaria and it's beergardens
2) When I return I order a new SSD. My SSD of my desktop crashed last night :cry: Unfortunately the desktop hosted my development environment including the regression test environment
3) Add NVMe in my regression test
4) Execute some unit tests for NVMe on my CM4
5) Run final regression tests
6) Publish 0.6.8 beta with NVMe support
7) Wait for feedback and publish 0.6.8 4 weeks later

Just go ahead and test thoroughly your use cases and let me know if you find any new issue. As long as I don't have to reproduce the issue on my CM4 at home I will be able to work on the issue and create any required fix on my Thinkpad.

Not sure whether I get much feedback for 0.6.8 beta because as far as I can see all users of raspiBackup use a Raspberry but no CM4.

framps, 2022-06-15:
I just double checked whether the current NVMe development commit of raspiBackup will be notified if 0.6.8 beta or even the final 0.6.8 release will be available: A notification message will be written.

When I tested this I noticed I didn't ask you whether the current support for NVMe I added is exactly what you need for your CM4s or if there is anything else missing ... I got the impression you're fine now.

Leidrin, 2022-06-16:
Enjoy your travels! That sounds like a wonderful trip. Sorry to hear about your SSD, and will again make sure to order the CM4 power adapter for you shortly before you come back (let me know if your timetable changes).

The support thusfar is exactly what I need. I look forward to testing the full flow with your UI configurator once the beta is ready, and will continue to run tests while you are travelling (have a cron job set for both my CM4s presently). Have yet to experience anything and believe we tested most backup/restore options, but if anything pops up I'll post right away.

Thank you for taking on this feature request for me! I hope to see the CM4/NVMe community grow, especially as more and more boards are being made available with NVMe support.

framps, 2022-07-13:
I finished all NVMe tests now and created a new release v0.6.7.1. It's not officially published because it's just v0.6.7 with NVMe support and I don't want to bother current raspiBackup users with a new release which just includes a feature they don't need.

raspiBackup 0.6.7.1 with NVMe support is available [here](https://linux-tips-and-tricks.de/de/component/joomdoc/raspibackup-0-6-7-1-sh/detail/?Itemid=200) for download and instructions how to install it.

NVMe support will become officially available in next raspiBackup release which will include in addition to NVMe support new features, enhancements and bug fixes. Users of raspiBackup 0.6.7.1 will get notified by raspiBackup when the next release will be available and then can upgrade to the official release which includes NVMe support.

@Leidrin Please close this issue if you're fine now.

Leidrin, 2022-07-16:
Hi Framp,

I am glad your trip went well! I am jealous of all the good food and beer :)

Thanks so much again, if you would still like me to order the power adapter for your CM4 kit let me know and I'll be happy to do so!

I believe you will have to close this since you opened the issue for me based upon my separate issue report. Take care and thank you again for this great utility!

framps, 2022-07-16:
Hi Leidrin,

> if you would still like me to order the power adapter for your CM4 kit let me know and I'll be happy to do so!

Just let me know whether another 2 A/10 W power adapter I have from Lenovo is capable to drive the CM4 reliable. If not please let me know what's the minimum amount of A/W the power adapter should have and I will order one.

> I believe you will have to close this since you opened the issue for me

You are right. I didn't remember it was me who created the issue on behalf of you.

Thank you very much again for your donation of a CM4. If you face any issues with NVMe support feel free to create another git issue :wink:

closed 2022-07-16 by framps

Leidrin, 2022-10-11:
Hi Peter!

Here is the traking - I was informed once this goes international you won't be able to track anymore but you can see up until it's put on the boat:

https://tools.usps.com/go/TrackConfirmAction?tLabels=CJ474926234US

This is everything except the power adapter (see attached text file for contents) which I will order directly from a German retailer for you since those are readily available. Going to wait and make sure the package is progressing as expected before ordering the adapter since I imagine you'll get that pretty quickly.

Thanks again Peter and sorry for the wait!

-Matthew

Included:

Standalone parts:
CM4 1gb w/ 32gb EMMC + wifi
Wavshare CM4 carrier board w/ PCIE M.2 slot
128gb M.2 2232 SSD

Parts bag:
4x small screw w/ cap (Feet for carrier board)
2x M.2 SSD screw (spare included!)
HDMI dummy plug (CM4 GPU acceleration needs a display present to activate, dummy plug does that)
Heatsink for CPU

Extra info:
Pi OS (Jan 2022) installed on both M.2 and EMMC w/ default user/pass pi/raspberry and DHCP set if you want to use that. M.2 should be default boot device once connected. OS should be identical except I changed wallpaper to differentiate between the EMMC and M.2 boot environments.

Otherwise image the CM4 using your PC and USB OTG (I can send you guides if needed!).

Leidrin, 2022-10-11:
Hi Peter!Package is finally sent! I probably should have just ordered from German merchants since the shipping was pretty high,  but it's finally on the way! I'll send you tracking once it's active. Your friend, Matt