framps/raspiBackup #413: Restore a backup program from a SSD to a smaller SD card

WolfS2711, 2021-12-13:
Backup of a FHEM program on a raspberry 4.
Trying to restore a "tar" - backup with 2,5 GB  from a SSD (120GB) to a SD-card with 16 GB ends with the failure 112. (Partitioning is not possible). I thought, that tar- programs can be transferred to smaller memories, if the content  of the backup is smaller than the memory size.
Thank you for any advice.
Best regards
Wolfgang
[raspiBackup.log](https://github.com/framps/raspiBackup/files/7705462/raspiBackup.log)

framps, 2021-12-13:
I was able to reproduce the issue and have to fix it. But this will take some time. There are a workarounds available:

## Workaround 1:
In the backup directory there is a file with extension .sfdisk. Just save the current file somewhere and modify the file. 

Example :

My sfdisk file of a 16GB SD card  backup is

```
label: dos
label-id: 0x3c3f4bdb
device: /dev/mmcblk0
unit: sectors
sector-size: 512

/dev/mmcblk0p1 : start=        8192, size=      524288, type=c
/dev/mmcblk0p2 : start=      532480, size=    29917184, type=83
```

In your case the number `29917184`  (approx 14GB) should be something about `251658240` (approx 120GB). Just change this number to some value which is less than 16GB, for example 15GB which is `31457280` `(15*1024*1024*1024/512)`. Then start the restore.

## Workaround 2:
Manually partition your SD card with a boot partition /dev/mmcblk0p1 of 253M and allocate the rest to /dev/mmcblk0p2. Then add option `-0` (dash zero) when you restore the backup.

Just pick one of the workarounds :wink:

WolfS2711, 2021-12-13:
> I created a branch for you issue which contains a fix for your
> scenario. Please download raspiBackup.sh from
> [here](https://raw.githubusercontent.com/framps/raspiBackup/issue413/raspiBackup.sh)
> into /home/pi and call `sudo ./raspiBackup.sh -d /dev/sda
> <yourBackupDir>` in directory /home/pi. Please note the leading dot !
> Otherwise you will not use the fixed raspIBackup code.

Thank you very much for the fix.
I'll try it as soon as possible.

Best regards 
Wolfgang  

⁣BlueMail for Android herunterladen ​

WolfS2711, 2021-12-16:
Thank you for the quick Hotfix; it works!

framps, 2021-12-16:
Thank you very much you verified the fix :+1: . I'll merge the fix into next release.

closed 2021-12-16 by framps