framps/raspiBackup #414: Restore bricht ab mit "no such file or directory"

Steinspiel, 2021-12-14:
Mein Setup:

pi4, nur mit 120GB M2.SSD,
diese hat eine 256 MB boot-Partition sowie eine ca. 28 GB root-Partition, der Rest ist nicht zugeordnet
der Pi bootet problemlos von SSD, openHABian 3.0 läuft.
eine 2 TB USB 3.0 Toshiba Platte (eine einzige Partition, ext4) für`s Backup, hängt am aktiven USB3.0 Hub am Raspi
eine zweite, baugleiche 120GB M2.SSD wie oben, auf der das restore landen soll, hängt ebenfalls am aktiven USB3.0 Hub am Raspi 

Um weitgehendst Hardwarefehler aus zu schließen ist die USB Platte durch eine heute gekaufte, neue Platte ersetzt worden.

Das Backup läuft Problemlos durch, beim Restore erscheinen Fehlermeldungen wie auf dem Screenshot_01.
https://imgur.com/DeBVV5f

Auf Screenshot_02 ist zu erkennen welche Files in /backup... enthalten sind
https://imgur.com/OjgsyMo

angehängt "raspiBackup.log" einmal **nach** erstellen des Backups, einmal **nach** Abbruch des Restore...

Danke

[raspiBackup_nach_backup.log](https://github.com/framps/raspiBackup/files/7714491/raspiBackup_nach_backup.log)
[raspiBackup_nach_restore.log](https://github.com/framps/raspiBackup/files/7714492/raspiBackup_nach_restore.log)
,

Steinspiel, 2021-12-15:
Moin framp,

Woher "4211.sfdisk" im Backupverzeichnis kommen kann, da habe ich auch keine Idee.
Allerdings habe ich festgestellt das in /home/openhabian/ zwei, vom Dateinamen her, ähnliche Dateien sind, wie Du auf angehängtem Screenshot sehen kannst...

Egal: "4211.sfdisk" habe ich im Backupverzeichnis gelöscht und das Restore lief danach glatt durch! Die wiederhergestellte SSD habe ich dann in den Raspi gebaut und sauber gebootet - alles bestens! :-)

Anschließend habe ich von der wiederhergestellten SSD ein weiteres Backup gemacht und abermals ein Restore - Super!

Ich bin wirklich völlig begeistert! Jetzt will ich sehen das ich regelmäßig ein automatisches Backup erstellen kann und diese "Telegram Sache" in Angriff nehmen...

Ich denke mein Ursprüngliches Problem war das wohl nicht mehr ganz sauber laufende Backup USB LW, hatte also **nichts** mit raspiBackup zu tun. Das anschließende Problem mit "4211.sfdisk" war dann auch nur ein blöder Zufall. Ich werde alles im Auge behalten und mich in ein paar Tagen noch mal melden und berichten wie es läuft.

Danke erst mal für Deine Geduld + Hilfe

[image]
,

closed 2021-12-15 by Steinspiel

framps, 2021-12-15:
> Woher "4211.sfdisk" im Backupverzeichnis kommen kann, da habe ich auch keine Idee.

raspiBackup legt temporaer eine Kopie der sfdisk Datei an und nimmt dazu den Namen $$.sfdisk wobei $$ die Prozessnummer von raspiBackup ist. Wieso sie plotzlich bei Dir im Backupverzeichnis auftaucht verstehe ich nicht. Anyhow habe ich in #415 der Code so geaendert so dass im naechsten Release diese Datei nicht mehr stoert. Ausserdem werde ich die temporaerer Datei zukuenftig definiert woanders ablegen.

Freut mich aber dass ich die Ursache aus dem Debuglog herausfinden konnte :smiley: und der Restore nun funktioniert.