framps/raspiBackup #418: Raspibackup restore goes to 148%

madbrain76, 2021-12-21:
I backed up a Raspberry Pi 3B running Bullseye 32-bit version onto my NAS using tar and samba. Here is the OS version for the backup.
pi@genie:~ $ cat /etc/os-release 
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
pi@genie:~ $ uname -a
Linux genie 5.10.63-v7+ #1496 SMP Wed Dec 1 15:58:11 GMT 2021 armv7l GNU/Linux
pi@genie:~ $ 

I tested the restore feature from my NAS running Ubuntu 20.04 . Here is the restore OS version :
root@server10g:/array/zfs/Backups/genie/genie-tar-backup-20211220-210403# uname -a
Linux server10g 5.11.0-37-generic #41~20.04.2-Ubuntu SMP Fri Sep 24 09:06:38 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux
root@server10g:/array/zfs/Backups/genie/genie-tar-backup-20211220-210403# cat /etc/os-release 
NAME="Ubuntu"
VERSION="20.04.3 LTS (Focal Fossa)"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 20.04.3 LTS"
VERSION_ID="20.04"
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
VERSION_CODENAME=focal
UBUNTU_CODENAME=focal

The restore worked. However,  the progress bar for the second partition went to 148% rather than 100%, which was surprising.

root@server10g:/array/zfs/Backups/genie/genie-tar-backup-20211220-210403# raspiBackup -g -d /dev/sdk .
--- RBK0009I: server10g: raspiBackup V0.6.6.1 (711a501) started at Mon 20 Dec 2021 09:09:28 PM PST.
!!! RBK0018W: Target /dev/sdk with 59.68 GiB is larger than backup source with 29.81 GiB. root partition will be expanded accordingly to use the whole space.
!!! RBK0065W: Device /dev/sdk will be repartitioned and all data will be lost.

--- RBK0066I: Device /dev/sdk will be overwritten with the saved boot and root partition.
--- RBK0069I: Bootpartition /dev/sdk1 will be formatted and will get the restored Boot partition.
--- RBK0070I: Rootpartition /dev/sdk2 will be formatted and will get the restored Root partition.
--- RBK0038I: Are you sure? y/N 
y
--- RBK0050I: Restoring backup from /array/zfs/Backups/genie/genie-tar-backup-20211220-210403.
--- RBK0004I: Adjusting second partition from 29.55 GiB to 59.43 GiB.
--- RBK0053I: Restoring first partition (boot partition) to /dev/sdk1.
 256MiB 0:00:04 [55.7MiB/s] [================================>] 100%            
--- RBK0055I: Restoring second partition (root partition) to /dev/sdk2.
8.43GiB 0:15:29 [9.28MiB/s] [=================================] 148%            

--- RBK0033I: Please wait until cleanup has finished.
--- RBK0076I: Restore finished successfully.
--- RBK0010I: genie: raspiBackup V0.6.6.1 (711a501) stopped at Mon 20 Dec 2021 09:28:53 PM PST with rc 0.
--- RBK0026I: Debug logfile saved in /home/madbrain/raspiBackup.log.

I have attached the log.

Also of note, the restore was quite slow at just 9.28MB/s. I was using a Sandisk reader with a Sandisk microSD capable of 170MB/s read speed / 90MB/s write speed with a Sandisk A2 card. I suspect the restore was slow because of sequential access to files in the tar restore. I just didn't expect a 90% slowdown in throughput.

[raspiBackup.log](https://github.com/framps/raspiBackup/files/7758724/raspiBackup.log)

framps, 2021-12-21:
The log shows

`20211220-211106 DBG 2273:                  --> executeCmd Command: pv -f /array/zfs/Backups/@HOSTNAME@/@HOSTNAME@-tar-backup-20211220-210403/@HOSTNAME@-tar-backup-20211220-210403.tar | tar --exclude /boot --same-owner --same-permissions --numeric-owner  -xf -`
20211220-212637 DBG 2293:                  <-- executeCmd 0`

and it took about 15 minutes to restore your backup with tar which should be fine.

raspiBackup just uses the Linux tools available for backup and restore and the local infrastructure. If you get 148% of CPU utilization that's what tar uses. Don't blame the user :wink: 

I frankly don't understand why you have an issue with 148% cpu usage used by tar. You just want to restore your backup which takes about 15 mins and is fine. why do you bother about the tar cpu usage to restore your backup? You finally want to have a  backup restored - and it shouldn't matter how long and how much cpu it costs :thinking:

madbrain76, 2021-12-21:
I don't really care about the percentage of CPU. I thought this was supposed to be the progress status. This percentage indeed increases from 0% to 148%. 148% is what it ends up with when the restore is finished.

It's very unlikely tar at 9MB/s would be using anywhere close to 148% of CPU on the NAS which is 4 core i5-6600k overclocked to 4.4 GHz.

madbrain76, 2021-12-22:
As to why it matters how long the restore is going to take - I like to have an idea upfront how long the process is going to take, so I know when to check when it's finished. I buy fast SD cards expecting fast restore times.

It also matters because of autosuspend. My system is setup to autosuspend after exactly 15 minutes, unless I'm logged in locally, or there is at least 600KBytes/s of network activity during. I usually control the system through VNC. I chose the 600Kbytes/s to allow the system to suspend even if I leave my VNC session open and inactive. I wake it back up with WOL even needed. Usually, I'll initiate a restore of an SD card from a VNC session. I don't want the machine to suspend in the middle of a restore. I would need to adjust the autosuspend heuristics to prevent that from happening. It could get complicated to figure out the proper heuristics. I guess I would need to use the KVM switch and login locally as a workaround instead of using VNC. But I would then need to remember to logout to allow the box to go back to sleep.

framps, 2021-12-22:
I see your point. I noticed you use option `-g` which because of a bug in current release writes the progress information into the debug log instead of the screen. I think the restore will be faster if you don't use this option for now.

I think the best solution for you would be to have an extension point which is called when the restore finishes and you can trigger any activity you need. I'll create a feature request on behalf of you.

closed 2022-01-06 by github-actions[bot]