framps/raspiBackup #419: Unable to restore backup from 64-bit Raspberry Pi OS

madbrain76, 2021-12-21:
I installed the latest 64-bit Raspberry Pi OS on my new Raspberry Pi 4B with 8GB, and backed it up using Raspibackup. Here is the log of the backup. It was a tar backup onto my NAS via cifs. The backup succeeded.

[raspiBackup.log](https://github.com/framps/raspiBackup/files/7758758/raspiBackup.log)

The OS on the Pi4B is as follows :
pi@pi64:~ $ uname -a
Linux pi64 5.10.63-v8+ #1488 SMP PREEMPT Thu Nov 18 16:16:16 GMT 2021 aarch64 GNU/Linux
pi@pi64:~ $ cat /etc/os-release 
PRETTY_NAME="Debian GNU/Linux 11 (bullseye)"
NAME="Debian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"
pi@pi64:~ $ 

I then tried to restore it on my Ubuntu 20.04 system. Here is the OS version on the NAS :
root@server10g:/array/zfs/Backups/pi64/pi64-tar-backup-20211220-203939# cat /etc/os-release 
NAME="Ubuntu"
VERSION="20.04.3 LTS (Focal Fossa)"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 20.04.3 LTS"
VERSION_ID="20.04"
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
VERSION_CODENAME=focal
UBUNTU_CODENAME=focal
root@server10g:/array/zfs/Backups/pi64/pi64-tar-backup-20211220-203939# uname -a
Linux server10g 5.11.0-37-generic #41~20.04.2-Ubuntu SMP Fri Sep 24 09:06:38 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux
root@server10g:/array/zfs/Backups/pi64/pi64-tar-backup-20211220-203939# 

The restore failed as follows :
root@server10g:/array/zfs/Backups/pi64/pi64-tar-backup-20211220-203939# raspiBackup -g -d /dev/sdk .
--- RBK0009I: server10g: raspiBackup V0.6.6.1 (711a501) started at Tue 21 Dec 2021 02:50:03 PM PST.
!!! RBK0006W: Target /dev/sdk with 29.81 GiB is smaller than backup source with 59.62 GiB. root partition will be truncated accordingly. NOTE: Restore may fail if the root partition will become too small.
!!! RBK0065W: Device /dev/sdk will be repartitioned and all data will be lost.

--- RBK0066I: Device /dev/sdk will be overwritten with the saved boot and root partition.
--- RBK0069I: Bootpartition /dev/sdk1 will be formatted and will get the restored Boot partition.
--- RBK0070I: Rootpartition /dev/sdk2 will be formatted and will get the restored Root partition.
--- RBK0038I: Are you sure? y/N 
y
--- RBK0050I: Restoring backup from /array/zfs/Backups/pi64/pi64-tar-backup-20211220-203939.
--- RBK0033I: Please wait until cleanup has finished.
??? RBK0077E: Restore failed. Check previous error messages.
--- RBK0010I: pi64: raspiBackup V0.6.6.1 (711a501) stopped at Tue 21 Dec 2021 02:50:13 PM PST with rc 112.
--- RBK0026I: Debug logfile saved in /home/madbrain/raspiBackup.log.
root@server10g:/array/zfs/Backups/pi64/pi64-tar-backup-20211220-203939# 

I have attached the corresponding restore log.
[raspiBackup.log](https://github.com/framps/raspiBackup/files/7758780/raspiBackup.log)

framps, 2021-12-21:
According the log

```
0211221-145005 DBG 3518:                  *** cat /array/zfs/Backups/@HOSTNAME@/@HOSTNAME@-tar-backup-20211220-203939/@HOSTNAME@-backup.sfdisk
20211221-145005 DBG 3518:                      label: dos
20211221-145005 DBG 3518:                      label-id: 0x8cf12228
20211221-145005 DBG 3518:                      device: /dev/mmcblk0
20211221-145005 DBG 3518:                      unit: sectors
20211221-145005 DBG 3518:                      sector-size: 512
20211221-145005 DBG 3518:                      
20211221-145005 DBG 3518:                      /dev/mmcblk0p1 : start=        8192, size=      524288, type=c
20211221-145005 DBG 3518:                      /dev/mmcblk0p2 : start=      532480, size=   124510208, type=83
```
you created the backup with a system which has a different sfdisk version than the system you use to restore the backup.

`>>> line 5: unsupported command`

If you backup the original sfdisk file and delete line 5 (sector-size: 512) you should be able to restore the backup. Please use an OS you created the backup with to restore the backup to have no issues with the restore :wink:

madbrain76, 2021-12-22:
Thanks. Removing line 5 from pi64-backup.sfdisk indeed fixed it.

I'm usually not using the same OS to restore the card as the one used to backup. Typically, I backup to NAS using a Raspberry Pi, and restore to the card from the NAS, not using the Pi. I did not know that the backup format used by raspiBackup was OS-dependent.

The version of sfdisk on the backup system is :
pi@pi64:~ $ sfdisk --version
sfdisk from util-linux 2.36.1

The version of sfdisk on the restore system is :
root@server10g:/array/zfs/Backups/pi64/pi64-tar-backup-20211220-203939# sfdisk --version
sfdisk from util-linux 2.34

The man page for sfdisk doesn't seem to define the format of the dump file, unfortunately. I'm not sure if it's defined elsewhere. There is no mention of compatibility between versions. Clearly there is no forward compatibility in this case. But it's not mentioned whether there is backwards compatibility. Unfortunately, the dump file itself doesn't include any versioning.

This would make it hard for you to implement a solution to this problem that doesn't involve manually editing the sfdisk file for every restore.

One suggestion I would make is to add some versioning in raspiBackup. Save the version of sfdisk used for the backup somewhere. And when restoring the backup, compare it. Proceed if it's the same. If it's not the same, print a warning. If backwards compatibility exists in sfdisk, maybe print that warning only if the version of sfdisk on the restore system is lower than on the backup system, as was the case for me.

framps, 2021-12-22:
> One suggestion I would make is to add some versioning in raspiBackup.

I see your point. But raspiBackup was designed to have no metadata reporsitory. 

> Typically, I backup to NAS using a Raspberry Pi, and restore to the card from the NAS, not using the Pi.

That's exactly the way I do it. I also already had to restore a Bullseye image and faced exactly the same issue. I keep raspiBackup backward compatible whenever I add new function or change code. But I don't add any compatibility code for the Linux tools. I cannot test all the different configurations and dependencies to OS releases.

One possible workaround for you would be to create a wrapper script which will delete line 5 and then start the restore.

madbrain76, 2022-01-01:
I wish there was a more transparent automatic solution. If you aren't going to fix this issue, just close it as wontfix.

closed 2022-01-15 by github-actions[bot]