framps/raspiBackup #420: Add a restore completion extension point

framps, 2021-12-22:
Would be nice to have an extension point which is called when the restore finishes.

madbrain76, 2021-12-22:
Thanks. This could be useful to enable/disable autosuspend during restores on my NAS.

framps, 2021-12-22:
This also can be used to do any postprocessing. For preprocessing another extension point will be added.

framps, 2022-01-22:
Just created branch https://github.com/framps/raspiBackup/tree/dev_restoreExtensions to add the extensionpoints

framps, 2022-01-28:
Implemented in https://github.com/framps/raspiBackup/commit/283fd6a0c2061009c10c48ebe56fabd4f08dd1bb

framps, 2022-01-28:
New config option `	DEFAULT_RESTORE_EXTENSIONS=""` similar to `DEFAULT_EXTENSIONS=""` is now available to define extensions executed pre and post restore operation.

closed 2022-01-28 by framps