framps/raspiBackup #424: Backup LW hängt sich aus...

Steinspiel, 2021-12-27:
Hallo,

RaspiBackup lief vor kurzem schon einmal auf meinem Pi4, an der Hardware hat sich nichts geändert und die Konfiguration habe ich auch bei den Standartwerten beibehalten:

-openHABian neu installiert, 
-dort einen Account eingerichtet, 
-raspiBackup installiert und wie vor Konfiguriert,
-externer USB Platte EXT4 formatiert,

Die Simulation mit 

`sudo raspiBackup -F`

lief erfolgreich durch... (s. Logfile)

Rufe ich raspiBackup von Hand auf bekomme ich die Fehlermeldungen "RC 1", soweit ich weiß ein Lese- oder Schreibfehler bzw. wird "Read-only file system" gemeldet. Erst dachte ich es handelt sich um die falsche Rechtevergabe für _/backup_. Aber 

`ls -l /backup`

zeigt meines Wissens Lese-,Schreib-,Ausführen Rechte an...

Dann habe ich festgestellt das **nach** der raspiBackup Fehlermeldung das ext. USB Laufwerk **nicht** nur nicht mehr in _/backup_ gemountet ist, sondern dann als "sdc1" eingebunden und nicht wie **vor** dem Backup als "sdb1"
So wie ich das verstehe wurde das LW während des Backups ausgehängt und dadurch treten auch die Lese- oder Schreibfehler. Nur wieso?

Anbei ein Screenshot des Terminals während des Backups und je ein Logfile von der raspiBackup Simulation und dem tatsächlichen Aufruf.

Falls ich irgendeine Info vergessen habe...
Danke fürs lesen und für die Hilfe natürlich 

Steinspiel
[raspiBackup.log](https://github.com/framps/raspiBackup/files/7781135/raspiBackup.log)
[simulation.log](https://github.com/framps/raspiBackup/files/7781137/simulation.log)
[image]

framps, 2021-12-27:
Im Debuglog finde ich

```
--- RBK0085I: Backuperstellung vom Typ rsync gestartet. Bitte Geduld.
20211227-160922 DBG 2273:                  --> executeCmd Command: rsync --exclude="/backup" 			--exclude="/home/@USER@/raspiBackup.log" 			--exclude="/home/@USER@/raspiBackup.msg" 			--exclude='.gvfs' 			--exclude=/proc/* 			--exclude=/lost+found/* 			--exclude=/sys/* 			--exclude=/dev/* 			--exclude=/boot/* 			--exclude=/tmp/* 			--exclude=/run/* 			--exclude=/@HOSTNAME@-backup.img --exclude=/@HOSTNAME@-backup.tmg --exclude=/@HOSTNAME@-backup.sfdisk --exclude=/@HOSTNAME@-backup.mbr --exclude=/raspiBackup.log --exclude=/raspiBackup.msg 			 			 			--numeric-ids 			-aHAx --delete 			 			 			/ 			"/backup/@HOSTNAME@/@HOSTNAME@-rsync-backup-20211227-160849" 			
20211227-160922 DBG 2274:                      --- Redirect: & - Skips: 
rsync: [receiver] write failed on "/backup/@HOSTNAME@/@HOSTNAME@-rsync-backup-20211227-160849/var/lib/apt/lists/raspbian.raspberrypi.org_raspbian_dists_bullseye_main_binary-armhf_Packages": Read-only file system (30)
rsync error: error in file IO (code 11) at receiver.c(378) [receiver=3.2.3]
rsync: [sender] write error: Broken pipe (32)
rsync error: error in socket IO (code 10) at io.c(823) [sender=3.2.3]
rsync: [generator] write error: Broken pipe (32)
20211227-161115 DBG 2293:                  <-- executeCmd 10
20211227-161115 DBG 4969:              <-- rsyncBackup 10
20211227-161115 DBG 3591:              --> colorAnnotation 1
20211227-161115 DBG 3615:              <-- colorAnnotation 
??? RBK0021E: Backupprogramm des Typs rsync beendete sich mit RC 10.
```

D.h. es war/dev/sdb1 als /backup gemounted und rsync wurde gestartet. Dann wird nichts umounted oder remounted. Der RC 10 bedeutet gemaess `man rsync`

```
Error in socket I/O
```

Den Fehler von rsync habe ich bislang noch nicht kennengelernt. Dass ein Filesystem auf RO gemounted wird kenne ich wenn Lese/Schreibprobleme auftreten. Das kann man in der fstab definieren wie im Fehlerfalle zu verfahren ist. 

So wie ich es sehe tritt irgendwie ein Lese/Schreibfehler auf waehrend gesichert wird. Es sieht mir so aus dass die Stromversorgung der Raspi oder der USB Platte ungenuegend ist und unter Last zusammenbricht. Sieh mal in /var/log/syslog nach zum Zeitpunkt des Backups.

Steinspiel, 2021-12-27:
Moin,

Gleich folgt ne kleine Lobeshymne, ;-) denn Du scheinst Dich wirklich sehr gut aus zu kennen! 

In /var/log/syslog findet sich zum fraglichen Zeitpunkt:
```
Dec 27 16:11:15 openhabian kernel: [ 1506.694305] usb 2-1: USB disconnect, device number 2
Dec 27 16:11:15 openhabian kernel: [ 1506.734416] EXT4-fs (sdb1): This should not happen!! Data wil
```
usw. usf.

Also es **gibt** ein Abbruch und der Grund den Du vermutest scheint auch hin zu kommen: Habe das USB LW nun **nicht** direkt an den Raspi angeschlossen sondern an nen aktiven HUB mit eigener Stromversorgung und was soll ich sagen? Das Backup lief  mit `Returncode 0` durch! :-)

Ich werde das morgen noch ein paar mal probieren, dann auch das automatische Backup... und mich noch einmal melden 

Auf jeden Fall erst einmal vielen Dank für Deine Hilfe! 
Und ich muss ja zugeben, durch die "Problemchen" lernt man raspiBackup und sogar Linux etwas besser kennen als wie es im Fehlerfreien Betrieb für mich der Fall wäre  ;-)

bis dann,

framps, 2021-12-28:
Die Stromversorgung der Raspi ist immer wieder in allen möglichen Situationen die Problemursache. Und wenn die Platte sich beim Betrieb abhängt ist das zu 99% die Ursache 😉

closed 2022-01-12 by github-actions[bot]