framps/raspiBackup #425: Am I supposed to be able to run custom commands with the -o argument during backups ?

madbrain76, 2022-01-01:
I tried to use the -o argument to run a script that wakes up my NAS prior to backing up. The script calls etherwake, then sleeps for a bit, then mounts the /backup directory. Unfortunately, it appears none of the commands I pass through the "-o" argument are executed, even a simple "touch /tmp/wokeupnas" does nothing.

pi@genie:~ $ sudo raspiBackup.sh -o "touch /tmp/wokeupnas"
--- RBK0009I: genie: raspiBackup.sh V0.6.6.1 (711a501) started at Fri 31 Dec 2021 07:20:58 PM PST.
mkdir: cannot create directory ‘/backup’: No such device
??? RBK0172E: Unable to create directory /backup/genie/genie-tar-backup-20211231-192056.
--- RBK0033I: Please wait until cleanup has finished.
??? RBK0005E: Backup failed. Check previous error messages for details.
--- RBK0010I: genie: raspiBackup.sh V0.6.6.1 (711a501) stopped at Fri 31 Dec 2021 07:21:20 PM PST with rc 119.
--- RBK0026I: Debug logfile saved in /home/pi/raspiBackup.log.
pi@genie:~ $ ls -al /tmp/wokeupnas
ls: cannot access '/tmp/wokeupnas': No such file or directory

Version info:
* Provide the output of `sudo raspiBackup.sh --version` or `sudo raspiBackupInstallUI.sh --version`
pi@genie:~ $ sudo raspiBackup.sh --version
Version: 0.6.6.1 CommitSHA: 711a501 CommitDate: 2021-12-08 CommitTime: 21:40:00
pi@genie:~ $ sudo raspiBackupInstallUI.sh --version
/usr/local/bin/raspiBackupInstallUI.sh: illegal option -- -
raspiBackupInstallUI.sh 0.4.3.6, 2021-11-12/23:11:19 - 91326c6

* Provide the output of `cat /etc/os-release`

pi@genie:~ $ cat /etc/os-release 
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
[VERSION_ID="10"](url)
[raspiBackup.log](https://github.com/framps/raspiBackup/files/7797348/raspiBackup.log)

VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

framps, 2022-01-01:
You cannot use option -o to mount /backup. Check and adapt your needs https://github.com/framps/raspiBackup/blob/master/helper/raspiBackupNfsWrapper.sh which mounts an nfs drive.

madbrain76, 2022-01-02:
OK. I wish it was possible to it this way. I have an existing script to wake up the NAS, which I sometimes call manually. It would be easier to invoke it using the -o option, or another raspiBackup command-line option, rather than having to write a wrapper script.
Just my 2 cents.

framps, 2022-01-02:
I see your point. 

Options `-o` and``-a` are supposed to start and stop services only but you can use this also to  mount `/backup`. I will check whether I can move the check for a mounted /backup behind the option `-o` calls in next release.. 

But you also can add an option in your script which controls the mount of `/backup`.

closed 2022-01-19 by github-actions[bot]