framps/raspiBackup #426: Have mounted a 2.5 disk on pi on /backup, but trying backup gives me:

pjdw, 2022-01-03:
Please execute all following actions first and provide all requested information:

* Check whether my error messages were listed on the messages page and all suggested actions don't fixes my issue. [English](https://linux-tips-and-tricks.de/en/rmessages) - [German](https://linux-tips-and-tricks.de/de/fehlermeldungen)
* Attach raspiBackup debug log. Just drag and drop raspiBackup.log into the issue and don't copy and paste the contents of raspiBackup.log into the issue.
* Issues are closed automatically after 2 weeks if there is no activity any more

pjdw, 2022-01-03:
`sudo bash ./raspiBackup.sh
--- RBK0009I: pimk: raspiBackup.sh V0.6.6.1 (711a501) started at ma  3 jan 2022 13:36:54 CET.
??? RBK0134E: Unable to use hardlinks on /backup for bootpartition files. RC %2.
--- RBK0033I: Please wait until cleanup has finished.
--- RBK0043I: Removing incomplete backup in /backup/pimk/pimk-rsync-backup-20220103-133652. This may take some time. Please be patient.
??? RBK0005E: Backup failed. Check previous error messages for details.
--- RBK0010I: pimk: raspiBackup.sh V0.6.6.1 (711a501) stopped at ma  3 jan 2022 13:36:57 CET with rc 102.
--- RBK0026I: Debug logfile saved in /home/pi/raspiBackup.log.`

framps, 2022-01-03:
Where is the debug log?

pjdw, 2022-01-03:
> Where is the debug log?

Could the cause be that target medium is formatted as exfat? Some where i saw that disk has to be formatted as ext?

framps, 2022-01-03:
Yes. Exfat does not support hardline. Use ext3 or even.better ext4

pjdw, 2022-01-03:
> Yes. Exfat does not support hardlinks Use ext3 or even.better ext4

pjdw, 2022-01-03:
Thanks! If you would like to keep things clean. Maybe delete the question.

closed 2022-01-03 by pjdw

framps, 2022-01-03:
> Maybe delete the question

This issue may be helpful for anybody who faces the same error message. I updated the issue title a little bit :wink: