framps/raspiBackup #430: Restoring TAR Backup not possible with one partition on armbian/ odroid n2

Peppa123, 2022-01-16:
Hi there, I would like to use raspiBackup on my armbian installations on some odroid devices. Corresponding to RBK0142E raspiBackup is for a longer time aware of it that there is only one partition on the sd card with the armbian installation.

The backup is possible and working and creating one tar file in the backup directory. But the restore is not possible and raspiBackup is searching for the information about the boot partition files:

--- RBK0009I: c4-db-omv5: raspiBackup V0.6.6.1 (711a501) Sun Jan 16 11:33:12 CET 2022 gestartet.
??? RBK0061E: Keine Bootpartitionsdateien in /srv/dev-disk-by-uuid-106c94c5-c9b3-4985-a41d-639a2fc149fe/Backup-C4/c4-db-omv5/c4-db-omv5-tar-backup-20220116-110135 gefunden die mit c4-db-omv5-backup beginnen.
??? RBK0061E: Keine Bootpartitionsdateien in /srv/dev-disk-by-uuid-106c94c5-c9b3-4985-a41d-639a2fc149fe/Backup-C4/c4-db-omv5 gefunden die mit c4-db-omv5-backup-20220116-110135 beginnen.
??? RBK0061E: Keine Bootpartitionsdateien in /srv/dev-disk-by-uuid-106c94c5-c9b3-4985-a41d-639a2fc149fe/Backup-C4/c4-db-omv5 gefunden die mit c4-db-omv5-backup beginnen.
--- RBK0033I: Bitte warten bis aufgeräumt wurde.
??? RBK0077E: Restore wurde fehlerhaft beendet. Siehe vorhergehende Fehlermeldungen.
--- RBK0010I: c4-db-omv5: raspiBackup V0.6.6.1 (711a501) Sun Jan 16 11:33:15 CET 2022 beendet mit Returncode 102.
--- RBK0026I: Debug Logdatei wurde in /root/raspiBackup.log gesichert.

Am I doing something wrong or is there still something to do after the older case.

Peppa123, 2022-01-16:
I have also one log file for restore but would like to send it via pm.

framps, 2022-01-16:
Did you check [this page](https://www.linux-tips-and-tricks.de/en/rmessages/#m0051) ? Looks like the backup run didn't create required files in the backup directory. 

Please note you're running raspiBackup in an unsupported environment :smirk:  . But I will check the debug log and try to get raspiBackup to succeed in your environment as long as the effort is accceptable :wink: 

Please provide the debug log of a backup run. If you don't want to attach it here just send it to my eMail address (See my [contact page](https://www.linux-tips-and-tricks.de/en/contact/))

Peppa123, 2022-01-16:
Yes but after reading the old issue RBK0142E I give it a new try to use raspiBackup for the IOT devices to keep backup simple and easy to use. Log file is underway.

framps, 2022-01-16:
> Log file is underway.

You sent the restore debug log to me. But I need the debug log of a backup run :wink:

framps, 2022-01-16:
Thank you very much for the backup debug log. There I found
```
20220116-110137 DBG 5953:                  *** ls -1 /dev/mmcblk*
20220116-110137 DBG 5953:                      /dev/mmcblk0
20220116-110137 DBG 5953:                      /dev/mmcblk0boot0
20220116-110137 DBG 5953:                      /dev/mmcblk0boot1
20220116-110137 DBG 5953:                      /dev/mmcblk0p1
20220116-110137 DBG 5953:                      /dev/mmcblk0rpmb
```
and
```
 /dev/mmcblk0p1 on / type ext4 (rw,noatime,nodiratime,errors=remount-ro,commit=600)
```
which is unusual for RaspberryOS. There exists all the time /boot either mounted from /dev/mmcblk0p1 or it's located on the root partition..

Even if /boot and /root exist on the same partition there has to exist /boot which is not true for your environment. Therefore no /boot partition is saved and not found by the restore. 

You can try what happens if you disable the search for the boot files when restoring the backup:

In line 7213 or raspiBackup there is following code

```
	if (( ! $PARTITIONBASED_BACKUP	 )); then
		findNonpartitionBackupBootAndRootpartitionFiles
	fi
```

Just insert `#` at the beginning of all three lines and try what happens when you restore the backup. 

Frankly I expect the restore still to fail. But give it a try.

Other alternatives which may work for you may be to use the partitionoriented backup mode or to use the dd backup type.

framps, 2022-01-16:
I just had another idea: Create an empty /boot directory on your root partition. Then an empty boot partition backup will be created an the restore will find a /boot partition backup and not fail.

Peppa123, 2022-01-24:
I tried the partitionoriented backup for now. The backup is created and the restore to sd card is possible but the os is not booting after inserting the sd card. The partitions seems to be identical but I am not sure if there is still something different on boot sector etc. so that the boot is failing.

Since now I was using the dd backup but it is quit space consuming and as you wrote yourself on kb possible filesystems errors are backup up and I want to change to tar backup.

Do you have some idea were to look to get the restore booting and find the missing detail in backup?

Peppa123, 2022-01-24:
One remark - there is a /boot directory existing on armbian os. Here are the details on the only patition with ext

-rw-r--r--  1 root root  5087370 Nov 22 13:00 System.map-4.9.277-meson64
-rw-r--r--  1 root root      180 Jan 24 09:17 armbianEnv.txt
-rw-r--r--  1 root root     1536 Mar 21  2020 armbian_first_run.txt.template
-rw-r--r--  1 root root     4882 Mar 21  2020 boot-desktop.png
-rw-r--r--  1 root root    38518 Mar 21  2020 boot.bmp
-rw-r--r--  1 root root     7823 Jan 24 09:33 boot.cmd
-rw-r--r--  1 root root     5579 Jan 24 09:33 boot.ini
-rw-r--r--  1 root root     7895 Jan 24 09:33 boot.scr
-rw-r--r--  1 root root   170684 Nov 22 13:00 config-4.9.277-meson64
-rwxr-xr-x  1 root root      120 Jan  1  1970 display.bin
lrwxrwxrwx  1 root root       19 Nov 30 11:08 dtb -> dtb-4.9.277-meson64
drwxr-xr-x  2 root root     4096 Aug 23  2020 dtb-4.9.219-meson64
drwxr-xr-x  2 root root     4096 Sep 21  2020 dtb-4.9.232-meson64
drwxr-xr-x  2 root root     4096 Oct 27  2020 dtb-4.9.236-meson64
drwxr-xr-x  2 root root     4096 Feb  7  2021 dtb-4.9.247-meson64
drwxr-xr-x  2 root root     4096 Mar 10  2021 dtb-4.9.250-meson64
drwxr-xr-x  2 root root     4096 May 12  2021 dtb-4.9.260-meson64
drwxr-xr-x  3 root root     4096 Nov 30 11:08 dtb-4.9.277-meson64
lrwxrwxrwx  1 root root       19 Aug 31 20:49 dtb.old -> dtb-4.9.275-meson64
-rwxr-xr-x  1 root root      256 Jan  1  1970 edid.bin
-rw-r--r--  1 root root 12140425 Jan 24 09:34 initrd.img-4.9.277-meson64
lrwxrwxrwx  1 root root       23 Jan 24 09:34 uInitrd -> uInitrd-4.9.277-meson64
-rw-r--r--  1 root root 12140489 Jan 24 09:34 uInitrd-4.9.277-meson64
-rw-r--r--  1 root root  9945413 Nov 22 13:00 vmlinuz-4.9.277-meson64
lrwxrwxrwx  1 root root       23 Nov 30 11:08 zImage -> vmlinuz-4.9.277-meson64

Peppa123, 2022-01-24:
Aaahh I see that the "/boot" directory is empty after restoring using partition based backup. It seems that actually it can't be backuped up by raspianbackup. Is this correct?

framps, 2022-01-24:
> Do you have some idea were to look to get the restore booting and find the missing detail in backup?

I'm kind of surprised because a partition oriented backup should just save and restore all partitions. I checked the debug log you sent to me again and which partitions are used and detected

```
20220116-110137 DBG 5953:                      /dev/mmcblk0
20220116-110137 DBG 5953:                      /dev/mmcblk0boot0
20220116-110137 DBG 5953:                      /dev/mmcblk0boot1
20220116-110137 DBG 5953:                      /dev/mmcblk0p1
20220116-110137 DBG 5953:                      /dev/mmcblk0rpmb
```
I have no idea what mmcblk0boot0 and mmcblk0rpmb are used for. They don't exist on RaspbianOS and may cause the issue. To verify this I need the debug logs of the partition oriented backup and restore.

> Since now I was using the dd backup but it is quit space consuming

You can reduce the dd image size (See [this page](https://www.linux-tips-and-tricks.de/en/faq/#a16) for details. 

> Aaahh I see that the "/boot" directory is empty after restoring using partition based backup. It seems that actually it can't be backuped up by raspianbackup. Is this correct?

/boot should not be empty with partition oriented backup. In the debug log I find
```
20220116-110137 DBG 6018:              --- Bootpartition is located on rootpartition /dev/mmcblk0p1
```
and thus /boot should exist in the backup on partition mmcblk0p1. Please check. If it's missing in the restore image please provide the debug log of the restore.

I also noticed in the log

```
20220116-110138 DBG 5034:                          /dev/mmcblk0p1 on /var/folder2ram/var/log type ext4 (rw,noatime,nodiratime,errors=remount-ro,commit=600)
20220116-110138 DBG 5034:                          folder2ram on /var/log type tmpfs (rw,relatime)
20220116-110138 DBG 5034:                          /dev/mmcblk0p1 on /var/folder2ram/var/tmp type ext4 (rw,noatime,nodiratime,errors=remount-ro,commit=600)
20220116-110138 DBG 5034:                          folder2ram on /var/tmp type tmpfs (rw,relatime)
```
which also may cause the issue.

In addition you use Armbian

```
20220116-110136 DBG 1862:          *** cat /etc/os-release
20220116-110136 DBG 1862:              PRETTY_NAME="Armbian 21.08.6 Buster"
20220116-110136 DBG 1862:              NAME="Debian GNU/Linux"
20220116-110136 DBG 1862:              VERSION_ID="10"
20220116-110136 DBG 1862:              VERSION="10 (buster)"
20220116-110136 DBG 1862:              VERSION_CODENAME=buster
20220116-110136 DBG 1862:              ID=debian
20220116-110136 DBG 1862:              HOME_URL="https://www.debian.org/"
20220116-110136 DBG 1862:              SUPPORT_URL="https://www.debian.org/support"
20220116-110136 DBG 1862:              BUG_REPORT_URL="https://bugs.debian.org/"
```
which also may cause the issue.

raspiBackup is tested and supported for a Raspberry running RaspianOS only. You can try raspiBackup and it works indeed on other OSes and HW but frankly I think your environment will not work.

If you send me the debug logs of the backup and restore of the partition oriented backup I will check the logs whether I find the root cause of the missing /boot directory. But if I don't find anything I have to close the issue because your environment is not supported. :cry:

Peppa123, 2022-01-25:
I send you the logs tomorrow via email. Thankx in advance.

Peppa123, 2022-02-09:
Did you find time to have a look at the log files?

framps, 2022-02-09:
I'm sorry. I didn't get any log files from you

closed 2022-02-11 by framps