framps/raspiBackup #434: smart recycle issue/question

upD8R, 2022-01-22:
Hi there,

new RPi here, I installed and configured your script in the same way as I did for my other Pis. This is about smart recycle strategy.
[raspiBackup.log](https://github.com/framps/raspiBackup/files/7919398/raspiBackup.log)

After initial setup I created yesterday a dd backup by executing raspiBackupWrapper.sh on the console (only pisjrink used in the wrapper script). Everything fine.
Today I wanted to create another backup after I installed additonal software, this was also successfull (see log) but I realized an issue on the console as pishrink was throwing an error.
Obviously this last backu was immeditealy smart recycled, which is a funny behavior.

So I do have a few questions/comments:

1. Why did it delete the last one and not the previous? Is this becaue **0** 4 12 1? Should it be 1 4 12 1 instead?
2. I raised that before, pishrink is so cool but sending the success mail before it is executed is suboptimal. I would have never realized that this backup actually failed.

Many thanks!

framps, 2022-01-22:
> 1. Why did it delete the last one and not the previous? Is this because **0** 4 12 1? Should it be 1 4 12 1 instead?

Yes. You should keep at least 1 daily backup. I will add some code to reject a daily backup count of less than 1 in next release.

> 2\. I raised that before, pishrink is so cool but sending the success mail before it is executed is suboptimal. I would have never realized that this backup actually failed.

raspiBackupWrapper is a sample script and not part of the official raspiBackup releases.  If you find an issue or have some code enhancement of the script please open a PR which I will merge.

upD8R, 2022-01-22:
Thanks indeed.
Regarding wrapper the script, I'm actually not a coder. But I see your point.

BTW, there's also the memory consumption before/after empty on that box.
This is casued by your grep:
free -m |grep -i 'Mem:'
In my Bullseye setup here this "free" gives  its results in Germany, so I changed that to Speicher:

framps, 2022-01-22:
> BTW, there's also the memory consumption before/after empty on that box.

Good catch :+1: Actually the extensions are also just samples and you're free to update the code. If you have an idea how to change the code to become language insensitive please either just create a PR or let me know and I'll update the code accordingly.

framps, 2022-02-04:
@upD8R I did some investidations and finally found out
```
function getMemoryFree() {
	eval set -- "$(LC_ALL=C free -m |grep -i 'Mem:')"
	echo "$3 $4"
}
```

should do the trick :wink:  Would be great if you can confirm the fix works for you.

upD8R, 2022-02-05:
Changed it, I'll let you know!

closed 2022-02-08 by upD8R