framps/raspiBackup #437: raspiBackup.msg not being created on new install?

g6nhu, 2022-01-28:
Hi,
I've got raspiBackup running successfully on a number of Pis, each time the backup runs, it creates three files, the backup file itself, a .log and a .msg file.

I've recently installed it to a new Pi and configured it (as far as I can tell) exactly the same as the previous ones.

When I run a backup, I don't get the summary .msg file.

I'm using raspiBackup.sh 0.6.6.1.

Have I missed something in the configuration?

Thanks.

framps, 2022-01-28:
Looks like you use the latest raspiBackup release on your new pi now. The msg file is only needed to send the messages via eMail or Telegram and is now deleted when raspiBackup terminates.

g6nhu, 2022-01-28:
Thanks for the very quick reply.

It's a really useful summary, which saves looking at the entire log.  Is there any way you'd consider leaving it there and not deleting it upon termination?

framps, 2022-01-28:
Interesting you find it useful ... It's just redundant information because as you noticed the messages are available in the debug log. Just use `grep RBK raspiBackup.log` and you will get the messages only :wink:

g6nhu, 2022-01-28:
It's just the convenience, I check my backups on my NAS from the GUI on a Mac and double clicking the .msg file takes a fraction of a second and all the info I need is there.  I could log onto the NAS, find the appropriate directory for the backup (I rotate six full backups and now backup thirteen Pis) and grep the file but realistically, I probably won't bother due to the time taken.  I'm just surprised that a very useful summary has been removed.

[image]

framps, 2022-01-28:
I see your point. Actually I thought it's useless and didn't expect somebody uses the message file. 

I'll keep the message file in next release again. A temporary workaround for you may be to delete [this line](https://github.com/framps/raspiBackup/blob/4911c95b2d177dad5ae911628adc488b396b663c/raspiBackup.sh#L2021) in raspiBackup.sh.

g6nhu, 2022-01-28:
Thank you so much for doing that, I really do appreciate it.  In the meantime, I've done what you suggested and removed that line.

g6nhu, 2022-01-29:
I removed that line and ran a backup this evening, the .msg file was still deleted.  Not sure if it's somewhere else in the script as well.

framps, 2022-01-30:
Hm ... I just had a quick look into the code and didn't do any tests. You have to wait until #438 is implemented and tested.

Sorry for the inconvenience.

framps, 2022-01-31:
@g6nhu I just had some spare time and created a fix. Would you please grab raspiBackup.sh from https://raw.githubusercontent.com/framps/raspiBackup/master_418/raspiBackup.sh and test again?

g6nhu, 2022-01-31:
@framps That has worked, thank you.

framps, 2022-01-31:
Great. Thank you for your test. I'll merge the fix into next release. 

I have no idea when this will be published. May be somewhere in summer this year. 

Just keep the current hotfix in the meantime and use `sudo raspiBackup.sh -U -S` if the next release is available.

closed 2022-02-03 by framps