framps/raspiBackup #440: RBK0021E: Backupprogram for type rsync failed with RC 23

WimBre, 2022-02-04:
Hi Framps,

I'm getting a "RBK0021E: Backupprogram for type rsync failed with RC 23." error.

I tried to open a page of the FAQ where I could possibly find more information on solving this problem but get there (https://www.linux-tips-and-tricks.de/en/faq/index. php#m0021) an error message.
Attached you will find my raspiBackup debug logfile. 

My environment:
Raspberry PI3 with Raspbian Jessie (kernel Linux raspi 4.9.35-v7+), USB boot device
raspibackup version raspiBackup 0.6.6.1
Backup on Sandisck Ultra USB3,0 256 GB (ext4) attached to Raspberry Pi
All devices attached via LAN

Regards,
Wim
[raspiBackup.log](https://github.com/framps/raspiBackup/files/8001982/raspiBackup.log)

framps, 2022-02-04:
```
rsync: mknod "/media/usb/@HOSTNAME@/@HOSTNAME@-rsync-backup-20220203-184747/var/lib/docker/volumes/backingFsBlockDev" failed: Operation not permitted (1)
rsync error: some files/attrs were not transferred (see previous errors) (code 23) at main.c(1207) [sender=3.1.3]
```
Looks like there is some authorization missing.

Please also check https://www.linux-tips-and-tricks.de/en/rmessages/#m0021

framps, 2022-02-04:
> get there (https://www.linux-tips-and-tricks.de/en/faq/index.php#m0021) an error message.

I have to fix this. Where did you find this link?

WimBre, 2022-02-04:
I went to this page: https://www.linux-tips-and-tricks.de/en/faq/index.php?option=com_content&view=article&id=515:raspibackup-messages&catid=38&lang=en-US

and then tried to open this link but this one failed.
[RBK0197W: eMail send command %s failed with RC %s.](https://www.linux-tips-and-tricks.de/en/faq/index.php?option=com_content&view=article&id=515:raspibackup-messages&catid=38&lang=en-US#m0197)

framps, 2022-02-04:
> I went to this page

All links on this page are dead :cry: . Where did you find the link? The correct link to FAQ is https://www.linux-tips-and-tricks.de/en/faq/ so I'm wondering where this ugly url comes from :thinking:

WimBre, 2022-02-04:
I tried to reproduce my search:

Home (https://www.linux-tips-and-tricks.de/en/)

-> choose -> raspiBackup - FAQ (https://www.linux-tips-and-tricks.de/en/faq)

-> choose -> [32) After upgrade to v0.6.3.2 I get errormessage RBK0021E: Backupprogram of type tar terminated with RC 1 or RBK0021E: Backupprogram of type rsync terminated with RC 23.h](https://www.linux-tips-and-tricks.de/en/faq/#a32)
raspiBackup - FAQ (https://www.linux-tips-and-tricks.de/en/faq)

- > 40) How can I update my local configuration with new options introduced in a new version after upgrade?

In general there are new options introduced in a new version. It's possible to add them manually. But it's much more convenient to use a helper script which creates a new configuration file with all the latest options and merges your local options. Merge conflicts have to be solved manually. You find details for the script on **[this page]**(https://www.linux-tips-and-tricks.de/en/configuration-update).

The last BOLD link (this page) led me to the " dead page".

framps, 2022-02-04:
> The last BOLD link (this page) led me to the " dead page".

Fixed now. Thank you very much for your detailed description :+1:

WimBre, 2022-02-05:
> ```
> rsync: mknod "/media/usb/@HOSTNAME@/@HOSTNAME@-rsync-backup-20220203-184747/var/lib/docker/volumes/backingFsBlockDev" failed: Operation not permitted (1)
> rsync error: some files/attrs were not [...]

Any idea if these authorization issues are related to the target (USB stick) or to certain directories on my raspberry?

framps, 2022-02-05:
> Any idea

Hm ... it's a rsync issue and no raspiBackup issue :wink: .

First of all  I would check the syslog for disk errors which happened during the backup. Next I would check the USB stick for errors (smartctl). I assume it's the only file which causes RC23. Next I would try to exclude the file from the backup to check whether RC23 is specific for this file. Next I would read [FAQ47](https://www.linux-tips-and-tricks.de/en/faq/#a47) :wink:

framps, 2022-02-10:
In the debuglog you shouldn't see any colors. It's just a plain text file. raspiBackup adds some VT100 coloring codes for error and warning messages written on the console but no coloring codes are written into the debug log. So I have no idea where this red bar comes from.

I also notice there is `@user@` and `@hostname@` used to masquerade sensitive information in the log. But raspiBackup replaces these strings in upper case `@USER@` and `@HOSTNAME@`. Any idea why it's in lower case in your debug log?

framps, 2022-02-10:
I just checked the log in https://github.com/framps/raspiBackup/issues/440#issue-1124051925 which has everything in uppercase :thinking:

WimBre, 2022-02-10:
should I delete the current log file and make a new backup to see what is in the newly created log file?

framps, 2022-02-10:
This will neither not change the root cause of your issue and help to locate the root cause.

closed 2022-02-25 by github-actions[bot]