framps/raspiBackup #442: Problem mit NOOBS

hajoko1, 2022-02-09:
Hi, habe 3 Raspis laufen und bin dabei, auf allen RaspiBackup zu installieren. Auf zweien davon läufts einwandfrei, bei einem ist auch schon der restore getestet. Nummer 3, mein frühest installierter, ist leider mit NOOBS eingerichtet worden (da wusste ich es nicht besser und habe mich auf Anleitungen verlassen), die anderen beiden sind "normal" installiert. Bei allen ist ist als Sicherungsmethode tar eingestellt.

Wähle ich jetzt in raspiBackup den Punkt "mehrere Patitionen sichern", kreuze alle an, so kommt das Auswahlfenster immer und immer wieder, die Auswahl wird nicht akzeptiert. Erneuter Aufruf von raspiBackupInstallUI zeigt, dass die Auswahl wieder auf den Standard-Partitionen steht. 

Ich traue mich auch nicht dran, den raspi1 zu "entnoobsen", Anleitungen gibts ja. Ich betreibe hier ein Raid-5 mit 6 TB, mit allen meinen Daten. Eine Störung dieses Systems wäre tödlich für mich.

Gibts irgendeine Lösung für mein Problem? Was mache ich falsch? Danke für Hinweise. Gruß, Hajo.

framps, 2022-02-09:
Eigentlich steht da bei Erstellen eines Issues dass Du die aktuelle Versionsnummer von dem Installer angeben sollst :wink: 

Anyhow - installiere mal den aktuellen Installer und teste noch mal. Wenn es dann immer noch nicht funktioniert kannst Du das auch ersatzweise in /usr/local/etc/raspiBackup.conf konfigurieren. Ich habe - aus Gruenden die Du schon genannt hast kein NOOBS um Dein Problem nachzuvollziehen. Mit 2 Partitionen funktioniert es bei mir in den P Modus umzuschalten.

Du musst darin die folgenden Zeilen definieren. Bei der zweiten Zeile musst Du alle Partitionsnummern angeben. Ich habe mal ein paar Nummern beispielhaft eingegeben. Das musst Du natuerlich entsprechend bei Dir anpassen.
```
# Options for partitionoriented mode
DEFAULT_PARTITIONBASED_BACKUP="1"
DEFAULT_PARTITIONS_TO_BACKUP="1 2 4 5 7 8"
```

hajoko1, 2022-02-10:
Moin. Danke für die schnelle Antwort. Ich hatte es bereits direkt über die Kommandozeile mit `-P -T "*"` versucht. Verzeichnis im Backuppfad wird angelegt, ebenso die Files .msg und .log. Das wars, weiter passiert auch lange nichts. Habe dann nach deiner Antwort den Config-File abgeändert. Gleiches Ergebnis. Ich würde dann das Ganze gern nochmal mit den Partitionsnummern anstelle "*" versuchen. Wie finde ich die Nummern der Partitionen?

Übrigens: sehr schönes Programm, danke! Erspart die Fummelei am Raspi für eine Karten-Kopie - Backup-Platte ab, da alle USB-Ports belegt sind, Kabelgewirr - und läuft vor allem täglich!

Vielleicht sollte ich mich doch mal ans entnoobsen machen.

Nachtrag: raspiBackup ist Version 0.6.6, Config-Tool 0.4.3.5. Sorry...  :-(
(alles erst diese Woche erstmals installiert)

framps, 2022-02-10:
> Wie finde ich die Nummern der Partitionen?

`blkid`

> Vielleicht sollte ich mich doch mal ans entnoobsen machen.

Sicherlich keine schlechte Idee :wink: 

Dass der Installer die Partitionen nicht annimmt liegt daran dass beim P Mode eine Partition 1 und 2 existieren muss. Bei noobs fehlt die Nr 2... Liegt aber daran dass NOOBs nicht mehr offiziell von raspiBackup unterstuetzt wird.

ich habe bei mir noch eine alte NOOBs SD Karte gefunden und habe raspiBackup wie folgt gestartet:

`sudo raspiBackup.sh -P -T "1 5 6 7"`

und er werden brav 5 Partitionen gesichert. Den Restore habe ich jetzt nicht getestet. Warum es mit "*" nicht funktioniert muss ich mir noch ansehen.

> (alles erst diese Woche erstmals installiert)

Uebrigens nutzt Du sowohl ein altes raspiBackup wie auch einen alten Installer :cry: 

So sieht es bei mir aus:
```
raspiBackupInstallUI -h
raspiBackupInstallUI 0.4.3.6, 2022-02-03/19:49:13 - 434af05
raspiBackupInstallUI ( -i [-e]? | -u | -U )
-e: unattended (re)install of raspiBackup extensions
-i: unattended (re)install of raspiBackup
-h: display this help
-U: unattended update of raspiBackupInstallUI
-u: unattended uninstall of raspiBackup
pi@raspberrypinoobs:~ $ raspiBackup.sh --version
Version: 0.6.6.1 CommitSHA: c674d9f CommitDate: 2022-02-01 CommitTime: 12:16:54
```

hajoko1, 2022-02-10:
Danke erstmal. Werde mich mal dranmachen.

> Du nutzt sowohl ein altes raspiBackup wie auch einen alten Installer 😢

Hmmmm - diese Woche erst von Deiner Seite heruntergeladen und installiert. Bis dahin wusste ich nichtmal etwas von der Existenz des raspiBackup.

sudo raspiBackup -U
--- RBK0031I: Prüfe ob eine neue Version von raspiBackup verfügbar ist.
--- RBK0073I: /usr/local/bin/raspiBackup bereits auf der aktuellen Version 0.6.6.1.

???

framps, 2022-02-10:
> sudo raspiBackup -U
> --- RBK0031I: Prüfe ob eine neue Version von raspiBackup verfügbar ist.
> --- RBK0073I: /usr/local/bin/raspiBackup bereits auf der aktuellen Version 0.6.6.1.

Oben hast Du geschrieben 

> Nachtrag: raspiBackup ist Version 0.6.6, Config-Tool 0.4.3.5.

Wie bist Du an diese Info gekommen?

hajoko1, 2022-02-10:
sudo raspiBackup -h
     _raspiBackup **0.6.6.1**, 2022-02-01/12:16:54 - c674d9f
     Aufruf: raspiBackup [Option]* {Backupverzeichnis}_

sudo raspiBackupInstallUI
     _raspiBackup Installations- und Konfigurations Tool V**0.4.3.6**_

hajoko1, 2022-02-10:
Ich habe vorhin nur ein `raspiBackup -U` gemacht und bin von 0.6.6 auf 0.6.6.1 gegangen. raspiBackupInstallUI ist dabei von 0.4.3.5 auf 0.4.3.6 gegangen...

framps, 2022-02-10:
Damit hast Du alles upgraded. Dann verstehe ich nicht wie Du die alten Versionen installieren konntest wenn es vor ein paar Tagen war :thinking: 

Anyhow - egal - jetzt bist Du auf dem letzten Stand.

hajoko1, 2022-02-18:
Es läuft jetzt. Winziger Bug in raspiBackupInstallUI - bei Auswahl der Partitionen kann man zwar alle mit einem * versehen, danach gehts aber nicht weiter. Schließt man dann und startet neu, ist wieder die erste aktiv. War aber kein Problem, das im Config-File einzustellen. Läuft.

In der Mail von den Backups steht bei zweien: "raspix: Backup erfolgreich beendet.", bei einem aber steht ";-) raspix: Backup erfolgreich beendet.". Warum kommt der Smiley? Wars besonders schön? :-)

Danke für Deine Hilfe. Super-Programm!

framps, 2022-02-18:
> Winziger Bug in raspiBackupInstallUI

Nein. Ist es nicht :wink: raspiBackup unterstuetzt offiziell nur RaspbianOS und der Installer will dass die Partitionen 1 und 2 gesichert werden. NOOBS hat keine Partition 2 und deshalb kommt diese Loop. Aber wie Du festgestellt hast kann raspiBackup schon NOOBS sichern. Nur unterstuetzt der Installer das nicht und man muss die entsprechende Konfiguration manuell in der Konfig Datei vornehmen.

> Warum kommt der Smiley?

Siehe dazu [hier](https://www.linux-tips-and-tricks.de/de/faq/#a37)

hajoko1, 2022-02-18:
`raspiBackup -U` schreibt: ```
--- RBK0031I: Prüfe ob eine neue Version von raspiBackup verfügbar ist.
--- RBK0073I: /usr/local/bin/raspiBackup bereits auf der aktuellen Version 0.6.6.1.
```
Und trotzdem kommt :-) in der Mail.

hajoko1, 2022-02-18:
Ok, :-) ist weg. Auf einen Update von 0.6.6.1 auf 0.6.6.1 wäre ich nie gekommen. :-) war echt nicht wichtig, hatte mich nur gewundert. Danke trotzdem. :-) kommt bei den anderen zwei Raspis nicht, dabei erfolgte der frühere Update von 0.6.6 auf 0.6.6.1 zeitgeich auf allen 3 Raspis.

framps, 2022-02-18:
Ist ein Feature ...

Ohne -S waere der Smiley morgen erst weg gewesen.