framps/raspiBackup #445: Create ansible role for installing raspiBackup ?

CyberGitJul, 2022-02-22:
Hi,

just a short question. I want to install raspiBackup on several Pi in my network.
Wouldn´t it be a good thing for your tool, to additionally provide an ansible role to roll it out?

I would think so :-)

Thanks and cheers,

Julian

framps, 2022-02-22:
I know what Ansible is used for but I don't know how to create a rule.

CyberGitJul, 2022-02-24:
Hi, tried to do it, but I have problems.
Thought of generating the configuration with jinja templates, which can be used quite nicely in ansible, but specifiying the variables over your default config file is quite tough.
The problem is, that you´re using numbers and "numbers" there 

```
# zip tar or dd backup
DEFAULT_ZIP_BACKUP: "0"

# dd backup will save space used by partitions only
DEFAULT_DD_BACKUP_SAVE_USED_PARTITIONS_ONLY: 0

# how many backups to keep
DEFAULT_KEEPBACKUPS: "3"
# how many backups to keep of the specific backup type. If zero DEFAULT_KEEPBACKUPS is used
DEFAULT_KEEPBACKUPS_DD: 0
DEFAULT_KEEPBACKUPS_DDZ: 0
DEFAULT_KEEPBACKUPS_TAR: 0
DEFAULT_KEEPBACKUPS_TGZ: 0
```

and when ansible reads that variables the type information gets lost.
So when I use 

this template for example

{{ DEFAULT_KEEPBACKUPS }}
{{ DEFAULT_KEEPBACKUPS_DD }}

the result would be 

3
0

and I suppose these values won´t be valid with raspiBackup, right ?

Trying to generate a shell script now, which calls raspiBackup directly, but god ... all these options :-D

Cheers,

Julian

framps, 2022-02-24:
If you read the config file you will get

```
# zip tar or dd backup
DEFAULT_ZIP_BACKUP=0

# dd backup will save space used by partitions only
DEFAULT_DD_BACKUP_SAVE_USED_PARTITIONS_ONLY=0

# how many backups to keep
DEFAULT_KEEPBACKUPS=3
# how many backups to keep of the specific backup type. If zero DEFAULT_KEEPBACKUPS is used
DEFAULT_KEEPBACKUPS_DD=0
DEFAULT_KEEPBACKUPS_DDZ=0
DEFAULT_KEEPBACKUPS_TAR=0
DEFAULT_KEEPBACKUPS_TGZ=0
DEFAULT_KEEPBACKUPS_RSYNC=0
```

which is different than what you wrote. 

You mentioned type information is lost. But `"3"` is a string and `3` is a number.  Can't you grab this difference (even it's not there in the config file) in Ansible?

CyberGitJul, 2022-02-24:
hmmmm, on my side I got this config file I wrote with the numbers as strings and numbers in it.
If you say, you don´t have that in your config it should be fine .....

I´ll test it again this weekend...... 

Quick thing for you to sleep well tonight :-)

It´s an awesome job you did with that raspiBackup tool ..... really ....... 

Cya dude and take care.

framps, 2022-02-24:
I just double checked the config file which is published and there is `0` instead of `"0"` written all the time :thinking:

closed 2022-03-12 by github-actions[bot]