framps/raspiBackup #446: Seltamkeiten

hajoko1, 2022-02-25:
Moin. Wahrscheinlich kein Problem des raspiBackup - aber ein zusammen damit auftretendes reproduzierbares Problem.

Situation: drei Raspis, #1, #2, #3. RaspiBackup auf allen dreien.

#1 ist ein Raid5-NAS für das gesamte Netz.
#2 betreibt die alte Menalto-Gallery (wegen PHP5 ausgelagert), Files per NFS vom NAS.
#3 ist mein Test-Raspi und ist u.a. für die Arduino-IDE zuständig, Files per NFS vom NAS.

#1 schreibt seinen Backup direkt auf die Platten, #2 und #3 schreiben per NFS.

Backup #1 läuft, Backup #2 läuft, Backup #3  lief bis vorgestern auch.

Seit gestern passiert Folgendes: Backup auf #3 läuft an, bricht ab, da das Ziel auf #1 weder beschreibbar, noch beim Aufräumen löschbar ist. Gehe ich per SSH auf #1, ist auf dem NAS das Ziel-Verzeichnis nicht mehr erreichbar, es kommt nur Ein-/Ausgabefehler. 

Reboot, danach ist wieder alles ok. RaspiBackup auf #3 händisch angeworfen, alles bricht ab, Ein-/Ausgabefehler. Erneuter reboot, alles ok, Backup gestartet, Ergebnis gleich... Habe den Backup auf #3 jetzt erstmal aus dem Cronjob genommen, #1 läuft seitdem einwandfrei.

Es kann wohl kaum der raspi-Backup sein - oder doch? Aber was kann es dann sein? Hat jemand eine Idee?

Danke für Ideen. Gruß, Hajo.

framps, 2022-02-25:
Du hast nicht geschrieben welchen Backuptyp Du nutzt. Aber in jedem Falle stresst das von raspiBackup genutzte Linuxtool die Rootpartition. Ich vermute die SD Karte verabschiedet sich langsam oder das Filesystem ist korrupt. Auch solltest Du Mal in den Systemlogs nach Fehlermeldungen suchen. Ich denke da wirst Du fündig. Und ansonsten verweise ich auf [FAQ47](https://www.linux-tips-and-tricks.de/de/faq/#a47) 😉

closed 2022-02-25 by framps

hajoko1, 2022-02-25:
raspiBackup 0.6.6.1 mit TAR. Gestresst ist ja wohl die Rootpartition von #3. Es macht für mich nicht so richtig Sinn, warum dann über NFS auf dem Zielsystem #1 dessen USB-Platte sich seltsam verhält und erst nach reboot wieder läuft. Und warum funktioniert es aber problemlos, wenn #2 seinen Backup per NFS auf #1 ablegt? Ich werde mal die Systemlogs ansehen -auf #1 oder #3? FAQ47 werde ich mir ansehen. Sehr verwirrend das Ganze.

hajoko1, 2022-02-25:
Ok, habe FAQ47 gelesen. Ich habe das hierher geschrieben, weil das Problem reproduzierbar einzig und allein bei raspiBackup auftritt. Alle anderen (teilweise sehr starken Schreib-/Lesezugriffe) von #2 auf #1 per NFS laufen schon immer ohne Probleme. Deshalb hatte ich gehofft, dass Du einen Tip für mich hast.

hajoko1, 2022-02-25:
Ok, Du hast den Fall geschlossen. Schade. Werde es dann im Forum versuchen, mache mir aber wenig Hoffnungen. Wenn der raspiBackup die Rootpartition stark stresst, werde ich wohl das Programm besser von allen Raspis entfernen, ständige korrupte Platten am NAS und tägliche Reboots brauche ich nicht. Da mache ich lieber von Zeit zu Zeit einen SD-Kopie.

framps, 2022-02-25:
> Es macht für mich nicht so richtig Sinn, warum dann über NFS auf dem Zielsystem #1 dessen USB-Platte sich seltsam verhält und erst nach reboot wieder läuft.

Sorry. Mir ist eben klar geworden dass ich Dein Problem falsch verstanden habe. Ich dachte #3 haengt sich auf :disappointed: Dass sich jetzt das RAID5 System aufhaengt beim Backup von #3 und Backup von #2 OK ist ist schon merkwuerdig. Wie gesagt wuerde ich mal in den Systemlogs von #1 nachsehen ob Du da Indikationen findest. Kann es sein dass die Backuppartition die von #3 genutzt wird irgendwie anders ist als die von #2 und die von #3 ploetzlich buggy ist? 

> Wenn der raspiBackup die Rootpartition stark stresst,

Es ist tar - welches von raspiBackup genutzt wird um ein Backup zu erstellen - liest saemtliche Daten um sie zu sichern. D.h. es finden viele Lesezugriffe auf der Rootpartition statt und natuerlich auch viele Schreibzugriffe auf die Backuppartition - in Deinem Falle #1. 

> werde ich wohl das Programm besser von allen Raspis entfernen

Ein jeder Backuplauf stresst ein Dateisystem und das Medium. Stress im Sinne von benutzen - nicht im Sinne von zerstoeren :wink: Ich wuerde mir das ueberlegen denn manuelle Backups sind immer unzuverlaessig und werden gerne mal vergessen. Das ist ja der Grund warum ich raspiBackup geschrieben habe. Auch bist Du der Erste Nutzer von raspiBackup der Bedenken hat weil bei jedem Backuplauf viel gelesen und geschrieben wird.

> Du hast den Fall geschlossen. Schade. 

Dein Problem hat mit Deiner HW und/oder SW bzw dem RAID zu tun. raspiBackup nutzt das nur und bringt Dein Problem zu Tage und deshalb habe ich FAQ47 verlinked und den Issue geclosed.

> Werde es dann im Forum versuchen, mache mir aber wenig Hoffnungen.

Sei nicht so pessimistisch :wink: . In dem von mir verlinkten Forum sind viele gute Leute die Dir durchaus helfen koennen die Ursache zu finden. Und das definitiv noch besser als ich :smirk: