framps/raspiBackup #447: Server certificates are out of date => no installation possible

ImzFan, 2022-02-26:
## Note
raspiBackup is supported on RaspberryOS as operating system and Raspberry HW only. There are environments out there which successfully run raspiBackup but any support requests on unsupported environments will be rejected. 

If you run raspiBackup on RaspberryOS and a Raspberry please execute all following actions first and provide all requested information:

* Check whether the error messages of raspiBackup were listed on the messages page and all suggested actions don't fixes the issue. [English](https://linux-tips-and-tricks.de/en/rmessages) - [German](https://linux-tips-and-tricks.de/de/fehlermeldungen)
* Attach raspiBackup debug log. Just drag and drop raspiBackup.log into the issue and don't copy and paste the contents of raspiBackup.log into the issue.

#### If any requested information was not provided the question will be rejected.

## Note
Issues are closed automatically after 2 weeks if there is no activity any more

framps, 2022-02-26:
Works for me. Which OS version do you use and which error message do you get?

ImzFan, 2022-02-26:
Linux raspberrypi_1 4.19.66-v7+ #1253 SMP Thu Aug 15 11:49:46 BST 2019 armv7l GNU/Linux

My uname -a above
It is on my old Raspberry 2 running Pi-hole and Cups since years ;)
Do sometimes updates by hand but do not remember when it was the last time....

framps, 2022-02-26:
There were some changes in the let's encrypt certificates I use on my website which cause the certificate to be recognized as invalid. Just update you OS with `sudo apt-get update` and `sudo apt-get upgrade`and the certificate will no longer be invalid.

ImzFan, 2022-02-26:
Many thanks i didn't know that :)
Have a nice weekend!

framps, 2022-02-26:
Please close the issue if the update fixed your issue :wink:

closed 2022-02-26 by ImzFan