framps/raspiBackup #449: Add config option which causes raspiBackup to initiate a reboot when the backup finished

framps, 2022-03-05:


framps, 2022-03-06:
## Note
If this option is used the option `DEFAULT_STARTSERVICES` will be ignored because all services will be restarted when the system reboots.

framps, 2022-03-06:
Implemented with ea003c965e25b3f4655fd531303f0c1930116ba6

H-a-m-i, 2022-05-13:
Hallo framps,

aufgrund meiner aktuellen probiererei mit raspibackup habe ich dann auch mal endlich ein update auf die aktuelle Version gemacht :D 
Wo finde ich denn diese Einstellungen zum autom. Neustarten nach dem backup? Im Konfigurationstool habe ich dazu nichts gefunden.

LG Hami

framps, 2022-05-13:
Moin Hami,

mein Nickname ist framp - nicht framps :wink: Ich musste framps als Repoownernamen waehlen da framp schon belegt ist :disappointed: .

> Im Konfigurationstool habe ich dazu nichts gefunden.

Es ist eigentlich unueblich unter Linux bei einem so kleinen Tool wie raspiBackup einen Installer der auch die Konfiguration ermoeglicht zu haben. Den habe ich nur geschrieben um es Raspberryeinsteigern leicht zu machen raspiBackup zu installieren damit sehr schnell ein Backup erstellt werden kann. Man sollte ein Backup schon sehr bald regelmaessig erstellen wenn man mit der Raspberry anfaengt.
Aber ein Installer ist aufwaendig zu schreiben und zu testen. Auch wenn neue Optionen aufgenommen werden. Ausserdem wird der Installer bei den vielen Optionen die raspiBackup mittlerweile hat auch sehr unuebersichtlich. 
Deshalb habe ich mich auf die absolut wichtigsten Optionen im Installer beschraenkt. Alle Optionen die man dort nicht konfigurieren kann muss man auf die normale Linuxart mit einem Editor in einer Konfigurationsdatei konfigurieren. Diese heisst /usr/local/etc/raspiBackup.conf. Du kannst natuerlich auch die Aufrufnoption nutzen. Details zu der Option findest Du [hier](https://www.linux-tips-and-tricks.de/de/raspibackup/#parm_rebootSystem).

framps, 2022-05-13:
> Sorry,

Kein Problem :smile:

H-a-m-i, 2022-05-16:
Jetzt muss ich aber nochmal nerven...

Erklär nem Nixblicker wie mir doch bitte mal was ich falsch mache 🙈 das mit dem reboot klappt einfach nicht. Habe folgendes in der config:

[image]

Wenn ich dann simuliere mit `sudo raspiBackup.sh -F` geht nix mehr und ich muss den pi vom Strom nehmen.

Was mach ich denn schon wieder falsch? 😞

framps, 2022-05-16:
Die Option ist richtig eingegeben. Allerdings wird kein reboot im fakemode vorgenommen. Ich werde das gleich noch entsprechend dokumentieren. 

Warum sich Dein System aufhaengt kann ich nicht nachvollziehen. Wenn ich Option -F nutze mit reboot endet raspiBackup ganz normal ohne reboot.

Wenn Du den Stecker ziehst sollte in /tmp das bis dahin geschriebene Debuglog zu finden sein.

H-a-m-i, 2022-05-16:
Hmm ok, komisch.

im /tmp sind nur ein paar Ordner, die aber keine log dateien enthalten:

[image]

framps, 2022-05-16:
Sehr komisch. Welche Meldungen siehst Du denn noch bevor sich die Raspi aufhaengt?

framps, 2022-05-16:
In der Beschreibung zu der Option steht dass die Startservices nicht ausgefuehrt werden wenn sie an ist. Das fuehrt dazu dass bei der Nutzung von Option `-F` die Servicess die Du vorher gestoppt hast nicht mehr gestartet werden. Das fuehrt natuerlich zu Fehlern. Ich habe eben einen kleinen Fix eingebaut der dafuer sorgt dass auch alles wieder startet wird wenn die Option `-F` genutzt wird - auch wenn REBOOT_SYSTEM an ist.

Mit `sudo raspiBackup -U -S` holst Du Dir den Fix.

H-a-m-i, 2022-05-16:
Ok die Simulation läuft jetzt einwandfrei durch. Allerdings gibt das normale update jetzt einige Fehler und wird nicht fertig gestellt. Hab die log mal in meine dropbox hochgeladen. Am Ende sind einige Fehler, unter anderem auch wieder der msmtp Fehler
[raspiBackup.log](https://www.dropbox.com/s/r5xyenb0ypaj4n7/raspiBackup.log?dl=0)

framps, 2022-05-16:
Danke fuer das DebugLog.

Zu den vielen Fehlern: Die kommen durch die ACLs die der IOBroker definiert. Ich hatte gerade den Fall eines anderen Nutzers der auch mit IOBroker und ACLs kaempft. Die Kommunikation kannst dazu Du [hier](https://www.linux-tips-and-tricks.de/de/raspibackup/benutzung-von-synology/#comment-4754) nachlesen. Er hat einfach das `-A` entfernt.

Zu dem msmtp Problem: Da kann ich Dir leider nicht helfen. Ich nutze kein msmtp und es hat auch nicht direkt mit raspiBackup zu tun. Siehe dazu auch [FAQ47](https://www.linux-tips-and-tricks.de/de/faq/#a47). Pruefe mal ob alle Deine notwendigen Services gestartet sind. Vielleicht rebootest Du auch Dein System einfach mal. Dann sollte ja alle Services wieder oben sein.

H-a-m-i, 2022-05-16:
Dieser Typ war tatsächlich ich :-D Ich bin der Hami der's oft nicht blickt :-D
Ich hatte mich gewundert dass in der upgedateten config meine Zeilen mit dem `DEFAULT_RSYNC_BACKUP_OPTIONS` weg waren und unten am Ende jetzt steht: `  OPTION DELETED in config version "0.1.6" DEFAULT_RSYNC_BACKUP_OPTIONS="-aHx --delete" `
Hab's jetzt aber wieder eingetragen und jetzt geht's. Das mit den msmtp Fehlermeldungen ist auch erledigt, wie du sagtest hat ein reboot die Sache gelöst. Zwar komisch, da ich kurz vor dem backup einen reboot gemacht hatte, aber nun ja, hauptsache die Fehlermeldungen sind weg. Habe zwar die letzte Mail gerade nicht bekommen, aber da kein Fehler im log ist, wird das irgendwo anders dran liegen. Muss ich mal sehen...

Was dich aber vlt noch interessiert: beim update mit `-g` kommt der Fehler `??? RBK0131E: Programm pv nicht gefunden. Mit 'sudo apt-get update; sudo apt-get install pv' wird das fehlende Programm installiert.` obwohl ich es installiert hatte. Ich brauche `-g` nicht, wollte es nur erwähnt haben...

Ich danke dir tausend mal für deine Hilfe und den schnellen fix der config :)

H-a-m-i, 2022-05-17:
So ein kurzer Nachtrag noch:

Nach gefühlten 100 Test Backups heute Nacht und auch dem Test mit alten zurückgespielten backups, ist nun klar, dass `DEFAULT_REBOOT_SYSTEM=1` bei mir dafür verantwortlich ist, dass raspiBackup keine Emails sendet. Es steht zwar im log, dass die Mail raus geht, geht sie aber nicht. Nach dem Löschen des Eintrags in der config, geht es ohne Probleme.
Vlt wird durch raspiBackup ein Dienst gestoppt, der für die Mail gebraucht wird (obwohl ich eigentlich keinen Email Dienst stoppen lasse) und raspiBackup schickt die Mail vor dem Neustart raus, sodass der gestoppte Dienst noch nicht wieder gestartet ist und die Mail deshalb nicht raus geht 🤷‍♂️ Keine Ahnung, wie gesagt hab ich vom dem Thema eigentlich kein Plan.

Lange Rede, kurzer Sinn :-D Ich werde nun einfach weiterhin den raspi per cronjob jede Nacht rebooten lassen und damit ist das Problem locker umgangen 😄

framps, 2022-05-17:
> Vlt wird durch raspiBackup ein Dienst gestoppt, der für die Mail gebraucht wird

raspiBackup macht nur was mit STOP_SERVICES konfiguriert wurde. 

D.h. also der Reboot funktioniert soweit wie beabsichtigt. Nur die eMail geht nicht raus? 

> ch werde nun einfach weiterhin den raspi per cronjob jede Nacht rebooten lassen und damit ist das Problem locker umgangen

Das ist ein Workaround aber ich moechte dass die Option richtig funtioniert - also auch mit eMail - und Du sie nutzen kannst. Die Option ist offiziell fuer 0.6.7 angekuendigt und muss funktionieren. Es fehlt wohl nur noch eine Kleinigkeit :wink: 

Was ich mir vorstellen kann ist dass der Reboot zu schnell kommt so dass die eMail nicht mehr raus geht. D.h. hinter dieser Zeile https://github.com/framps/raspiBackup/blob/c51e541e6f867d1aebb5a15aa35e5cc05a7493f2/raspiBackup.sh#L9019 muss noch eine Zeile `sleep 1m` oder `sleep 3m` rein um sicherzustellen dass die eMail rausgeht. Kannst Du mal austesten ob es daran liegt und z.B. mal ein `sleep 5m` einfuegen? Wenn dann eine eMail kommt wissen wir dass es daran liegt.

framps, 2022-05-17:
> Was dich aber vlt noch interessiert: beim update mit `-g` kommt der Fehler `??? RBK0131E: Programm pv nicht gefunden. Mit 'sudo apt-get update; sudo apt-get install pv' wird das fehlende Programm installiert.` obwohl ich es installiert hatte. Ich brauche `-g` nicht, wollte es nur erwähnt haben...

Danke fuer den Hinweis. Ich sehe mir das mal an.

H-a-m-i, 2022-05-18:
Hmm also ich glaub der hat nicht 5 Min gewartet. Bin beim reboot zwar grad nicht davor gesessen, aber ich meine ich war auch keine 5 Minuten vom Bildschirm weg... Hab ich es falsch eingetragen? Eine neue Zeile hast du geschrieben:

[image]

framps, 2022-05-18:
Sorry. Mein Fehler. Die Zeile muss VOR den Reboot geschrieben werden.

H-a-m-i, 2022-05-18:
ok habs geändert. Er hat auch diesmal gewartet mit dem reboot, mail ist aber trotzdem nicht raus, obwohl im log steht mail gesendet

framps, 2022-05-18:
Hm ... Dann muss ich das bei mir mal nachstellen mit eMail. Dauert aber da ich gerade ziemlich busy mit andren Sachen bin.

H-a-m-i, 2022-05-18:
Ja kein Stress, wie gesagt, ich umgehe das Problem ja eh per cronjob 😉

framps, 2022-05-26:
Eben habe ich einen 1m sleep an der richtigen Stelle eingefuegt und ich bekomme eine eMail auch wenn ein reboot ausgefuehrt wird. Allerdings nutze ich exim. 

Die Aenderung habe ich published, d.h. Du kannst wenn Du willst 1m in 15m aendern um zu testen ob es ein Timingproblem bei Dir ist :wink:

H-a-m-i, 2022-05-27:
Ja geil, hab's als erstes mit der Minute getestet und es hat gleich funktioniert. 

Top DANKE DIR! =)

H-a-m-i, 2022-05-27:
Ach Mist, zu früh gefreut. Ich hatte beim "Einschalten" des auto reboot in der raspiBackup.conf einen Fehler gemacht und somit war der reboot gar nicht eingeschaltet beim Testen vorhin… 🙈 Das ist mir jetzt vorhin zufällig aufgefallen. Habe das gerade korrigiert, den auto reboot wirklich eingeschalten, den timer auf 15 minuten gestellt und noch mal ein backup gemacht. Leider immer noch das alte Ergebnis: backup läuft durch, aber mail wird nicht gesendet. 
Komische Sache echt...

framps, 2022-05-28:
Das ist dumm 😢 . Dann liegt es vermutlich am msmtp. Da kenne ich mich leider nicht aus. Wenn Du das gelöst haben willst solltest Du im Raspberryforum nach Rat suchen. Kannst mich da gerne in den Thread einladen um Dir Unterstützung zu geben.

framps, 2022-05-29:
Wenn Du da noch weiterbohrst update bitte diesen Issue mit Deinen Erkenntnissen. Ansonsten schliesse bitte diesen Issue.

H-a-m-i, 2022-05-30:
Ja prima, danke für das Angebot. Ich guck mir das im Laufe der Woche noch mal in Ruhe an und gucke dann wie es weiter geht und werde dann alles hier erweitern oder schließen...

H-a-m-i, 2022-06-10:
Also ich komme momentan nicht wirklich dazu, mich länger damit zu beschäftigen. Habe heute mal auf die schnelle mein Mail Versand neu eingerichtet bzw. installiert mit postfix, da ich Probleme mit gmail hatte. Läuft nun alles wieder, hab dann auch mal kurz raspiBackup getestet mit 15 minuten, aber es kommt immer noch keine Mail an.
Egal, ich lass es dann jetzt auch so, hab ja eine andere Lösung...

Sollte ich mich irgendwann mal wieder damit befassen und eine Lösung finden, werde ich bescheid geben.

✌️

H-a-m-i, 2022-06-10:
Ähm wie schließ ich denn datt ding hier? 😂

framps, 2022-06-10:
Unten ist ein close issue Button 😉

H-a-m-i, 2022-06-10:
Hmm entweder bin ich blind oder mein Browser ist kaputt 😂😂
[image]

framps, 2022-06-10:
:laughing: Sorry. Du kannst jeden Issue closen den Du erstellt hast. Diesen habe ich aber erstellt :wink: Also schliesse ich ihn fuer Dich.

commit c51e541e (2022-05-16): Fixed #449 - Issue with fake option
closed 2022-06-10 by framps