framps/raspiBackup #451: Cannot Upgrade to v0.6.7

SJMarty, 2022-03-14:
I have four Raspberry Pis running raspiBackup and I cannot get any of them to upgrade to v0.6.7.  They are all running v0.6.6.1.  I am running the following command to attempt to upgrade...

`sudo raspiBackup.sh -U -S`

I get the following message and then the terminal prompt returns...

`--- RBK0031I: Checking whether a new version of raspiBackup.sh is available.`

I have attached the log from one of my Pis.
[raspiBackup.log](https://github.com/framps/raspiBackup/files/8241027/raspiBackup.log)

framps, 2022-03-14:
There is no v0.6.7 out there as of now. I'm working on v0.6.7 beta which will be published the next couple of weeks..

FYI: raspiBackup will write a message when there is a new beta or new release available.

SJMarty, 2022-03-14:
Strange.  I received this e-mail notification from GitHub...

[From:]() framp <notifications@github.com> 
Sent: Wednesday, March 2, 2022 3:56 PM
To: framps/raspiBackup <raspiBackup@noreply.github.com>
Cc: Subscribed <subscribed@noreply.github.com>
Subject: [framps/raspiBackup] Release [v0.6.7](https://github.com/framps/raspiBackup/releases/tag/v0.6.7) - v0.6.7

v0.6.7
Repository: [framps/raspiBackup](https://github.com/framps/raspiBackup) · Tag: [v0.6.7](https://github.com/framps/raspiBackup/tree/v0.6.7) · Commit: [3e24488](https://github.com/framps/raspiBackup/commit/3e24488971d20ac858d51016e6c2d8f4eab3cec8) · Released by: [framps](https://github.com/framps)
Features
•	Add a "restore completed" and "restore ready" extension point https://github.com/framps/raspiBackup/issues/420
Enhancements
•	raspiBackup.msg not being created on new install? https://github.com/framps/raspiBackup/issues/437
•	Disable non supported environments https://github.com/framps/raspiBackup/issues/435
•	Create different Logfiles for Backup and Restore https://github.com/framps/raspiBackup/issues/432
•	Allow to customize color codes for console and html in eMail https://github.com/framps/raspiBackup/issues/360
Bugfixes
•	Make lookup for meta data files more robust https://github.com/framps/raspiBackup/issues/415
•	Restore with option -g misses progress indication for root partition https://github.com/framps/raspiBackup/issues/406
•	Check whether pv is installed if option -g is used https://github.com/framps/raspiBackup/issues/405
•	Restore terminates with RC112 for a Bullseye backup with no error message https://github.com/framps/raspiBackup/issues/400
•	Some files were not deleted when raspiBackup was uninstalled https://github.com/framps/raspiBackup/issues/397
For a detailed list of issues in this release see [here](https://github.com/framps/raspiBackup/issues?q=is%3Aissue+milestone%3A0.6.7+is%3Aclosed)
—
This release has 2 assets:
•	Source code (zip)
•	Source code (tar.gz)
Visit the [release page](https://github.com/framps/raspiBackup/releases/tag/v0.6.7) to download them.
—
You are receiving this because you are watching this repository.
[View it on GitHub](https://github.com/framps/raspiBackup/releases/tag/v0.6.7) or [unsubscribe](https://github.com/framps/raspiBackup/unsubscribe_via_email/ACWPXHXNIHM6UBB5SKNCC43U57IUFANCNFSM4BB3RFXQ) from all notifications for this repository.

SJMarty, 2022-03-14:
No problem at all.  Glad there is no issue.  🙂

closed 2022-03-14 by SJMarty