framps/raspiBackup #460: pihole backup using raspiBackup.sh V0.6.6 (83ecd03)

angelperezleon, 2022-03-27:
## Unable to backup pihole using raspiBackup
Using a Raspberry Pi 4b (4GB) with Linux rpi4-prox 5.10.103-v7l+ Raspi

I have "raspiBackup" working on 4 other Pi's (1x Pi 4b, 2 x Pi3b & 1x pi0 2w)

I Usually use this command:
`/usr/local/bin/raspiBackup.sh -o "systemctl stop bluetooth && pihole disable" -a "systemctl start bluetooth && pihole enable" -p /mnt/RemoteBackups2019 -t tar -k 4`

`--- RBK0009I: pi-hole: raspiBackup.sh V0.6.6 (83ecd03) started at Sun 27 Mar 11:19:01 CEST 2022.

--- RBK0151I: Using backuppath /mnt/nfs-mount with partition type nfs4.
  [i] Disabling blocking
  [✓] Pi-hole Disabled
--- RBK0036I: Saving partition layout.
256+0 records in
256+0 records out
268435456 bytes (268 MB, 256 MiB) copied, 6.2502 s, 42.9 MB/s
--- RBK0158I: Creating native tar backup "/mnt/nfs-mount/pi-hole/pi-hole-tar-backup-20220327-111900/pi-hole-tar-backup-20220327-111900.tar".
--- RBK0085I: Backup of type tar started. Please be patient.
tar: Removing leading `/' from member names
tar: /var/log/pihole.log: file changed as we read it
tar: /etc/pihole/pihole-FTL.db: file changed as we read it
tar: /etc/pihole: file changed as we read it
tar: Removing leading `/' from hard link targets
??? RBK0021E: Backupprogram for type tar failed with RC 1.
--- RBK0033I: Please wait until cleanup has finished.
  [i] Enabling blocking
  [✓] Pi-hole Enabled
--- RBK0043I: Removing incomplete backup in /mnt/nfs-mount/pi-hole/pi-hole-tar-backup-20220327-111900. This may take some time. Please be patient.
--- RBK0049I: Messages saved in /home/pi/raspiBackup.msg.
--- RBK0026I: Debug logfile saved in /home/pi/raspiBackup.log.
--- RBK0010I: pi-hole: raspiBackup.sh V0.6.6 (83ecd03) stopped at Sun 27 Mar 11:22:35 CEST 2022 with rc 109.
??? RBK0005E: Backup failed. Check previous error messages for details.
`
In past 6-month I probably had 3-4times success in backing up, just don't find a working method that help keep a weekly backup consistent without failing.
So hoping someone here can advise of a working method..

I am guessing the backups fails due to Pi-hole still getting traffic and logs getting update and pihole-FTL.db getting updated while backup is in operation, as shown in the above logs.
But what every I tried I cant get a consistent/predictable result to ensure raspiBackup is able to backup all the time without failures.

angelperezleon, 2022-03-27:
`/usr/local/bin/raspiBackup.sh -o "systemctl stop bluetooth && pihole disable && service pihole-FTL stop" -a "systemctl start bluetooth && pihole enable && service pihole-FTL start" -p /mnt/synology-nsf-share -t tar -k 4`

Looks like I got a working backup now for Pi-hole

Stopping the **pihole-FTL** service might be the key to consistent backups now, but I won't be convinced until the next schedule weekly backup is done, to be happy its fixed with this.

Just thought I share my findings here for all to see.

framps, 2022-03-27:
I also use pihole  and use `pihole disable` only. Actually I create the backup early in the morning where there is no traffic. Looks like you have traffic which causes an update of the FTL db when the backup is created.

You also may exclude the FTL db from the backup process. Then your backup will not have the latest backup of the db but as far as I understand that's just DNS caching stuff and will not make your backup inconsistent. But please double check this with pihole dev if you decide to go this way.

Crossing fingers the backup now will be reliable all the time by stopping the FTL. Please keep me updated so I can pass on your experiences to other pihole and raspiBackup users :wink:

angelperezleon, 2022-03-27:
Probably, but in my cronjobs i start it at 5:03AM every sundays, and chose this time/day as i have no activity around this time on the network, i.e. other backups, people browsing the web, updates, etc...
Only thing I haven't really looked at is any pihole logrotate scripts i need to deal with, but for now we can close this case if you want, and I will wait till next Sunday0s scheduled backup.
I am happy I have todays backup done, as its been months since last one!

Thanks for excellent support of this script!
It's should be bookmarked as Top 100 (at the very least) Raspberry Pi projects to be honest!

framps, 2022-03-27:
> pihole logrotate scripts i need to deal with

Some raspiBackup users stop cron which triggers logrotate :wink:

angelperezleon, 2022-04-03:
I can confirm todays weekly (Sunday) back-up has also worked, so closing this issue down and the aforementioned steps I took, worked for me in avoiding failed backups of a PiHole raspberry pi.

closed 2022-04-03 by angelperezleon