framps/raspiBackup #471: Renaming of backup directories may affect Smartrecycle and keep backup logic

framps, 2022-04-01:
The current logic assumes all backup directories were created by raspiBackup and follow a strict naming convention. If users manually rename backup directories for SR the wrong daily backup may be deleted and for keep backup the number of kept backups will be less than requests. In addition the debug logfile will be saved in home directory instead in the backup directory.

Workaround: If for some reasons a backup should be kept with a different name never use a path starting with `<backupPath>/<hostname>`.  Use for example `<backupPath>/mySavedBackupOf<hostname>`.

Make the SR and keep logic robust to ignore any other directories or write an error message and abort the backup if those kind of directories are detected.

framps, 2022-04-04:
The logic to clean up backups with the keep count assumes the directory names can be sorted in alphabetical order to identify the oldest backup to delete. If for example the hostname is `raspberry` and there exists a directory `SomeImportantBackupi` this will ` exclude this directory from the deletion process because it's alphabetically all the time the leading directory. For a keep count of 3 this directory will never be deleted and infact the number of normal created backup and kept dirs will be reduced to 2

framps, 2022-04-04:
Both keep and SR strategy assume the backup directories of <hostname> start with <hostname>/ <hostname>, followed by the backup type followed by the backup creation date and time. If the directories don't follow this naming convention the strategies are screwed up. Thus it's important to add an additional check during startup all existing backup directories follow this naming convention and abort the backup run.

framps, 2022-04-12:
Every backup run will check whether the existing backup directories follow the backup directory naming used by raspiBackup. If there are exist any other backup directories the backup raspiBackup will terminate with error message RBK0271E

closed 2022-04-12 by framps

bcutter, 2022-05-29:
Is this the reason for removing this in the 0.6.7 config update?

```
# >>>>> OPTION DELETED in config version "0.1.6" <<<<<
# DEFAULT_USE_HARDLINKS=1
```

I was/am using this for my rsync-Backups. Not sure if those still work as before udating to 0.6.7 and I really could find *nothing* about `DEFAULT_USE_HARDLINKS` in whole https://www.linux-tips-and-tricks.de/de/raspibackup/#alphabetisch :-(

framps, 2022-05-29:
Good catch :+1: 

In one of the first releases of raspiBackup you could set `DEFAULT_USE_HARDLINKS=1` to `0` to use rsync if no hardlink support of the underlying filesystem is available. This option is obsolete for a long time already and if you use rsync there has to be hardlink support of the underlying filesystem. Otherwise rsync cannot be used. That's why I removed this option from the config file.

Re `RSYNC_IGNORE_ERRORS="23 24"` it's not officially documented because it's discouraged to be used but mentioned at some places as a way to disable raspiBackup termination for these RCs and thus can still be used.