framps/raspiBackup #474: First message is "System will be rebooted", which is not configured that way

UomoInvisibile, 2022-04-03:
Hello framp,

on my Raspberry 4b (Arch Linux ARM - aarch64) with the current beta version 0.6.7 the first message issued is "RBK0268I: System will be rebooted at the end of the backup run.". Even though "DEFAULT_REBOOT_SYSTEM=0" is specified in the configuration file.

I hope I didn't do anything stupid wrong.

Thank you for your great job, Andreas

[raspiBackup.log](https://github.com/framps/raspiBackup/files/8404510/raspiBackup.log)

framps, 2022-04-03:
Thank you very much for the log. I just fixed the issue caused by a duplicate message number. Actually you should get a warning you use an unsupported environment but the message you got was defined later on with the same message number so you got the wrong message :disappointed_relieved: 

Fixed in https://github.com/framps/raspiBackup/commit/e9c1be6946d6f3a883df516002ed90822c2a00a7

closed 2022-04-03 by framps

framps, 2022-04-03:
@UomoInvisibile Just one question: I noticed in the log

`20220403-135458 DBG 1975:  --- raspiBackup.sh 0.6.7-beta, $/$ -`

which means the commit date and sha is missing. Did you grab raspiBackup directly from github? If you follow the official upgrade via `sudo raspiBackup.sh -U -S` you will get a deployed version which includes both information and makes my debugging life much easier :wink:

UomoInvisibile, 2022-04-03:
Thank you for providing a new version so quickly. 

To your question: yes, that's right, "my" script was a direct download from GitHub. I had an issue with the default way, but can't remember which one. Sorry for the inconvenience.