framps/raspiBackup #475: kein freier Speicher auf Ziel obwohl Quelle kleiner

TylonHH, 2022-04-06:
Moin,

ich versuche gerade ein Backup zurückzuspielen. (Mein eigentliches Ziel ist es ein laufendes System zu klonen um zu testen ob ich es auf 64bit umstellen kann -> [CrossGrading](https://wiki.debian.org/CrossGrading) )
Meine Backups liegen auf ein NAS welches gemountet ist. Quell- und Ziel SD-Karten sind 32GB.

Das Backup schlägt fehl mit:
`/dev/sda2: No free sectors available.`
`Failed to add #2 partition: No space left on device`

Hier noch ein paar nützliche Infos:
```
root@linux:/mnt/NAS/raspberrypi# sudo du -shl *
5.8G    raspberrypi-rsync-backup-20220326-000908
5.8G    raspberrypi-rsync-backup-20220331-030001
```

```
root@linux:/mnt/NAS/raspberrypi# sudo raspiBackup.sh -d /dev/sda /mnt/NAS/raspberrypi/raspberrypi-rsync-backup-20220331-030001
--- RBK0009I: linux: raspiBackup.sh V0.6.6.1 (b4749b7) Wed  6 Apr 23:45:24 CEST 2022 gestartet.
!!! RBK0018W: Ziel /dev/sda mit 29.81 GiB ist größer als die Backupquelle mit 29.27 GiB. Die root Partition wird entsprechend vergrößert um den ganzen Platz zu benutzen.
!!! RBK0065W: Gerät /dev/sda wird repartitioniert und die gesamten Daten werden gelöscht.
--- RBK0067I: Momentane Partitionen auf /dev/sda:
Number  Start   End      Size     Type     File system  Flags
 1      1.05MB  32010MB  32009MB  primary  fat32        lba
--- RBK0066I: Gerät /dev/sda wird überschrieben mit der gesicherten Boot- und Rootpartition.
--- RBK0069I: Bootpartition /dev/sda1 wird formatiert und erhält die zurückgespielte Bootpartition.
--- RBK0070I: Rootpartition /dev/sda2 wird formatiert und erhält die zurückgespielte Rootpartition.
--- RBK0038I: Bist Du sicher? j/N
j
--- RBK0050I: Backup wird von /mnt/NAS/raspberrypi/raspberrypi-rsync-backup-20220331-030001 zurückgespielt.
--- RBK0004I: Zweite Partition wird von 29.23 GiB auf 29.76 GiB angepasst.
--- RBK0033I: Bitte warten bis aufgeräumt wurde.
??? RBK0077E: Restore wurde fehlerhaft beendet. Siehe vorhergehende Fehlermeldungen.
```

```
root@linux:/mnt/NAS/raspberrypi# fdisk -l | egrep "^Disk /|^/dev"
Disk /dev/ram0: 4 MiB, 4194304 bytes, 8192 sectors
Disk /dev/ram1: 4 MiB, 4194304 bytes, 8192 sectors
Disk /dev/ram2: 4 MiB, 4194304 bytes, 8192 sectors
Disk /dev/ram3: 4 MiB, 4194304 bytes, 8192 sectors
Disk /dev/ram4: 4 MiB, 4194304 bytes, 8192 sectors
Disk /dev/ram5: 4 MiB, 4194304 bytes, 8192 sectors
Disk /dev/ram6: 4 MiB, 4194304 bytes, 8192 sectors
Disk /dev/ram7: 4 MiB, 4194304 bytes, 8192 sectors
Disk /dev/ram8: 4 MiB, 4194304 bytes, 8192 sectors
Disk /dev/ram9: 4 MiB, 4194304 bytes, 8192 sectors
Disk /dev/ram10: 4 MiB, 4194304 bytes, 8192 sectors
Disk /dev/ram11: 4 MiB, 4194304 bytes, 8192 sectors
Disk /dev/ram12: 4 MiB, 4194304 bytes, 8192 sectors
Disk /dev/ram13: 4 MiB, 4194304 bytes, 8192 sectors
Disk /dev/ram14: 4 MiB, 4194304 bytes, 8192 sectors
Disk /dev/ram15: 4 MiB, 4194304 bytes, 8192 sectors
Disk /dev/mmcblk0: 58.94 GiB, 63281561600 bytes, 123596800 sectors
/dev/mmcblk0p1        8192    532479    524288  256M  c W95 FAT32 (LBA)
/dev/mmcblk0p2      532480 123596799 123064320 58.7G 83 Linux
Disk /dev/sda: 29.81 GiB, 32010928128 bytes, 62521344 sectors
/dev/sda1        2048 62519295 62517248 29.8G  c W95 FAT32 (LBA)
```

```
root@linux:/mnt/NAS/raspberrypi# raspiBackup.sh --version
Version: 0.6.6.1 CommitSHA: b4749b7 CommitDate: 2022-03-08 CommitTime: 18:19:59
```

Hier noch die LOG
```
Disk /dev/sda: 29.81 GiB, 32010928128 bytes, 62521344 sectors
Disk model: Storage Device  
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0xdfdb1ad0

Old situation:

Device     Boot Start      End  Sectors  Size Id Type
/dev/sda1        2048 62519295 62517248 29.8G  c W95 FAT32 (LBA)

>>> Script header accepted.
>>> Script header accepted.
>>> Script header accepted.
>>> Script header accepted.
>>> Created a new DOS disklabel with disk identifier 0x12edf225.
/dev/sda1: Created a new partition 1 of type 'W95 FAT32 (LBA)' and of size 43.9 MiB.
/dev/sda2: No free sectors available.
Failed to add #2 partition: No space left on device
Leaving.
```

So ganz plausibel erscheint es mir nicht, oder?

Gruß

framps, 2022-04-06:
Zeige doch mal die Ausgaben von

```
sudo fdisk -l /dev/mmcblk0 | grep "Disk /dev"
```
und
```
sudo fdisk -l /dev/sda | grep "Disk /dev"
```

TylonHH, 2022-04-06:
> Zeige doch mal die Ausgaben von
> 
> ```
> sudo fdisk -l /dev/mmcblk0 | grep "Disk /dev"
> ```
`Disk /dev/mmcblk0: 58.94 GiB, 63281561600 bytes, 123596800 sectors`
> und
> 
> ```
> sudo fdisk -l /dev/sda | grep "Disk /dev"
> ```
`Disk /dev/sda: 29.81 GiB, 32010928128 bytes, 62521344 sectors`

TylonHH, 2022-04-06:
Hier die Ausgabe der **Quelle**
```
pi@raspberrypi:~ $ df -h
Filesystem                                       Size  Used Avail Use% Mounted on
/dev/root                                         29G  5.3G   23G  20% /
devtmpfs                                         430M     0  430M   0% /dev
tmpfs                                            462M  4.5M  458M   1% /dev/shm
tmpfs                                            462M   47M  415M  11% /run
tmpfs                                            5.0M  4.0K  5.0M   1% /run/lock
tmpfs                                            462M     0  462M   0% /sys/fs/cgroup
/dev/mmcblk0p1                                    44M   25M   19M  58% /boot
192.168.178.58:/volume1/Backup/Raspberry/PiHole  2.7T  1.9T  843G  70% /mnt/NAS
tmpfs                                             93M     0   93M   0% /run/user/999
tmpfs                                             93M     0   93M   0% /run/user/1000
```

framps, 2022-04-07:
> Anscheinend habe ich doch eine 64GB welche ich sichere

Wenn Du dd als Backupmethode nutzt muss das Geraet auf welches restored werden soll wenigstens genausogross sein wie das Backup. D.h. ein 64GB Backup passt nicht auf 32GB.

Lies Dir mal [FAQ16](https://www.linux-tips-and-tricks.de/de/faq/#a16) durch. Dort sind Loesungsmoeglichkeiten genannt. Oder Du nutzt ein tar oder rsync Backup. Dann werden von den 64GB nur die existierenden Daten gesichert und restored.

TylonHH, 2022-04-07:
> Oder Du nutzt ein tar oder rsync Backup.

> root@linux:/mnt/NAS/raspberrypi# sudo du -shl *
> 5.8G    raspberrypi-rsync-backup-20220326-000908
> 5.8G    raspberrypi-rsync-backup-20220331-030001

Müsste doch rsync sein, oder? Die Ordner des Backups sind knappe 6GB groß
Hier das Ergebnis des letzten Backups
`RBK0081I: Backup vom Typ rsync wird in /mnt/NAS/@HOSTNAME@/@HOSTNAME@-rsync-backup-20220407-030001 erstellt.`

framps, 2022-04-07:
Stimmt. Zeige doch mal das Debuglog vom Restore.

TylonHH, 2022-04-07:
https://pastebin.com/AcZ0YDYa

framps, 2022-04-07:
Vielen Dank fuer das Log. Darin finde ich

```
20220407-153428 DBG 3522:          --> calcSumSizeFromSFDISK /mnt/NAS/@HOSTNAME@/@HOSTNAME@-rsync-backup-20220407-030001/@HOSTNAME@-backup.sfdisk
20220407-153428 DBG 3526:                  *** cat /mnt/NAS/@HOSTNAME@/@HOSTNAME@-rsync-backup-20220407-030001/@HOSTNAME@-backup.sfdisk
20220407-153428 DBG 3526:                      label: dos
20220407-153428 DBG 3526:                      label-id: 0x12edf225
20220407-153428 DBG 3526:                      device: /dev/mmcblk0
20220407-153428 DBG 3526:                      unit: sectors
20220407-153428 DBG 3526:                      
20220407-153428 DBG 3526:                      /dev/mmcblk0p1 : start=        8192, size=       89854, type=c
20220407-153428 DBG 3526:                      /dev/mmcblk0p2 : start=       98304, size=    61306880, type=83
20220407-153428 DBG 3566:          <-- calcSumSizeFromSFDISK 31439322112
20220407-153428 DBG 7223:          --- soureSDSize: 31439322112 - targetSDSize: 32010928128
```

Die sourceSDSize ist falsch. Warum weiss ich momentan nicht und muss es mir genauer ansehen. 

Falls es dringend bei Dir ist: Es gibt einen Workaround: Du musst die Ziel SD Karte vorpartitionieren mit /boot (256M) und / (den Rest). Dann muss beim Restore zusaetzlich noch die Option `-0` (dash zero) mitgegeben werden. Dann sollte der Restore erfolgreich sein.

TylonHH, 2022-04-08:
> die Ziel SD Karte vorpartitionieren mit /boot (256M) und / (den Rest). Dann muss beim Restore zusaetzlich noch die Option `-0` (dash zero) mitgegeben werden

Danke, hat geklappt 👍 
Wenn du noch Infos brauchst wegen der falschen sourceSDSize sag Bescheid.

framps, 2022-04-08:
Danke :+1: . Das Log zeigt eine Menge und ich hoffe ich finde da alles was ich brauche. Ansonsten melde ich mich noch mal.

framps, 2022-04-08:
Ich habe den Fehler gefunden und einen [Fix](https://github.com/framps/raspiBackup/commit/7d86a3b9827319edbbc57e83921c54136ac8bd45) in [diesem Branch](https://github.com/framps/raspiBackup/tree/master_475) erstellt. Es waere gut wenn Du den Restore mit dem geaenderten [raspiBackup.sh](https://raw.githubusercontent.com/framps/raspiBackup/master_475/raspiBackup.sh) noch einmal auf dieselbe SD Karte vornehmen wuerdest um den Fix zu verifizieren :wink:

TylonHH, 2022-04-09:
Well done.
Ich habe die [raspiBackup.sh](https://raw.githubusercontent.com/framps/raspiBackup/master_475/raspiBackup.sh) nach /usr/local/bin kopiert. Läuft! Vielen Dank für den schnellen Support.

Hat geklappt:
```
pi@linux:~ $ sudo raspiBackup.sh -d /dev/sda /mnt/NAS/raspberrypi/raspberrypi-rsync-backup-20220407-030001
--- RBK0009I: linux: raspiBackup.sh V0.6.6.1 () Sat  9 Apr 00:05:47 CEST 2022 gestartet.
--- RBK0031I: Prüfe ob eine neue Version von raspiBackup.sh verfügbar ist.
--- RBK0168I: :-D raspiBackup.sh Beta Version 0.6.7 ist verfügbar. Hilfe beim Testen dieser Beta ist sehr willkommen. Einfach auf die neue Beta Version mit der Option -U upgraden. Die vorhergehende Version kann mit der Option -V wiederhergestellt werden
--- RBK0114I: Besuche https://www.linux-tips-and-tricks.de/de/versionshistorie/ um die Änderungen in der neuen Version kennenzulernen.
!!! RBK0018W: Ziel /dev/sda mit 29.81 GiB ist größer als die Backupquelle mit 29.27 GiB. Die root Partition wird entsprechend vergrößert um den ganzen Platz zu benutzen.
!!! RBK0065W: Gerät /dev/sda wird repartitioniert und die gesamten Daten werden gelöscht.
--- RBK0067I: Momentane Partitionen auf /dev/sda:
Number  Start   End      Size     Type     File system  Flags
 1      1.05MB  32010MB  32009MB  primary  fat32        lba
--- RBK0066I: Gerät /dev/sda wird überschrieben mit der gesicherten Boot- und Rootpartition.
--- RBK0069I: Bootpartition /dev/sda1 wird formatiert und erhält die zurückgespielte Bootpartition.
--- RBK0070I: Rootpartition /dev/sda2 wird formatiert und erhält die zurückgespielte Rootpartition.
--- RBK0038I: Bist Du sicher? j/N
j
--- RBK0050I: Backup wird von /mnt/NAS/raspberrypi/raspberrypi-rsync-backup-20220407-030001 zurückgespielt.
--- RBK0004I: Zweite Partition wird von 29.23 GiB auf 29.76 GiB angepasst.
--- RBK0053I: Erste Partition (Bootpartition) wird auf /dev/sda1 zurückgespielt.
--- RBK0055I: Zweite Partition (Rootpartition) wird auf /dev/sda2 zurückgespielt.
```

framps, 2022-04-09:
Perfekt :+1: . Vielen Dank fuer die Verifikation. Den Fix schiebe ich gleich noch in die aktuelle Beta 0.6.7 rein.

TylonHH, 2022-04-09:
Ich werde das Backup nochmal drauf spielen, ich habe irgendwie Probleme mit der dpkg
```
sudo raspi-config
dpkg-query: error: cannot scan updates directory '/var/lib/dpkg/updates/': No such file or directory
```

framps, 2022-04-09:
Das hat aber definitiv nichts mit diesem Issue zu tun :wink: 

Sieh mal im rsync Backup nach ob das Verzeichnis dort drin ist. Wenn es dort fehlt kann es auch im restoretem Backup nicht sein. Dann fehlte es offensichtlich im Original.

Falls Du rausfindest dass es doch ein Problem mit raspiBackup sein koennte mache bitte einen neuen Issue auf.

closed 2022-04-09 by framps