framps/raspiBackup #477: Benachrichtigungen an Telegram ermöglichen

m4ding, 2022-04-18:
Es wäre super, wenn Benachrichtigungen an einen Telegram Bot möglich wären. Eigentlich ist es nur eine default URL sowie der persönliche Token und der Text welcher gesendet werden soll.

Viele Grüsse
Martin

framps, 2022-04-18:
Ich bin kein Telegram Spezl. Aber Telegram wird doch unterstützt. Was fehlt denn da?

m4ding, 2022-04-18:
Ach mist, sorry übersehen. Hatte es im UI gesucht. Danke! Eine Frage habe ich noch: kann ich die Parameter in DEFAULT_TELEGRAM_NOTIFICATIONS kombinieren - mit Komma getrennt?

danke!

framps, 2022-04-18:
> kann ich die Parameter in DEFAULT_TELEGRAM_NOTIFICATIONS kombinieren - mit Komma getrennt?

Ja, aber ohne Kommata. Beispiel: `"SFm"` oder `"SFM"`. `m` und `M` koennen nicht gemeinsam genutzt werden.

Die Doku dazu ist wie ich jetzt gesehen habe etwas unklar. Ich werde sie jetzt noch updaten.

m4ding, 2022-04-18:
Tiptopp danke!

framps, 2022-04-18:
> Tiptopp danke!

Demnach kann der Issue geclosed werden? Das macht ueblicherweise er Ersteller des Issues :wink:

m4ding, 2022-04-18:
Sorry bin mit git nicht so firm

framps, 2022-04-18:
> Sorry bin mit git nicht so firm

No problem :smiley: