framps/raspiBackup #479: RaspiBackup: Restore fail

bad-friday, 2022-04-27:
On a Reaspberry Pi 4 B I try to restore from a RaspiBackup with
sudo raspiBackup.sh -d /dev/sdb /backup/raspberrypi/raspberrypi-rsync-backup-20220426-131409.
I get the error message «RBK0185E: Rootpartition creation failed with RC 1». Unfortunately this errocode isn't mentioned on https://www.linux-tips-and-tricks.de/en/faq/rmessages/.
df -k
Dateisystem 1K-Blöcke Benutzt Verfügbar Verw% Eingehängt auf
/dev/root 238982804 4824364 221946012 3% /
devtmpfs 1773756 0 1773756 0% /dev
tmpfs 1938620 0 1938620 0% /dev/shm
tmpfs 775452 1216 774236 1% /run
tmpfs 5120 4 5116 1% /run/lock
/dev/sda 239631980 4426188 222960392 2% /backup
tmpfs 387724 20 387704 1% /run/user/1000 
[raspiBackup.log](https://github.com/framps/raspiBackup/files/8570946/raspiBackup.log)

framps, 2022-04-27:
You try to restore the backup to the current active system running on '/dev/sdb' which doesn't make sense and is impossible. You have to at plugin a target SD card in an USB card reader and use something like `sudo raspiBackup -d /dev/sdc /backup/@HOSTNAME@/@HOSTNAME@-rsync-backup-20220426-131409` if your target SD card is connected as /dev/sdc.

Looks like I should add another test in raspiBackup to write a user friendly error message when you try to restore to the running system :smiley:

bad-friday, 2022-04-27:
Thank you framp, what you say is obvious, but the error message…

closed 2022-04-27 by bad-friday

framps, 2022-04-27:
> but the error message…

Have you seen issue #480 I just created to make sure there will be a more user friendly error message and this kind of restore request will be rejected :wink:

bad-friday, 2022-04-27:
That's great