framps/raspiBackup #482: 0.6.7-beta / raspiBackup does not run on one Pi / RBK0268E

bondskin, 2022-05-01:
Hi,

I am running three Raspberry Pi hosts, all with beta version 0.6.7-beta.
Pi hosts without issues: 

> pi@raspberrypi:~ $ uname -a
> Linux raspberrypi 5.10.103-v7l+ #1529 SMP Tue Mar 8 12:24:00 GMT 2022 armv7l GNU/Linux

The pi which produces RBK0268E:

> pi@raspberrypidns:~ $ uname -a
> Linux raspberrypidns 5.10.103-v8+ #1529 SMP PREEMPT Tue Mar 8 12:26:46 GMT 2022 aarch64 GNU/Linux

If I add parameter "--unsupportedEnvironment" it works.
The latest scheduled backup caused failed with

> !!! RBK0269W: @@@@@@@@@> HINWEIS <@@@@@@@@@
> !!! RBK0268W: Die Umgebung wird nicht unterstützt. raspiBackup kann korrekt funktionieren aber es wird keine Unterstützung gegeben!
> !!! RBK0268W: @@@@@@@@@> HINWEIS <@@@@@@@@@
> --- RBK0009I: raspberrypidns: raspiBackup.sh V0.6.7-beta (2022-04-15/6f85dbe) Sun May 1 05:00:02 CEST 2022 gestartet.
> !!! RBK0165W: =========> HINWEIS <=========
> !!! RBK0165W: Dieses ist eine Betaversion welche nicht in Produktion benutzt werden sollte.
> !!! RBK0165W: =========> HINWEIS <=========
> --- RBK0031I: Prüfe ob eine neue Version von raspiBackup.sh verfügbar ist.
> ??? RBK0239E: https://www.linux-tips-and-tricks.de/downloads/raspiBackup0613.properties/download?version=0.6.7-beta%1type=rsync%1mode=N%1keep=3%1func=B%1srs= kann nicht aus dem Netz geladen werden. HTTP code: 301. RC: 0
> --- RBK0033I: Bitte warten bis aufgeräumt wurde.
> --- RBK0043I: Unvollständiges Backup in /mnt/nas_nfs/raspiBackup/raspberrypidns/raspberrypidns-rsync-backup-20220501-050001 wird gelöscht. Das kann etwas dauern. Bitte Geduld.
> ??? RBK0005E: Backup fehlerhaft beendet. Siehe vorhergehende Fehlermeldungen.
> --- RBK0010I: raspberrypidns: raspiBackup.sh V0.6.7-beta (6f85dbe) Sun May 1 05:01:04 CEST 2022 beendet mit Returncode 135.

framps, 2022-05-01:
It looks like my provider had some networking issue this morning at 5AM. Try again and then it should work. I also created #483 for next release.

bondskin, 2022-05-01:
thanks @framps , I was not so concerned about the download, more about the unsupported environment.

framps, 2022-05-01:
> more about the unsupported environment.

Do I get you right you run a Raspberry with Raspberry OS but the the unsupported message? 

Please post results of following commands:

```
cat /sys/firmware/devicetree/base/model
cat /etc/os-release
cat /etc/rpi-issue
```

bondskin, 2022-05-01:
yes, correct

> pi@raspberrypidns:~ $ cat /sys/firmware/devicetree/base/model
> Raspberry Pi 4 Model B Rev 1.1pi

> pi@raspberrypidns:~ $ cat /etc/os-release
> PRETTY_NAME="Debian GNU/Linux 11 (bullseye)"
> NAME="Debian GNU/Linux"
> VERSION_ID="11"
> VERSION="11 (bullseye)"
> VERSION_CODENAME=bullseye
> ID=debian
> HOME_URL="https://www.debian.org/"
> SUPPORT_URL="https://www.debian.org/support"
> BUG_REPORT_URL="https://bugs.debian.org/"

> pi@raspberrypidns:~ $ cat /etc/rpi-issue
> Raspberry Pi reference 2020-02-13
> Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 5f884374b6ac6e155330c58caa1fb7249b8badf1, stage2

framps, 2022-05-01:
I forgot one additional command:

```
dpkg --print-architecture
```

bondskin, 2022-05-01:
no problem

> pi@raspberrypidns:~ $ dpkg --print-architecture
> armhf

framps, 2022-05-01:
Could you please provide the same info of the 4 commands also for one system which runs without the unsupported message?

bondskin, 2022-05-01:
working config (also on a Pi4 model)

> pi@raspberrypi:~ $ cat /sys/firmware/devicetree/base/model
> Raspberry Pi 4 Model B Rev 1.1
> 
> pi@raspberrypi:~ $ cat /etc/os-release
> PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
> NAME="Raspbian GNU/Linux"
> VERSION_ID="10"
> VERSION="10 (buster)"
> VERSION_CODENAME=buster
> ID=raspbian
> ID_LIKE=debian
> HOME_URL="http://www.raspbian.org/"
> SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
> BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
> 
> pi@raspberrypi:~ $ cat /etc/rpi-issue
> Raspberry Pi reference 2020-08-20
> Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 9a3a10bf1019ebb2d59053564dc6b90068bad27d, stage4
> 
> pi@raspberrypi:~ $ dpkg --print-architecture
> armhf

bondskin, 2022-05-01:
Buster vs Bullseye

framps, 2022-05-01:
Fix: https://github.com/framps/raspiBackup/commit/0ad0259d2b249bdda911febcad39bd04ae1aea70

Please verify https://github.com/framps/raspiBackup/blob/beta_482/raspiBackup.sh fixes your issue :wink:

bondskin, 2022-05-01:
no improvement,  log file attached
[raspiBackup.log](https://github.com/framps/raspiBackup/files/8598823/raspiBackup.log)

framps, 2022-05-01:
Thank you very much for the test. I now removed one check which seems to be too restrictive. Now there may be some false negatives I have to deal with but no false positives any more.

Please verify the fix again :wink:

bondskin, 2022-05-02:
it's the same URL, right?

??? RBK0268E: Only Raspberries running Raspberry PI OS are supported. Use option --unsupportedEnvironment to invoke raspiBackup3 WITHOUT ANY SUPPORT.
--- RBK0026I: Debug logfile saved in /home/pi/raspiBackup3.log
[raspiBackup3.log](https://github.com/framps/raspiBackup/files/8600876/raspiBackup3.log)
.

framps, 2022-05-02:
I'm sorry. I was 100% sure the previous fix works but I unfortunately missed to update the [final return value](https://github.com/framps/raspiBackup/commit/6fac740b193ef290e715c0d352f4c5658673c3b3#diff-1f94627977853a94ac6ec67a52f59e64a74be29d61890a56a98d61260c90bbf5R2267) in the check function :worried: 

Should work now :wink: 

Thank you very much for your patience.

bondskin, 2022-05-02:
No worries. Thank you for your great support. 

I'll test within next two days as I am traveling right now

bondskin, 2022-05-05:
works as designed, thank you :-)

closed 2022-05-05 by bondskin