framps/raspiBackup #484: Error 105

MrR341-GER, 2022-05-10:
Just something about my mounted cifs-volume I think...

Here's the log
[raspibackup.log](https://github.com/framps/raspiBackup/files/8659796/raspibackup.log)

MrR341-GER, 2022-05-10:
btw. the Host is down at the time of the backup... (and won't be up)

MrR341-GER, 2022-05-10:
And I forgot to explain the Problem:
After the backup eth0 isn't coming up again.
ifup eth0 also results in Problems.

Till I reboot I can't etablish any connection to my raspi (ssh and whatelse)

framps, 2022-05-10:
Accoring following messages in the debug log
```
20220510-030607 DBG 3014:              --> stopServices
20220510-030607 MSG 3020:                  --- RBK0008I: Services werden gestoppt: 'systemctl stop alsa-restore %1%1 systemctl stop binfmt-support %1%1 systemctl stop console-setup %1%1 systemctl stop cron %1%1 systemctl stop dbus %1%1 systemctl stop haveged %1%1 systemctl stop ifupdown-pre %1%1 systemctl stop keyboard-setup %1%1 systemctl stop kmod-static-nodes %1%1 systemctl stop ModemManager %1%1 systemctl stop networking %1%1 systemctl stop NetworkManager %1%1 systemctl stop plymouth-quit-wait %1%1 systemctl stop plymouth-quit %1%1 systemctl stop plymouth-read-write %1%1 systemctl stop polkit %1%1 systemctl stop rsyslog %1%1 systemctl stop rtkit-daemon %1%1 systemctl stop smartmontools %1%1 systemctl stop ssh %1%1 systemctl stop ufw %1%1 systemctl stop wpa_supplicant %1%1 systemctl stop nginx %1%1 systemctl stop php8.1-fpm %1%1 systemctl stop plexmediaserver %1%1 systemctl stop plymouth- %1%1 systemctl stop rpc-statd-notify'.
20220510-030607 DBG 3021:                  --- systemctl stop alsa-restore && systemctl stop binfmt-support && systemctl stop console-setup && systemctl stop cron && systemctl stop dbus && systemctl stop haveged && systemctl stop ifupdown-pre && systemctl stop keyboard-setup && systemctl stop kmod-static-nodes && systemctl stop ModemManager && systemctl stop networking && systemctl stop NetworkManager && systemctl stop plymouth-quit-wait && systemctl stop plymouth-quit && systemctl stop plymouth-read-write && systemctl stop polkit && systemctl stop rsyslog && systemctl stop rtkit-daemon && systemctl stop smartmontools && systemctl stop ssh && systemctl stop ufw && systemctl stop wpa_supplicant && systemctl stop nginx && systemctl stop php8.1-fpm && systemctl stop plexmediaserver && systemctl stop plymouth- && systemctl stop rpc-statd-notify
20220510-030607 DBG 2301:                  --> executeShellCommand systemctl stop alsa-restore && systemctl stop binfmt-support && systemctl stop console-setup && systemctl stop cron && systemctl stop dbus && systemctl stop haveged && systemctl stop ifupdown-pre && systemctl stop keyboard-setup && systemctl stop kmod-static-nodes && systemctl stop ModemManager && systemctl stop networking && systemctl stop NetworkManager && systemctl stop plymouth-quit-wait && systemctl stop plymouth-quit && systemctl stop plymouth-read-write && systemctl stop polkit && systemctl stop rsyslog && systemctl stop rtkit-daemon && systemctl stop smartmontools && systemctl stop ssh && systemctl stop ufw && systemctl stop wpa_supplicant && systemctl stop nginx && systemctl stop php8.1-fpm && systemctl stop plexmediaserver && systemctl stop plymouth- && systemctl stop rpc-statd-notify
Warning: Stopping dbus.service, but it can still be activated by:
  dbus.socket
Warning: Stopping rsyslog.service, but it can still be activated by:
  syslog.socket
Failed to stop plymouth-.service: Unit plymouth-.service not loaded.
20220510-030639 DBG 2308:                  <-- executeShellCommand 5
20220510-030639 MSG 3025:                  ??? RBK0048E: Ein Fehler trat beim Beenden von Services auf. RC 5.
20220510-030639 DBG 2251:                  --> exitError 105
20220510-030639 DBG 2258:                  <-- exitError 105
```

there is an issue when some services are stopped. Looks like dbus and rsyslog are not active and thus cannot be stopped. In addition there is some issue with plymouth-service.

I suggest to remove the inactive services from stopping first.

Next I found

```
20220510-030505 DBG 3294:              --- Mountpoint: /media/mnt
df: /mnt/winshare/videos: Host is down
```
and
```
20220510-030505 DBG 5627:              /dev/sda1 on /media/mnt type fuseblk (rw,nosuid,nodev,relatime,user_id=0,group_id=0,allow_other,blksize=4096)
```

I have no idea about your infrastructure but it looks like `20220510-030505 DBG 5627:              //PASQUAL-STANDPC/Videos on /mnt/winshare/videos type cifs (rw,relatime,vers=3.1.1,cache=strict,username=@@@@,domain=@@@@,uid=0,noforceuid,gid=0,noforcegid,addr=192.168.178.20,file_mode=0755,dir_mode=0755,soft,nounix,serverino,mapposix,rsize=4194304,wsize=4194304,bsize=1048576,echo_interval=60,actimeo=1)` is down but  raspiBackup will not fail. 
If you fix the previous issue you will still get the `host is down` message but raspiBackup will create a backup. If you don't want to see the error message that's the way you can fix this: 

Add `umount -f /mnt/winshare/videos` in `DEFAULT_STOPSERVICES` and the corresponding mount in `DEFAULT_STARTSERVICES` or use `DEFAULT_EXCLUDE_LIST` to exclude this directory

MrR341-GER, 2022-05-10:
The Problem is that after the backup is done, my raspi has no Internet connection (eth0: link is down)
I thought maybe the cifs is the Problem, since it shows up in the error log..
But if thats not the problem, my problem persists.
Should I do a new Ticket for the specidfied "eth0 link is down" problem?

framps, 2022-05-10:
> The Problem is that after the backup is done, my raspi has no Internet connection (eth0: link is down)

Hm ... that's strange. You create a backup on a cifs mounted partition which uses eth0 and the network connection shouldn't be stopped. So from my point of view you either have a network issue or an issue with your samba server. 

But I just detected  `systemctl stop ifupdown-pre` and systemctl stop networking` which may be the root cause for your issue. I think you should streamline your stop and start list - also to fix  your 105 issue. 

From my point of view you should delete (no guarantee the list is OK. Just my 2 cents)

```
systemctl stop alsa-restore 
systemctl stop binfmt-support
systemctl stop console-setup
systemctl stop haveged 
systemctl stop ifupdown-pre
systemctl stop keyboard-setup
systemctl stop kmod-static-nodes
systemctl stop ModemManager
systemctl stop networking
systemctl stop NetworkManager
systemctl stop plymouth-quit-wait
systemctl stop plymouth-quit
systemctl stop plymouth-read-write
systemctl stop polkit
systemctl stop rsyslog
systemctl stop rtkit-daemon
systemctl stop smartmontools
systemctl stop ssh
systemctl stop wpa_supplicant 
systemctl stop plymouth
systemctl stop rpc-statd-notify
```

MrR341-GER, 2022-05-12:
I updated it... but now it seemed that it didn't worked the previous night.. so I give it a few new runs... I try a manual run now and then I'll reboot my raspi and try 2 automatic runs the next 2 days.

closed 2022-05-27 by github-actions[bot]