framps/raspiBackup #487: raspiBackup

bad-friday, 2022-05-12:
I have set up a new USB stick with Raspibian Buster.
As I wanted to back up the system with raspiBackup.
I get the error message
"RBK0178E: Creation of /backup/raspPieo/raspPieo-rsync-backup-20220512-143931/raspPieo-backup.img failed with RC 1".
I don't know what to do with this.
The error code is not explained anywhere.

framps, 2022-05-12:
dd fails with rc 1 when raspiBackup tries to create a backup of the boot partition. rc 1 is an input/output error. 

I cannot help without the debug log.

bad-friday, 2022-05-12:
That's what I've done: sudo raspiBackup.sh -t rsync /backup
[raspiBackup.log](https://github.com/framps/raspiBackup/files/8681346/raspiBackup.log)

framps, 2022-05-12:
I found following two log statements in the debug log:

```
20220512-143936 DBG 5203:                          /dev/sdb on /media/swe/d8fea947-fe38-404b-a6bf-9c21550ea1a3 type ext4 (rw,nosuid,nodev,relatime,uhelper=udisks2)
20220512-143936 DBG 5203:                          /dev/sdb on /backup type ext4 (rw,relatime)
```
and
```
20220512-143937 DBG 5060:                      *** ls /backup/@HOSTNAME@
20220512-143937 DBG 5060:                          
20220512-143937 DBG 4694:                  --> bootPartitionBackup 
20220512-143937 DBG 4698:                      --- Starting boot partition backup...
20220512-143937 DBG 3811:                      --> colorAnnotation 1
20220512-143937 DBG 3835:                      <-- colorAnnotation 
--- RBK0044I: Creating backup of boot partition in /backup/@HOSTNAME@/@HOSTNAME@-rsync-backup-20220512-143931/@HOSTNAME@-backup.img.
20220512-143937 DBG 2435:                      --> executeCmd Command: dd if=/dev/sda1 of="/backup/@HOSTNAME@/@HOSTNAME@-rsync-backup-20220512-143931/@HOSTNAME@-backup.img" bs=1M
20220512-143937 DBG 2436:                          --- Redirect: S - Skips: 
dd: failed to open '/backup/@HOSTNAME@/@HOSTNAME@-rsync-backup-20220512-143931/@HOSTNAME@-backup.img': No such file or directory
20220512-143937 DBG 2459:                      <-- executeCmd 1
20220512-143937 DBG 3811:                      --> colorAnnotation 1
20220512-143937 DBG 3835:                      <-- colorAnnotation 
??? RBK0178E: Creation of /backup/@HOSTNAME@/@HOSTNAME@-rsync-backup-20220512-143931/@HOSTNAME@-backup.img failed with RC 1.
```

I see following untested environmental config on your side which may be the root cause for your issue:
1) You use no partition /dev/sdb1 but a full device /dev/sdb. I'm frankly not sure whether this is the root cause. As far as I know it should work - but I never tested raspiBackup with this kind of backup device.
2) You have mounted /dev/sdb on two different mountpoints. 

Are you able to execute following command?

`dd if=/dev/sda1 of="/backup/backup.img" bs=1M`

framps, 2022-05-12:
Hold on. I just detected I added an important fix this morning which is buggy and the root cause for you issue. I just fixed the issue.

Please execute `sudo raspiBackup.sh -U -S` to grab the fix and try again. 

Sorry for the inconvenience :worried:

bad-friday, 2022-05-12:
> Sorry for the inconvenience 😟

No problem, I'm retired and I'm not in a hurry.

> `sudo dd if=/dev/sda1 of="/backup/backup.img" bs=1M`

/home/swe/ sudo dd if=/dev/sda1 of="/backup/backup.img" bs=1M
256+0 records in
256+0 records out
268435456 bytes (268 MB, 256 MiB) copied, 4.69813 s, 57.1 MB/s
/home/swe/ 
Unfortunately the backup still didn't work.
/home/swe/ sudo raspiBackup.sh -U -S
--- RBK0031I: Checking whether a new version of raspiBackup.sh is available.
--- RBK0195I: Update raspiBackup.sh to version 0.6.7.
--- RBK0038I: Are you sure? y/N 
y
--- RBK0057I: Downloading file raspiBackup.sh from https://www.linux-tips-and-tricks.de.
--- RBK0072I: /usr/local/bin/raspiBackup.sh updated from version 0.6.7 to version 0.6.7. Previous version saved as /usr/local/bin/raspiBackup.0.6.7.sh. Don't forget to test backup and restore with the new version now.
--- RBK0026I: Debug logfile saved in /home/swe/raspiBackup.log.

/home/swe/ sudo raspiBackup.sh -t rsync /backup
--- RBK0009I: raspPieo: raspiBackup.sh V0.6.7 (2022-05-12/90c0c08) started at Thu 12 May 2022 09:22:53 PM CEST.
--- RBK0151I: Using backuppath /backup with partition type ext4.
!!! RBK0157W: No services to stop.
--- RBK0044I: Creating backup of boot partition in /backup/raspPieo/raspPieo-rsync-backup-20220512-212251/raspPieo-backup.img.
??? RBK0178E: Creation of /backup/raspPieo/raspPieo-rsync-backup-20220512-212251/raspPieo-backup.img failed with RC 1.
--- RBK0033I: Please wait until cleanup has finished.
??? RBK0005E: Backup failed. Check previous error messages for details.
--- RBK0010I: raspPieo: raspiBackup.sh V0.6.7 (90c0c08) stopped at Thu 12 May 2022 09:22:58 PM CEST with rc 114.
--- RBK0026I: Debug logfile saved in /home/swe/raspiBackup.log.
/home/swe/ df -k
Filesystem     1K-blocks     Used Available Use% Mounted on
/dev/root       29248956  3164900  24842348  12% /
devtmpfs         1826956        0   1826956   0% /dev
tmpfs            1959052        0   1959052   0% /dev/shm
tmpfs            1959052     8812   1950240   1% /run
tmpfs               5120        4      5116   1% /run/lock
tmpfs            1959052        0   1959052   0% /sys/fs/cgroup
/dev/sda1         258095    50884    207212  20% /boot
tmpfs             391808        4    391804   1% /run/user/1000
/dev/sdb       239631980 10606780 216779800   5% /backup

[raspiBackup.log](https://github.com/framps/raspiBackup/files/8682016/raspiBackup.log)

bad-friday, 2022-05-12:
> Complicate boolean expressions in bash are a bit tricky ...
You're telling me. Not only in bash…
This fix did it. Thank you.
/home/swe/ sudo raspiBackup.sh -t rsync /backup              
--- RBK0009I: raspPieo: raspiBackup.sh V0.6.7 (2022-05-12/993abe4) started at Thu 12 May 2022 09:52:41 PM CEST.
--- RBK0151I: Using backuppath /backup with partition type ext4.
!!! RBK0157W: No services to stop.
--- RBK0044I: Creating backup of boot partition in /backup/raspPieo/raspPieo-rsync-backup-20220512-215239/raspPieo-backup.img.
--- RBK0158I: Creating native rsync backup "/backup/raspPieo/raspPieo-rsync-backup-20220512-215239".
--- RBK0085I: Backup of type rsync started. Please be patient.
--- RBK0078I: Backup time: 00:04:51.
--- RBK0033I: Please wait until cleanup has finished.
--- RBK0159I: 3 backups kept for rsync backup type. Please be patient.
--- RBK0017I: Backup finished successfully.
--- RBK0010I: raspPieo: raspiBackup.sh V0.6.7 (993abe4) stopped at Thu 12 May 2022 09:57:44 PM CEST with rc 0.
--- RBK0026I: Debug logfile saved in /backup/raspPieo/raspPieo-rsync-backup-20220512-215239/raspiBackup.log.

framps, 2022-05-12:
Great. Thank you very much for your patience. And I'm glad you reported the issue immediately :+1: 

Please close the issue if you're fine now.

bad-friday, 2022-05-12:
Thank you for your effort.

closed 2022-05-12 by bad-friday

bad-friday, 2022-05-17:
Is it possible to delete a single backup which isn't needed anymore for sure?

framps, 2022-05-17:
Not sure I get your question. Do you think it's not possible because rsync uses hardlinks? It's indeed possible. See [this page](https://www.linux-tips-and-tricks.de/en/all-pages-about-raspibackup/572-raspibackup-how-do-hardlinks-work/) where I try to explain the way hardlinks work.

bad-friday, 2022-05-17:
Am I correct: one can delete every backup following the first one? From the youngest to the oldest?

framps, 2022-05-17:
Yes. As long as a backup includes a file linked with a hardlink this file will not be deleted even it was created in an earlier backup :wink:

bad-friday, 2022-05-17:
Thank you for the explanation.