framps/raspiBackup #488: Backup failed with Returncode 136

jl61, 2022-05-14:
Der Fehler tritt erst seit der aktuellen Version 0.6.7 auf. Mit der Betaversion hat das Backup noch funktioniert.
Das Problem scheint der Aufruf von egrep mit der Option -E zu sein.
Das Debug Log habe ich angehängt.
Hier der Abschnitt mit der Fehlermeldung:

--- RBK0151I: Backuppfad /home/@USER@/PiHole_Backup mit Partitionstyp nfs4 wird benutzt.
20220514-141359 DBG 3522:          --> isPathMounted /home/@USER@/PiHole_Backup
20220514-141359 DBG 3531:              --- Path: /home/@USER@/PiHole_Backup
20220514-141359 DBG 3539:          <-- isPathMounted 0
20220514-141359 DBG 6553:              *** ls -1 /home/@USER@/PiHole_Backup/@HOSTNAME@ | egrep -Ev "@HOSTNAME@\-(dd|ddz|rsync|tar|tgz)\-backup\-([0-9]){8}.([0-9]){6}"
20220514-141359 DBG 6553:                  ls: invalid option -- 'E'
20220514-141359 DBG 6553:                  Try 'ls --help' for more information.
20220514-141359 DBG 2353:          --> exitError 136
20220514-141359 DBG 2360:          <-- exitError 136
20220514-141359 DBG 4313:          --> cleanup 
20220514-141359 DBG 3815:              --> colorAnnotation 1
20220514-141359 DBG 3839:              <-- colorAnnotation 
[raspiBackup.log](https://github.com/framps/raspiBackup/files/8692928/raspiBackup.log)
[raspiBackup.log](https://github.com/framps/raspiBackup/files/8692929/raspiBackup.log)

framps, 2022-05-14:
Wie Du richtig herausgefunden hast gibt es hier ein Problem:
```
20220514-141359 DBG 6553:              *** ls -1 /home/@USER@/PiHole_Backup/@HOSTNAME@ | egrep -Ev "@HOSTNAME@\-(dd|ddz|rsync|tar|tgz)\-backup\-([0-9]){8}.([0-9]){6}"
20220514-141359 DBG 6553:                  ls: invalid option -- 'E'
20220514-141359 DBG 6553:                  Try 'ls --help' for more information.
```
Leider kann ich durch die Maskierung nicht erkennen was genau  bei Dir @USER@ und @HOSTNAME@ ist. Ich vermute damit haengt das Problem zusammen. Kannst Du mir sagen wie genau die beiden Teile aussehen? Wenn es Dir zu sensitiv ist kannst Du mir die Info auch gerne per eMail (Siehe meine Konatktseite) zuschicken.

framps, 2022-05-14:
Mir ist noch etwas anderes als moegliche Ursache eingefallen. Kann es sein dass im Backupverzeichnis Dateien oder Verzeichnisse stehen die nicht von raspiBackup erstellt wurden?

Kannst Du dazu mal die Ausgabe von

`ls -1 /home/@USER@/PiHole_Backup/@HOSTNAME@` 

zeigen?

jl61, 2022-05-14:
Der User müsste "nfs" sein, denn es wird ein Verzeichnis auf einem Synology NAS gemounted unter: /home/nfs/PiHole_Backup

Der Hostname ist "pi-hole-jl"

Ich hänge mal die Log-Datei vom letzten Backup mit der 0.6.7 beta an, das ja noch funktioniert hat.

[raspiBackup.log](https://github.com/framps/raspiBackup/files/8693567/raspiBackup.log)

jl61, 2022-05-14:
Screenshot der Ausgabe ist angehängt
[image]
:

framps, 2022-05-14:
Ok. Das ist es :smile: 

Deine Betaversion ist eine der ersten gewesen. Bis zum GA hat sich noch vieles getan. U.A. wird jetzt geprueft ob nur Backupverzeichnisse dort liegen. Die beiden Dateien raspibackup.msg und raspiBackup.log sind keine Backupverzeichnisse. Deshalb der RC 136. Eigentlich sollte eine sprechende Fehlermeldung kommen. Warum die nicht kommt muss ich noch untersuchen.

Loesche die beiden Dateien und dann sollte es keinen Abbruch mehr geben.

jl61, 2022-05-14:
Ja, das war es. Nach dem Löschen der beiden Dateien ist das Backup fehlerfrei durchgelaufen.
Danke für die Hilfe und noch ein schönes Restwochenende.

framps, 2022-05-14:
Pefekt :+1: 

Ich habe das bei mir noch mal getestet (auch ein raspiBackup.log erstellt) und ich bekomme eine Fehlermeldung wie es sein soll.

Koenntest Du noch mal die aktuelle Version mit `sudo raspiBackup.sh -U -S` downloaden und eine raspiBackup.log in dem Verzeichnis erstellen und testen ob Du nun die Fehlermeldung bekommst?

Falls nicht bitte noch einmal das Log zeigen. Ansonsten bitte den Issue closen.

jl61, 2022-05-14:
Ja, jetzt kommt eine Fehlermeldung, dass in dem Backup-Verzeichnis "Nicht von raspiBackup generierte" Dateien sind.
Ich hänge die beiden Screenshots an und schließe den Issue.
[image]
[image]

framps, 2022-05-14:
> Nicht von raspiBackup generierte"

Ich sehe die Meldung aber nicht in Deinem Screenshot :confused: 

Bei mir sieht es wie folgt aus:

```
sudo ./raspiBackup.sh -G de
--- RBK0009I: raspberrypi-bullseye: raspiBackup.sh V0.6.7 (2022-05-12/5d28645) Sat 14 May 2022 09:15:02 PM CEST gestartet.
--- RBK0262I: Dynamischer mount von /backup nicht ausgeführt da es schon gemounted ist.
--- RBK0116I: Konfigurationsdatei /usr/local/etc/raspiBackup.conf wird benutzt.
--- RBK0151I: Backuppfad /backup mit Partitionstyp nfs wird benutzt.
??? RBK0273E: 1 ungültige Backupverzeichnis(se) in /backup/raspberrypi-bullseye gefunden.
--- RBK0033I: Bitte warten bis aufgeräumt wurde.
--- RBK0043I: Unvollständiges Backup in /backup/raspberrypi-bullseye/raspberrypi-bullseye-rsync-backup-20220514-211500 wird gelöscht. Das kann etwas dauern. Bitte Geduld.
??? RBK0005E: Backup fehlerhaft beendet. Siehe vorhergehende Fehlermeldungen.
--- RBK0010I: raspberrypi-bullseye: raspiBackup.sh V0.6.7 (5d28645) Sat 14 May 2022 09:15:07 PM CEST beendet mit Returncode 136.
--- RBK0026I: Debug Logdatei wurde in /home/pi/raspiBackup.log gesichert.
```

jl61, 2022-05-15:
Diese Fehlermeldung bekomme ich nicht. Ich habe ein "test-verzeichnis" im Backup-Verzeichnis angelegt, damit das Backup wieder in Fehler läuft, aber die Fehlermeldung wie bei dir bekomme ich nicht.
Ich hänge die Log-Datei an.
[raspiBackup.log](https://github.com/framps/raspiBackup/files/8694516/raspiBackup.log)

jl61, 2022-05-15:
Als Unterschied sehe ich nur, dass Du Raspbian "Bullseye" nutzt und ich "Buster".

framps, 2022-05-15:
Ok. Vielen Dank fuer Deinen Hinweis :+1: . Ich teste das mal auf meinem Buster. 

Ausserdem reopene ich den Issue denn er ist zwar fuer Dich geloest aber die Fehlermeldung sollte schon auch geschrieben werden :smirk:

framps, 2022-05-15:
An Buster liegt es nicht :cry: Dachte ich mir auch fast. D.h. Ich kann das Problem nur fixen wenn Du mir hilfst da ich es nicht reproduzieren kann :wink: 

Kannst Du mal vor https://github.com/framps/raspiBackup/blob/a4819253e90278d5b148569ad4214f1e9e58c05d/raspiBackup.sh#L6557 eine neue Zeile mit `set -x` und nach https://github.com/framps/raspiBackup/blob/a4819253e90278d5b148569ad4214f1e9e58c05d/raspiBackup.sh#L6560 noch eine neue Zeile mit `set +x` einfuegen und das Debuglog providen?

Vielen Dank.

jl61, 2022-05-15:
Zeilen eingefügt und Backup "mit Fehler" durchgeführt.
Log-Datei hänge ich an.

[raspiBackup.log](https://github.com/framps/raspiBackup/files/8694800/raspiBackup.log)

closed 2022-05-15 by framps

framps, 2022-05-15:
> Log-Datei hänge ich an.

Perfekt. Damit habe ich den Fehler gefunden :smile: 

Ich habe meine Tests mit Msglevel `debug` ausgefuhert. Du mit `minimal`. Bloederweise wurde die Meldung nur ausgegeben wenn man Detailed angegeben hat :persevere: . Ein typischer Copy/Paste Fehler eben.

Vielen Dank fuer Deine Hilfe den Fehler zu finden :+1: .

jl61, 2022-05-15:
Bitteschön.
Ich habe gerade nochmal ein "fehlerhaftes" Backup mit der aktuellen Version erstellt. Die Fehlermeldung sieht nun so aus:

Ich denke, so sollte es auch sein.

[image].

framps, 2022-05-15:
Exakt so sollt es aussehen. Noch einmal vielen Dank fuer Deine lokale Verifikation des Fixes :+1:

commit d3985d6e (2022-05-15): Fixed #488