framps/raspiBackup #489: Fortsetzung keine Hardlinks

DarthWeber, 2022-05-16:
Hi Framp,

wie in den Kommis angekündigt mache ich hier weiter, mein Problem war ja, dass mein raspiBackup iwie keine Hardlinks erzeugen wollte. Also ich habe jetzt einen neuen Versuch gewagt und lande bei "Backupprogramm des Typs rsync beendete sich mit RC 23" - was ja laut deiner FAQ darauf hindeutet, dass sich während des Backups eine Datei verändert hat. 

Da da erste Backup mit rsync ca. 4-5h dauert, ist das ja auch nicht so ungewöhnlich. Meine Frage wäre: Wie finde ich raus, welche Datei sich geändert hat bzw. welchen Dienst ich stoppen sollte bzw. wer da Ärger macht ? 

Danke,
Tom

framps, 2022-05-16:
> Wie finde ich raus, welche Datei sich geändert hat bzw. welchen Dienst ich stoppen sollte bzw. wer da Ärger macht ?

[Hier ](https://www.linux-tips-and-tricks.de/de/faq/fehlermeldungen/#M0021) wird die Fehlermeldung RBK0021E und Aktionen wie man sie beseitigt beschrieben :wink:

DarthWeber, 2022-05-17:
Danke für die Quelle, damit habe ich - wenn auch über Umwege - dass Problem gelöst. Falls es jemanden hilft:

Ich habe die export auf dem NFS Server wie folgt geändert:  (fsid=0,rw,async,no_subtree_check,no_root_squash)

damit lief das backup erstens wesentlich schneller (einn paar min statt 4-5h) und zweitens auch fehlerfrei. Nun habe ich auch das zweite Backup gestartet und komme wieder zum eigtl. Phänomen, beide Backups scheinen fast gleich groß zu sein und die Angabe von du -sh mit oder ohne l erzeugt keinen Unterschied.

`
16G     /backup/Uhura4/Uhura4-rsync-backup-20220517-030843/
18G     /backup/Uhura4/Uhura4-rsync-backup-20220517-054703/
`
andererseits lief das zweite Backup so schnell durch (wenige Minuten), dass ich am Zweifeln bin, dass tatsächlich 18GB kopiert wurden. Hier nun die log Datei vom **zweiten** Durchlauf (eine msg habe ich nicht gefunden):
[raspiBackup.log](https://github.com/framps/raspiBackup/files/8705010/raspiBackup.log)

DarthWeber, 2022-05-17:
Hm, möglicherweise habe ich ein Verständnisproblem und alles war richtig.

Ausgabe von du -sh Uhura4-rsync-backup-20220517-054703 (es ist das **zweite** Backup):
`18G     Uhura4-rsync-backup-20220517-054703`

aber bei du -sh * (jetzt also das * statt des konkreten Verzeichnisnamens) erhalte ich:
`3,4G    Uhura4-rsync-backup-20220517-054703`

mit dem l erhalte ich stets 18G, egal ob sternchen oder nicht.

beim **ersten** Backup erhalte ich bei beiden Versionen die gleiche Größe von 16GB. Tja, war wohl alles Fehlalarm, sorry. Mir erscheinen 3,4G zwar immer noch viel für 2h, aber das kann ja vielleicht sein.

closed 2022-05-17 by DarthWeber

framps, 2022-05-17:
> Hier nun die log Datei vom **zweiten** Durchlauf (eine msg habe ich nicht gefunden):

Im Log finde ich 
```
20220517-054720 DBG 4883:                  --- LinkDest: --link-dest="/mnt/nfs/r2d2/@HOSTNAME@/@HOSTNAME@-rsync-backup-20220517-030843"
```
und diese Zeile sagt dass Hardlinks genutzt werden. Die von Mir genannte Meldung die auf die  Nutzung von Hardlinks hinweist bekommst Du nur mit MSG_LEVEL =  1. Du nutzt MSG_LEVEL=0.

> Tja, war wohl alles Fehlalarm, sorry.

Kein Problem. Ich sehe mal wie ich die FAQ noch zu dem Thema noch verbessern kann.

> Mir erscheinen 3,4G zwar immer noch viel für 2h, aber das kann ja vielleicht sein.

Das kannst nur Du beurteilen ob das OK ist. Jedenfalls werden Hardlinks genutzt :-)