framps/raspiBackup #494: backups corrupted

bad-friday, 2022-05-29:
Because of typing errors, I have destroyed all my backups.
Surely my mistake but this should be prevented by the software (backup medium the same as the target medium)
[raspiBackup.log](https://github.com/framps/raspiBackup/files/8793285/raspiBackup.log)

framps, 2022-05-29:
Ouch. That's bad :cry: . I'll add some additional code to reject such an ugly option combination.

framps, 2022-05-29:
There is already a check in place. Unfortunately you used `/dev/sdb/` instead of `/dev/sdb`. Because of the trailing `/` the check didn't succeed. I updated the check in https://github.com/framps/raspiBackup/commit/eebe9e362d6f3dcca2a813bdb72d4c9cbe72117b which will remove any trailing `/` now.

framps, 2022-05-29:
Not sure whether you want to verify the fix :woozy_face:

bad-friday, 2022-05-29:
No chance to test it because all my backups are shot 😢

bad-friday, 2022-06-02:
What went wrong here?
[/home/swe]sudo raspiBackup -d /dev/sdb /backup/DebianBullseye/DebianBullseye-rsync-backup-20220601-153441
--- RBK0009I: Bullseye: raspiBackup.sh V0.6.7 (2022-05-12/5d28645) started at Thu  2 Jun 13:18:37 CEST 2022.
ls: cannot access '/backup/DebianBullseye/DebianBullseye-rsync-backup-20220601-153441/DebianBullseye-backup.blkid': No such file or directory
??? RBK0034E: File /backup/DebianBullseye/DebianBullseye-rsync-backup-20220601-153441/DebianBullseye-backup.blkid not found.
--- RBK0033I: Please wait until cleanup has finished.
??? RBK0077E: Restore failed. Check previous error messages.
--- RBK0010I: DebianBullseye: raspiBackup.sh V0.6.7 (5d28645) stopped at Thu  2 Jun 13:18:41 CEST 2022 with rc 108.
--- RBK0026I: Debug logfile saved in /home/swe/raspiBackup.logr.

framps, 2022-06-02:
https://github.com/framps/raspiBackup/blob/master/.github/ISSUE_TEMPLATE/problem-report.md

bad-friday, 2022-06-02:
I have no log file available because I've restored from a different backup. But I believe the error is not caused by the restore rather through the backup. A missing file in the backup? Strange.

framps, 2022-06-02:
You have a log file. See lost message

bad-friday, 2022-06-02:
I agree but the log I have is from the successfull restore nothing else.

framps, 2022-06-02:
Looks like your backup is not complete for some reasons. Please show the contents of the backup directory.

bad-friday, 2022-06-02:
[backup.dir.txt](https://github.com/framps/raspiBackup/files/8825416/backup.dir.txt)

bad-friday, 2022-06-03:
I've found the logs of the two failed backups
[failed backups..zip](https://github.com/framps/raspiBackup/files/8830015/failed.backups.zip)
.

framps, 2022-06-04:
I'm now back home and thus was able to inspect your logs and the source code now in detail. It's impossible to give support on a mobile :weary: 

In backup.dir.txt you provided I detected

```
drwxr-xr-x   2 root root      4096 Jun  1 17:11 sdb
drwxr-xr-x   2 root root      4096 May 31 13:59 sdb1
drwxr-xr-x   2 root root      4096 May 31 19:02 sdb2
```

which cause raspiBackup to assume you created a partition oriented backup and thus expects a .blkid file in order to restore the backup. The backup logs you provided and also the contents of the backup dir shows you created a normal backup.

Frankly I have no clue why you created these subdirectories on your raspberry: It's very uncommon to have directories with these names in /. If these directories are empty in the backup directory just delete them and you will be able to restore the backup. If they are not empty save them and move them somewhere else and you also should be able to restore the backup.

bad-friday, 2022-06-04:
I've creates those directories to mount partitions. The directories are empty -usually-. The raspberry isn't up right now. I'll have a look tomorrow. 🥴

framps, 2022-06-04:
> I've creates those directories to mount partitions.

That's what I expected. Just use some subdirectory like /disk/sdb1, /disk/sdb2 and so on on your Raspberyy and you will not face this issue again with raspiBackup :wink:

bad-friday, 2022-06-04:
But I like those short name.
Should I remve them from the backups?

framps, 2022-06-04:
If they are empty just remove them and you will be able to restore the backup. 

If you want to be able to restore future backups created with raspiBackup without any issues you have to delete these directories on your system. If you insist to keep theses directories even there is a better way to store them you have remember to delete these directories in the backup every time you want to restore a backup.

bad-friday, 2022-06-04:
Thank you. So I'll follow your advice even is /sdb1 is shorter than /disk/sdb1.
No need anymore to restore those backups but for the future…

bad-friday, 2022-06-07:
I've removed a /sdc2 from my backup.
Now the restore hangs somewhwere
[raspiBackup.logr.txt](https://github.com/framps/raspiBackup/files/8855068/raspiBackup.logr.txt)
e

bad-friday, 2022-06-07:
I was waiting for an answer and forgot the close.

bad-friday, 2022-06-08:
Next Problem:
I want to "restore". raspiBackup always complains that partitions of the target device are already mounted.
[raspiBackup Problem.txt](https://github.com/framps/raspiBackup/files/8861098/raspiBackup.Problem.txt)

framps, 2022-06-08:
Partitions used for the restore should not be mounted. Unfortunately there is no `mount` log debug statement in raspiBackup.

Please show the output of `mount` just before you start the restore which fails with RBK0274E.

bad-friday, 2022-06-08:
That's what I have:
[/home/swe]sudo mount /dev/sda /backup
[/home/swe]df -k
Filesystem     1K-blocks     Used Available Use% Mounted on
/dev/root      238982804  3531700 223238676   2% /
devtmpfs         1773760        0   1773760   0% /dev
tmpfs            1938624        0   1938624   0% /dev/shm
tmpfs             775452      924    774528   1% /run
tmpfs               5120        4      5116   1% /run/lock
/dev/sdb1         258095    52555    205540  21% /boot
tmpfs             387724       16    387708   1% /run/user/1000
/dev/sda       239631980 20904576 206482004  10% /backup
[/home/swe]lsblk
NAME   MAJ:MIN RM   SIZE RO TYPE MOUNTPOINT
sda      8:0    1 233.3G  0 disk /backup
sdb      8:16   1 232.9G  0 disk 
sdb1   8:17   1   256M  0 part /boot
sdb2   8:18   1 232.6G  0 part /
sdc      8:32   1 232.9G  0 disk 
sdc1   8:33   1   256M  0 part 
sdc2   8:34   1 232.6G  0 part

bad-friday, 2022-06-08:
[raspiBackup Problem.txt](https://github.com/framps/raspiBackup/files/8863684/raspiBackup.Problem.txt)

bad-friday, 2022-06-09:
But mine is raspiBackup.sh V0.6.7 (2022-05-12/5d28645). What have you changed in the meantime?
How do you determine whether a partition is "mounted"?

framps, 2022-06-09:
Latest version is 
```
raspiBackup : Version: 0.6.7 Date: 2022-06-06 17:42:10  Sha: afaeb8b   
```

> What have you changed in the meantime?

See [here](https://github.com/framps/raspiBackup/pull/491/files)

> How do you determine whether a partition is "mounted"?

See link above

bad-friday, 2022-06-09:
But raspiBackup checks for a newer version. Why didn't it recognize the newer version?

framps, 2022-06-09:
It's not a new version: It's still 0.6.7. It's just a minor new code level which doesn't need a new release. That's why you manually have to trigger the update.

bad-friday, 2022-06-09:
I've tried the new version. It seems to be ok now.

closed 2022-06-09 by framps