framps/raspiBackup #496: Status feststellen

DarthWeber, 2022-06-07:
Hi Framp, ich möchte gerne den aktuellen Stand meiner Backups anzeigen. Kann ich irgendwie festellen, ob raspiBackup gerade läuft, idealerweise auch welcher Typ und wie lange ?

Ich dachte, ich werte einfach das lockfile aus. Aber das führt nur zum Datum und außerdem bleibt es nach Erfolg ja erhalten. Mit ps komme ich auch nicht so richtig weiter, da dort immer 3 Prozesse sind und natürlich nicht zu erkennen ist, ob die letzte Ausführung fehlerfrei war.

framps, 2022-06-07:
> Ich dachte, ich werte einfach das lockfile aus.

/var/lock/raspiBackup wird erstellt wenn raspiBackup laeuft und am Ende wieder geloescht. D.h. darueber kannst Du feststellen ob raspiBackup laeuft.

> idealerweise auch welcher Typ und wie lange ?

Was meinst Du mit Typ? Wenn es der Backuptyp ist: Den kannst Du indirekt aus dem Backupverzeichnisnamen des aktuellsten Backups herleiten. Wie lange: Das koenntest Du indirekt aus der Msgdatei lesen. Die Message RBK0009I enthaelt die Startzeit. Solange raspiBackup laeuft findest Du die Messages in /tmp/raspiBackup.msg.

DarthWeber, 2022-06-07:
> > Ich dachte, ich werte einfach das lockfile aus.
> 
> /var/lock/raspiBackup wird erstellt wenn raspiBackup laeuft und am Ende wieder geloescht. D.h. darueber kannst Du feststellen ob raspiBackup laeuft.
> 
das klappt leider nicht. Auf einigen meiner Hosts ist das Backup erfolgreich durch:
RBK0017I: Backup erfolgreich beendet.
aber die lock-Datei ist noch da. Das Script selbst stört es nicht, wenn ich raspiBackup starte, aktualisiert es nur den Timestamp und rennt los.
> Was meinst Du mit Typ? Wenn es der Backuptyp ist: Den kannst Du indirekt aus dem Backupverzeichnisnamen des aktuellsten Backups herleiten. Wie lange: Das koenntest Du indirekt aus der Msgdatei [...]
Ja, aber an dem Backupverzeichnis sehe ich nicht, ob es noch läuft. Die Msgdatei finde ich sowohl in /tmp als auch im /home. Die Datei unter /tmp ist aber nach dem Backup weg.

Ich nutze 3 unterschiedliche Backuptypen (rsync, tgz und seltener ddz). Da hätte ich gerne eine einfache Übersicht über alle Hosts, wann welche Backups erfolgreich beendet werden konnten oder in die Hose gingen.

framps, 2022-06-07:
> aber die lock-Datei ist noch da

Tatsaechlich. Das sollte eigentlich nicht so sein. Ich sehe mir das noch mal genauer an und werde im naechsten Release dafuer sorgen dass diese Datei am Ende wieder geloescht wird.

Als Workaround fuer Dich koenntest Du ein winziges Plugin schreiben welches am Ende des Backups die Datei wieder loescht.

> Die Msgdatei finde ich sowohl in /tmp als auch im /home. Die Datei unter /tmp ist aber nach dem Backup weg.

Waehrend das Backup laeuft steht die Datei in /tmp und wird am erfolgreichen Ende ins Backupverzeichnis  oder wenn der Backup failed ins Homeverzeichnis gemoved.  Ich habe Dich so verstanden dass Du zur Laufzeit Infos haben moechtest und zu dem Zeitpunkt steht die Datei in /tmp.

> Ich nutze 3 unterschiedliche Backuptypen (rsync, tgz und seltener ddz). Da hätte ich gerne eine einfache Übersicht über alle Hosts, wann welche Backups erfolgreich beendet werden konnten oder in die Hose gingen.

Interessanter Ansatz. Finde ich fuer mich unnoetig da meine Backups zu 99.9% immer erfolgreich sind und ich sofort per email mitbekomme dass ein Backup gefailed ist. Aber Du wirst Deine Gruende dafuer haben.

Dir ist klar dass Du in dem Moment wo Du auf solche internen Dinge von raspiBackup zugreifst - die ich Dir gerne mitteile - eine starke Abhaengigkeit auf die Implementierung bzw die jeweilige Release von raspiBackup bekommst und bei einem Releasewechsel Probleme bekommen kannst. Ich denke das Verhalten mit den Logs und Msgs  wird sich nicht mehr aendern und die Messages sind eigentlich auch stabil. Aber ich kann Dir nicht garantieren dass sich zukuenftig dort doch irgendwas aendern wird. Ich achte streng auf Backwardcompatibility bei neuen Releases - aber wer auf Internas von raspIBackup zugreift muss sich bewusst sein dass die sich jederzeit aendern koennen.

DarthWeber, 2022-06-08:
Ja, das hatte ich auch schon gedacht: Wenn man da auf die internen Dinge zugreift, funktioniert das u.U. nicht mehr. 

Einen Ansatz für das Backup ... naja, ich hab immer dd gemacht, aber aus Erfahrung waren meine Probleme bisher eher, dass die primäre kleine Partition futsch war. Da ist ddz zwar rettend, aber auch extrem umständlich.

rsync mag ich sehr, funktioniert aber nicht über das Internet. Ich würde gerne einen Backupsatz "außer Haus" lagern. Da wäre aufgrund der geringen Datenmenge natürlich rsync ideal, aber damit komme ich nicht über webdav in meine cloud. Also tgz und rsync zum Sichern auf mein Heim-NAS....

framps, 2022-06-08:
Das Thema Cloud als Backuptarget kommt immer mal wieder hoch. Ich hatte mal versucht einen rsync Server als Target einzubinden. Aber der Aufwand ist schon sehr hoch wie auch die Logik nicht ganz unkompliziert ist. Deshalb habe ich es wieder eingestellt. 

Du koenntest wenn Du programmieren kannst ein Wrapperscript ([Beispiel](https://github.com/framps/raspiBackup/blob/master/helper/raspiBackupWrapper.sh) schreiben wo nach erfolgtem Backup das Backup in Deine Cloud geschoben wird. Das der Upload sicherlich recht lange dauert wg der geringen Bandbreite ist wohl ein asynchroner Ansatz ratsam.

closed 2022-06-23 by github-actions[bot]