framps/raspiBackup #499: Backupfiles mit Passwort schützen

DarthWeber, 2022-06-08:
Hi Framp, es ist wohl ein feature-request: Da ich meine Backups teilweise auf einem share speichere, der von mehreren Leuten zugänglich ist, würde ich gerne meine Backupfiles (tgz, ddz) gerne mit einem Passwort versehen können. Oder geht das schon mit irgendwelchen Tricks?

Danke,
Tom

framps, 2022-06-08:
Das ist in der Tat ein Feature request. Bei denen betrachte zuerst ich immer das Verhaeltnis Aufwand zu Anzahl Nutzer. Da Du der Einzige bist der dieses Feature haben moechte ist das Verhaeltnis sehr schlecht. Anyhow stelle ich den Request mal in den Backlog. Vielleicht finden sich ja noch weitere Interessenten.

Es gibt aber einen Loesungsweg fuer Dich: Nutze ein Wrapperscript ([Beispiel](https://github.com/framps/raspiBackup/blob/master/helper/raspiBackupWrapper.sh)). Nach erfolgtem Backup greifst Du auf die Backups zu und verschluesselst sie. Das natuerlich nicht auf einer oeffentlichen Partition. Danach movest Du sie auf eine oeffentlich Partition und gut ist. Beim Restore musst Du natuerlich entweder ein Wrapperscript fuer Restore schreiben (Da gibt es auch schon [eines](https://github.com/framps/raspiBackup/blob/master/helper/raspiBackupRestoreHelper.sh)) oder Du musst vor dem Restore manuell das Backup erst entschluesseln.

Alles erfordert aber ein wenig Programmierarbeit ...

DarthWeber, 2022-06-08:
das wrappen etc. würde aber nicht mit dem smart recycle harmonieren, oder ? Ich glaube, da warte und hoffe ich lieber auf Resonanz. Eigtl ist das Sichern auf externen Medien doch ganz sinnvoll und sensible Daten sind ja auch immer vorhanden, also ist das Verschlüsseln eigtl. ganz sinnvoll. 

Schade, ich dachte auch, das wäre einfacher. Im Grunde müsste man das ja nach dem gzip nur durch openssl oder gpg pipen bzw. umgekehrt beim Auspacken. Das ist jetzt aber auch nur meine naive Vorstellung, so gut kenne ich dein Script ja auch nicht.

Naja, vielleicht findet es mal seinen Weg in dein Script. Wenn ich etwas beitragen kann, lass es mich wissen ;-)

framps, 2022-06-08:
> das wrappen etc. würde aber nicht mit dem smart recycle harmonieren, oder

Das sollte funktioniern.

> Im Grunde müsste man das ja nach dem gzip nur durch openssl oder gpg pipen bzw. umgekehrt beim Auspacken.

Das stimmt. Das zu codieren ist denke ich ueberschaubar. Aber das richtig zu testen ist zeitaufwaendig :sob: Denn es muss der Backup als auch Restore sorgsam getestet werden. Dann muss zum Schluss dieser Test auch noch in die Regressionsuite eingebaut werden da das eine Erweiterung der Backup/Restore Funktionalitaet von raspiBackup ist und in jedem neuen Release regressiongetestet werden muss.

framps, 2022-06-10:
Moved in backlog

closed 2022-06-10 by framps