framps/raspiBackup #500: smartRecycle Option

DarthWeber, 2022-06-10:
kurze Frage: ie intelligente Rotationsstrategie schaltet man mit der Option --smartRecycle an. Wenn ich aber DEFAULT_SMART_RECYCLE="1" im conf file habe - wie schalte ich das dann per command line aus? Oder **muss** ich dann im conf file DEFAULT_SMART_RECYCLE="0" setzen und dann ggf. mit --smartRecycle in der Oprion einschalten ?

framps, 2022-06-10:
> Oder **muss** ich dann im conf file DEFAULT_SMART_RECYCLE="0" setzen und dann ggf. mit --smartRecycle einschalten ?

Nein. Das waere ziemlich umstaendlich. Einfach ein `-` dranhaengen. Also mit `--smartRecycle-`. Ist auch [hier](https://www.linux-tips-and-tricks.de/de/raspibackup/#parameter) beschrieben :wink:

DarthWeber, 2022-06-12:
sowas, wieso nutze ich -z- und komme da nicht drauf ? DANKE !

closed 2022-06-12 by DarthWeber