framps/raspiBackup #501: Rootpartitionserstellung endet fehlerhaft mit RC 1.

DarthWeber, 2022-06-10:
sorry, ich bin echt ein Pflegefall. Nachdem die Backups prima laufen, scheitere ich nun am restore (unter ubuntu20):

??? RBK0185E: Rootpartitionserstellung endet fehlerhaft mit RC 1.

es ist komisch: Die sd hängt unter /dev/sdd und ist nicht gemounted:

umount /dev/sdd1
umount: /dev/sdd1: not mounted.
root@r2d2:~# umount /dev/sdd2
umount: /dev/sdd2: not mounted.

und auch im log:

20220610-092647 DBG 5268:                  --- Creating mountpoint /tmp/raspiBackup
20220610-092647 DBG 5271:                  --- Umounting boot partition /dev/sdd1
umount: /dev/sdd1: not mounted.
20220610-092647 DBG 5278:                  --- Umounting root partition /dev/sdd2
umount: /dev/sdd2: not mounted.
20220610-092647 MSG 5301:                  --- RBK0052I: Partitionen werden auf /dev/sdd erstellt.

aber laut log ist sie es kurz danach doch:
20220610-092654 MSG 5408:                  --- RBK0054I: Zweite Partition (Rootpartition) /dev/sdd2 wird formatiert.
mke2fs 1.44.1 (24-Mar-2018)
/dev/sdd2 is mounted; will not make a filesystem here!
[

Anbei das komplette log (musste das r von der extension abschneiden)

[raspiBackup.log](https://github.com/framps/raspiBackup/files/8876776/raspiBackup.log)
](url)

DarthWeber, 2022-06-10:
öööh - das ist nicht nur komsich, sondern sogar komisch:

ich habe jetzt mit windows eine exFat erstellt, eigentlich nur, um die Karte zu testen. Könnte ja kaputt sein, war aber brandneu.

kurz: Danach wieder in den Ubuntu und raspiBackup läuft, der Fehler von oben kommt jetzt nicht mehr. Muss /Soll man die sd immer vorher manuell partitionieren ?

DarthWeber, 2022-06-10:
Noch mal getestetd und siehe da: Wenn ich den kodi Dienst vorher stoppe, dann läuft es durch. Da wäre nur noch die Frage, ob diese -a und -o Option auch beim Restore greifen ?

Ansonsten ist das hier erst mal erldeigt.

closed 2022-06-10 by DarthWeber

framps, 2022-06-10:
> Möglicherweise liegt das an kodi, was auch darauf läuft Irgendwo gab es da eine Option, dass wechseldatenträger automatisch gemounted werden.

In Deiner vorherigen Antwort hast Du es ja bestaetigt :wink: 

> Da wäre nur noch die Frage, ob diese -a und -o Option auch beim Restore greifen ?

Nein. Die greifen nur fuer den Backup. Du musst manuell wenn Du restorest eben dafuer sorgen dass da nichts gemounted ist. Ansonsten kann keine Partitionierung von raspiBackup vorgenommen werden.