framps/raspiBackup #506: setting backup path for mounted cloud service

Bra1nsen, 2022-06-16:
thanks for sharing your work!

I mounted google drive /home/pi/mnt/gdrive          (how to: https://www.youtube.com/watch?v=Zotvv52k7lE)

Now I tried to set this as backup path..error : does not exist

What am I missing/doing wrong?

happy greetings
Paul

framps, 2022-06-16:
Would you please provide the debuglog?

framps, 2022-06-16:
Ach so. Du haengst am Installer ... Da ist das debuggen leider nicht leicht :cry: 

Aendere doch mal in /usr/local/etc/raspiBackup.conf die Zeile `DEFAULT_BACKUPPATH="/backup"`
so ab dass dort Dein gemounteter Pfad von Google steht und rufe dann mal raspiBackup auf. Das Debuglog steht dann in Deinem Homeverzeichnis. Da bekomme ich eine Menge Informationen und kann sicherlich weiterhelfen :smile:

framps, 2022-06-16:
> Ich glaub ich stell mich nur dumm an, sorry =)

No problem. Wie gesagt bist Du leading edge mit Deinem Googlemount ... und da ist es nicht einfach.

Bra1nsen, 2022-06-16:
[raspiBackup.log](https://github.com/framps/raspiBackup/files/8921084/raspiBackup.log)

framps, 2022-06-16:
Danke. Im Debuglog finde ich 
```
mkdir: cannot create directory ‘/home/@USER@/mnt/gdrive’: Permission denied
```
Warum der User `root` auf dem Mountpoint nichts anlegen darf weiss ich nicht. Normalerweise darf der User `root` in Linux ueberall zugreifen. Wie ich gesehen habe nutzt Du rclone zum Mounten. Das Tool ist mir bislang noch nicht untergekommen. Offensichtlich gibt es da besondere Security. Du musst rausfinden (vermutlich irgendeine Configsache) wie der user `root` auf dem google Drive Dateien anlegen kann.

Sprich: Du musst in der Lage sein folgenden Befehl ohne Fehler absetzen zu koennen:
```
sudo touch /home/@USER@/mnt/gdrive/dummy.txt
```

Solange das nicht moeglich ist kann raspiBackup dort auch kein Backup erstellen.

Bra1nsen, 2022-06-16:
Ok bin ein Stück weiter.. sudo touch geht jetzt.

neuer error

[raspiBackup.log](https://github.com/framps/raspiBackup/files/8922636/raspiBackup.log)

hab auch eine raspiBackup.msg datei ..

framps, 2022-06-17:
```
??? RBK0164E: Unable to create hardlinks. RC /home/@USER@/mnt/gdrive.
```
Die Meldung sagt die Ursache: Wenn Du rsync nutzen willst muessen Hardlinks unterstuetzt sein. Entweder schaffst Du es durch irgendwelche Konfiguration dass google drive die unterstuetzt und ansonsten musst Du auf den Backuptype tar wechseln :wink:  Das hat natuerlich den Nachteil dass Du keine Deltabackups erstellen kannst.

Bra1nsen, 2022-06-17:
hardslinks sind noch nicht möglich .. leider

hab auf tar eingestellt.

ist das so richtig/complete?

[image]
[image]

viele grüße

framps, 2022-06-17:
raspiBackup schreibt am Ende eine Meldung mit dem Completioncode. Da sich bei Influx waehrend des Backups was geaendert hat  musst Du noch vir dem Backup Influx stoppen.

Ps: Kannst Du kurz beschreiben was Du machen musstest damit root auf google drive schreiben kann?

Bra1nsen, 2022-06-17:
ich habe ein python skript das kontinuierlich Messdaten trackt (Hardware Sicherheit) - daran könnte es liegen?

rclone mount gdrive: $HOME/mnt/gdrive --allow-other

damit die flag auch funktioniert muss man unter  /etc/fuse.conf 'user_allow_other' entkommentieren

framps, 2022-06-17:
Danke fuer den Hinweis. Ich werde das auch mal demnaechst ausprobieren und dokumentieren. Wie gesagt kommen immer mal wieder Fragen wie man raspiBackup nutzen kann um Backups in der Cloud abzulegen.

> ich habe ein python skript das kontinuierlich Messdaten trackt (Hardware Sicherheit) - daran könnte es liegen?

tar meckert wenn sich waehren des Backups Dateien aendern und bricht ab. Siehe dazu auch [FAQ18](https://www.linux-tips-and-tricks.de/de/faq/#a18). Besser ist es immer vor dem Backup eine DB zu stoppen und anschliessend wieder zu starten um nich vielleicht inkonsistente Daten durch Verlust von Instoragedaten zu bekommen.

Solange Du nicht folgende Meldungen bekommst

```
--- RBK0017I: Backup finished successfully.
--- RBK0010I: asterix raspiBackup.sh V0.6.7 (22aa285) stopped at Fri 17 Jun 2022 05:08:28 AM CEST with rc 0.
```
ist Dein Backup nicht OK.

framps, 2022-06-17:
> Now I tried to set this as backup path..error : does not exist

Nachdem Du ein weiteres Log provided hast habe ich die Mountzeile gefunden:
```
20220616-234532 DBG 5825:              gdrive: on /home/@USER@/mnt/gdrive type fuse.rclone (rw,nosuid,nodev,relatime,user_id=1000,group_id=1000,allow_other)
2
```
Dieses Format erkennt der Installer nicht und weist es dementsprechend ab. Ich erstelle dazu mal einen Issue und fixe das :wink:

closed 2022-07-02 by github-actions[bot]