framps/raspiBackup #508: trotz Keepbackup werden keine Backups gelöscht

DarthWeber, 2022-06-18:
Hallo Framp,

laut log hab ich: (komplett im Anhang)
`
20220617-012352 DBG 2525:  --> logOptions Invocation options
20220617-012352 DBG 2527:      --- Options:  -t tgz --smartRecycle- /mnt/sshfs/GymWild/
...
20220617-012352 DBG 2553:      --- KEEPBACKUPS=10
20220617-012352 DBG 2554:      --- KEEPBACKUPS_DD=0
20220617-012352 DBG 2555:      --- KEEPBACKUPS_DDZ=3
20220617-012352 DBG 2556:      --- KEEPBACKUPS_TAR=0
20220617-012352 DBG 2557:      --- KEEPBACKUPS_TGZ=5`
...
20220617-012353 DBG 2580:      --- SMART_RECYCLE=0
20220617-012353 DBG 2581:      --- SMART_RECYCLE_DRYRUN=0
`
aber meine täglichen tgz sind nun schon bei 10, ich hätte jetzt max. 5 erwartet - oder habe ich etwas missverstanden ?

Grüße,
Tom

framps, 2022-06-18:
Das sollte natürlich nicht sein. Kannst Du Mal das Debuglog vom letzten Backuplauf zeigen?

DarthWeber, 2022-06-18:
Gerne. Aber welches meinst du? Die log hängt oben d an. Oder meinst du die msg?

framps, 2022-06-18:
Sorry. Habe ich uebersehen :face_with_spiral_eyes: 

Im Log finde ich

```
20220608-084034 DBG 3586:      --- DEFAULT_KEEPBACKUPS="10"
20220608-084034 DBG 3586:      --- DEFAULT_KEEPBACKUPS_DD=0
20220608-084034 DBG 3586:      --- DEFAULT_KEEPBACKUPS_DDZ=3
20220608-084034 DBG 3586:      --- DEFAULT_KEEPBACKUPS_TAR=0
20220608-084034 DBG 3586:      --- DEFAULT_KEEPBACKUPS_TGZ=5
20220608-084034 DBG 3586:      --- DEFAULT_KEEPBACKUPS_RSYNC=10
20220608-084034 DBG 3586:      --- DEFAULT_SMART_RECYCLE="1"
20220608-084034 DBG 3586:      --- DEFAULT_SMART_RECYCLE_DRYRUN="0"
20220608-084034 DBG 3586:      --- DEFAULT_SMART_RECYCLE_OPTIONS="7 4 12 1"
```
D.h. Du hast SmartRecycle eingeschaltet. Die Keepoptionen wirken aber nur wenn kein Smartercycle genutzt wird.

DarthWeber, 2022-06-18:
Versteh ich nicht. Das sind doch nur die defaults? Ich setze ja --smartRecycle- also explizit aus. Oder hab ich die logs durcheinander gekriegt? Hab gerade nur ein Smartphone zur Hand, ich schaue sonst morgen noch Mal genau nach

framps, 2022-06-18:
```
20220608-084034 DBG 2527:      --- Options:  -t tgz /mnt/sshfs/GymWild/
```
Das sind die Aufrufoptionen :wink: 

Ausserdem finde ich im Log 

```
20220608-104354 DBG 5512:      --> applyBackupStrategy 
20220608-104354 DBG 5519:          --- SR Parms: 7 4 12 1
20220608-104354 MSG 5524:          --- RBK0218I: Wende smarte Backupstrategie an. Täglich:7 Wöchentlich:4 Monatlich:12 Jährlich:1 
```
> Oder hab ich die logs durcheinander gekriegt?

Vermutlich schon.

DarthWeber, 2022-06-19:
Hi Framo, hier nun die passende log zum Problem. 

[raspiBackup.log](https://github.com/framps/raspiBackup/files/8935523/raspiBackup.log)

framps, 2022-06-19:
Danke fuer das Log. Da stimmt wirklich was nicht. 

Interessant ist die Meldung
```
20220619-030021 MSG 5570:          --- RBK0159I: 5 Backups werden für den Backuptyp tgz aufbewahrt. Bitte Geduld. 
```
aber man sieht dass noch 9 tgz Backups anschliessend stehen bleiben :cry: 

Ich sehe mir das heute Abend genauer an. Falls ich Dein Senario nachstellen muss dauert es aber ein wenig da ich gerade keinen Zugriff auf mein Testenvironment habe.

framps, 2022-06-19:
Habe den Fehler schon gefunden. Ist ein bloeder Typo :angry: Die gefixte Version steht [hier](https://raw.githubusercontent.com/framps/raspiBackup/master_508/raspiBackup.sh). Bitte die downloaden und den Fix verifizieren :wink:

framps, 2022-06-19:
Du kannst z.B. [dieses Script](https://github.com/framps/raspiBackup/blob/master/scripts/raspiBackupFromGitBranch.sh) nehmen um den Fix ohne lokale Installation aufzurufen und zu testen. 

Aufruf:

```
./raspiBackupFromGitBranch master_508 <raspiBackupOptionen>
```

DarthWeber, 2022-06-19:
zu spät - hab schon lokal installiert ;-)

jedenfalls prima, es scheint jetzt zu gehen - ich habe nun noch tgz Backups, wie es bei mir auch eingestellt ist. 

Danke !
DarthWeber

closed 2022-06-19 by DarthWeber

framps, 2022-06-19:
> zu spät - hab schon lokal installiert ;-)

Ok. Dann bist Du ein versierter RaspberryPi Nutzer :+1:  Es gibt viele Nutzer von raspiBackup die kommen da nicht so ganz klar mit. Deshalb habe ich vorhin noch eben das Script geschrieben. Werde ich zukuenftig immer bei Fixes anpreisen.

> es scheint jetzt zu gehen

Das habe ich erwartet. Ist wirklich ein bloeder Typo. Danke fuer die Verifikation. Ich werde den Fix gleich in den master mergen und publishen. D.h. wenn Du in ca 1 Stunde
`sudo raspiBackup.sh -U -S`
eingibst bekommst Du den offiziellen Fix :wink:

framps, 2022-06-19:
Fix: https://github.com/framps/raspiBackup/commit/860971ec56e5fc15a68f4f14aeca1f58722687c1