framps/raspiBackup #510: Rootpartitionscheck endet fehlerhaft mit RC 8.

DarthWeber, 2022-07-06:
Auf einem meiner Raspis läuft Raspian, allerdings:
`PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
`

da schimpft das script, beim Typ tgz, läuft aber mit --unsupportedEnvironment brav durch. Beim restore gibt es auch Gemecker:
??? RBK0183E: Rootpartitionscheck endet fehlerhaft mit RC 8.

was damit zusammen hängt  - oder auch nicht ?

Das img ist ein fertiges Image mit eben diesem Raspbian. Kann ich mich auf dieses Backup verlassen oder sollte ich etwas anders machen ?

framps, 2022-07-06:
> die zweite Partition ist clean.

OK. Du nutzt eine Disk - keine SD Karte. Die zweite Partition ist die die das Rootfilesystem enthaelt. 

Wenn der Check nun OK ist ist die Frage warum er als er von raspiBackup ausgefuehrt wurde nicht OK war :thinking: .

Kannst Du das Debuglog zur Verfuegung stellen vom Restore?

DarthWeber, 2022-07-06:
Nein, das ist keine disk sondern eine 64gb SD.

Das Log vom restore - magst du mir kurz sagen, wo das ist?

DarthWeber, 2022-07-07:
Danke, der Hinweis mit dem logr reichte schon - hab natürlich nur nach log gersucht. Ich habs hier reingepackt, musste es aber wieder in log umbenennen, da githib das sonst nicht mag.
[raspiBackup.log](https://github.com/framps/raspiBackup/files/9060150/raspiBackup.log)

DarthWeber, 2022-07-07:
Ui, ich glaiube es lag wieder daran, dass kodi auf dem System zwischendurch bzw. wenn die sd partitioniert wird immer versucht, die sd gleich zu mounten. Jedenfalls kann ich bei gestopptem kodi RC8 nicht reproduzieren. Es läuft im Prinzip durch, hat aber Probleme mit den UUIDs. Siehe Anhang.
[Uploading raspiBackup.log…]()

framps, 2022-07-07:
> Es läuft im Prinzip durch, hat aber Probleme mit den UUIDs. Siehe Anhang.
> Uploading raspiBackup.log…

Irgendwas ist beim Upload des Logs schiefgelaufen. Es ist keines da :wink:

DarthWeber, 2022-07-09:
[raspiBackup.log](https://github.com/framps/raspiBackup/files/9075855/raspiBackup.log)

framps, 2022-07-09:
Im Log finde ich nichts mit UUIDs. Ich finde dagegen

```
20220706-170635 MSG 5469:                  --- RBK0184I: Rootpartitionscheck gestartet. 
fsck from util-linux 2.31.1
/dev/sdd2 is mounted.
e2fsck: Cannot continue, aborting.
```

D.h. /dev/sdd2 wird waehrend des Restores wieder gemounted. Das darf nicht passieren.

DarthWeber, 2022-07-10:
Hab leider wieder das Umbenennen vor dem upload vergessen ... hier aber
[raspiBackup.log](https://github.com/framps/raspiBackup/files/9078463/raspiBackup.log)

framps, 2022-07-10:
Der Fehler kommt daher dass das die /etc/fstab sowie die /boot/cmdline.txt in Deinem efa Image nicht so aussieht wie bei dem originalen Buster :wink: 

Im Original werden PARTUUIDs genutzt und die /etc/fstab sieht wie folgt aus:

```
PARTUUID=738a4d67-01  /boot           vfat    defaults          0       2
PARTUUID=738a4d67-02  /               ext4    defaults,noatime  0       1
```

und die /boot/cmdline.txt

```
console=serial0,115200 console=tty1 root=PARTUUID=738a4d67-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait
```

Wie Du siehst wird immer die PARTUUID genommen um eine Partition zu identifizieren.

So sieht Deine /boot/cmdline.txt aus

```
20220710-064254 DBG 1929:          *** cat /etc/fstab
20220710-064254 DBG 1929:              proc /proc proc defaults 0 0
20220710-064254 DBG 1929:              /dev/mmcblk0p1 /boot vfat defaults 0 0
```

und so die /etc/fstab

```
20220710-064639 DBG 7834:                              *** cat /tmp/boot/cmdline.txt
20220710-064639 DBG 7834:                                  dwc_otg.lpm_enable=0 root=/dev/mmcblk0p2 rootfstype=ext4 rootwait rw splash logo.nologo loglevel=1 quiet vt.global_cursor_default=0 plymouth.ignore-serial-consoles
```

So wie ich es sehe haben die Kollegen vom efa nie richtig von Wheezy, bei dem diese Notation noch gebraeuchlich war, auf Buster umgestellt. 

Ich denke das restorte Image wird trotz der Meldung starten. Probiere es einfach aus. Allerdings wuerde ich Dir empfehlen die beiden Dateien auf dem Image mal auf PARTUUID Nutzung umzustellen.

closed 2022-07-25 by github-actions[bot]