framps/raspiBackup #511: Fehler bei Raspbian buster

DarthWeber, 2022-07-06:
Auf einem meiner Raspis läuft Raspian, allerdings:
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)" NAME="Raspbian GNU/Linux" VERSION_ID="10" VERSION="10 (buster)" VERSION_CODENAME=buster ID=raspbian ID_LIKE=debian HOME_URL="http://www.raspbian.org/" SUPPORT_URL="http://www.raspbian.org/RaspbianForums" BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

da schimpft das script, beim Typ tgz, läuft aber mit --unsupportedEnvironment brav durch. 

das log mit der Option --unsupportedEnvironment 
[raspiBackup.log](https://github.com/framps/raspiBackup/files/9057174/raspiBackup.log)

und das log ohne diese Option, wenn es dann fehlschlägt
[raspiBackup.log](https://github.com/framps/raspiBackup/files/9057192/raspiBackup.log)

framps, 2022-07-06:
Danke fuer den separaten issue :+1: 

Ich habe genau dasselbe System bei mir und muss keine unsupportedEnvironment Option angeben. Dem Log nach fehlt bei Dir /etc/rpi-issue. Hast Du die Datei versehentlich geloescht?

Bei mir steht da drin

```
cat /etc/rpi-issue 
Raspberry Pi reference 2020-02-13
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 5f884374b6ac6e155330c58caa1fb7249b8badf1, stage2
```

DarthWeber, 2022-07-06:
Nee, hab ich nicht. Es ist aber ein spezielles Image für eine Anwendung. Evt fehlt das da schon.

framps, 2022-07-06:
> Evt fehlt das da schon.

Was bekommst Du denn mit `sudo cat /etc/rpi-issue` ?

DarthWeber, 2022-07-07:
Naja, ein "Datei oder Verzeichnis nicht gefunden"

framps, 2022-07-07:
> Naja,

Ok. Ich habe 

> Evt fehlt das da schon.

so interpretiert dass Du es nicht kontrolliert hast sondern nur eine Vermutung ist :smiley: 

D.h. es ist kein Bug in raspiBackup. Wenn Du die Option `--unsupportedEnvironment` zukuenftig nicht mehr eingeben willst musst Du die fehlende Datei mit dem von mir o.g. Inhalt erstellen. 

Was ist das denn fuer ein spezielles Image? Ist schon merkwuerdig dass die Datei nicht existiert obwohl es ein RpiImage ist.

DarthWeber, 2022-07-09:
Hab den ersten Satz nicht verstanden, die Datei "rpi-issue" ist definitiv nicht da, ob die da sein muss, weiß ich nicht. Wer oder was erstellt diese Datei denn ? Klar kann ich die anlegen, aber mit der Option  "--unsupportedEnvironment" komme ich auch klar. Letztendlich ist doch nur wichtig, dass der restore funktioniert und dem scheint nach einem ersten Test so zu sein. Erkennst du anhand dieser issue-Datei, ob das OS gültig ist ?

Das Image kommt von [http://efa.nmichael.de/download.html.de](url)

framps, 2022-07-09:
> Hab den ersten Satz nicht verstanden

Wollte damit nur erklaeren warum ich nach der Ausgabe von `sudo cat /etc/rpi-issue` gefragt habe :wink: 

> ob die da sein muss, weiß ich nicht.

Soweit ich weiss existiert sie nur aus Dokumentationsgruenden von der RPi Foundation. (Siehe auch [hier](https://github.com/RPi-Distro/pi-gen/issues/126)). Wenn ich es richtig verstehe ist es der git Commit des Imagebuilds. Somit kannst Du die Datei loeschen.

Ich nutze diese Datei um festzustellen ob das Image offiziell von der RPi Foundation kommt. Wenn Du kein Problem mit der zusaetzlichen Option hast nutze sie weiter. Du bekommst dann zwar immer den Hinweis dass Dein System nicht supported wird aber funktional gibt es keine Einschraenkung. Wie schon geschrieben: Alternativ erstelle die Datei auf dem Image und alles ist wie sonst auf einem Buster.

Warum die Datei bei efa nicht existiert obwohl es ein Buster Image ist weiss ich nicht. Vielleicht ist das ein Buildproblem bei efa denn eigentlich sollte das Image doch ein original Buster sein mit der zusaetzlichen efa SW. Vermutlich muesste man dazu mal bei den efa Leuten nachfragen.

framps, 2022-07-09:
Ich habe eben ein paar DebugMessageUpdates vorgenommen um solche FalsePositives wie bei Dir im Debuglog erkennen zu koennen :wink: 

https://github.com/framps/raspiBackup/commit/f9d4978598cda7c08008f7892bd5d8956e229d4b

closed 2022-07-25 by github-actions[bot]