framps/raspiBackup #523: raspiBackup NFS und WrapperScript

bracewGitHub, 2022-07-20:
## Note
raspiBackup is supported on RaspberryOS as operating system and Raspberry HW only. There are environments out there which successfully run raspiBackup but any support requests on unsupported environments will be rejected. 

If you run raspiBackup on RaspberryOS and a Raspberry please execute all following actions first and provide all requested information:

* Check whether the error messages of raspiBackup were listed on the messages page and all suggested actions don't fixes the issue. [English](https://linux-tips-and-tricks.de/en/rmessages) - [German](https://linux-tips-and-tricks.de/de/fehlermeldungen)
* Use your preferred search engine and look up the error message number
* Attach raspiBackup debug log. Just drag and drop raspiBackup.log into the issue and don't copy and paste the contents of raspiBackup.log into the issue.

#### If any requested information was not provided the issue will be rejected.

## Note
Issues are closed automatically after 2 weeks if there is no activity any more

Hallo von Bracew

framps, 2022-07-20:
Perfekt. Lass uns jetzt hier weitermachen :wink: 

> Mach ich das gleiche über einen Cron-Job, wird zwar die Start E-Mail gesendet, aber eine Stop E-Mail nicht und das Backup enthält nicht raspiBackup.log.

D.h. aber dass ein neues Backup im Backupverzeichnis erstellt wird aber dort das Logfile fehlt?

bracewGitHub, 2022-07-21:
Hallo Framp, 

aufgrund dem Footer Deiner E-Mail:
"_...
habe ich per Mail geantwortet.

Ich habe keine Erfahrung mit Github. Ist das bei Dir angekommen, wird das hier dargestellt oder muss ich es hier einstellen?

bracewGitHub, 2022-07-21:
Hallo Framp,

vielen Dank für Deine Mühen mit mir.

Zunächst möchte ich gerne nochmals den Werdegang und die Situation zusammenfassen, bevor ich auf Deine Frage Antworte.

Ich habe mehrere RasPi. Es gibt einen Haupt-RasPi mit Namen Pi und mehrere andere RasPi (Satelliten-RasPi) mit Namen HeizPi, HuhnPi und EPi.

Seit ungefähr August 2020 habe ich raspiBackup zusätzlich zu Backups via Win32 Disk Imager genutzt, da sich die Backups so über Nacht und ohne Benutzerinteraktion, ohne Wege, ohne Gehäuse aufschrauben und ohne SD-Karten Transport sehr bequem von alleine erledigen. Früher hatte ich Backups via Win32 Disk Imager bei jeder Änderung gemacht, seit raspiBackup funktioniert mache ich Backups via Win32 Disk Imager nur noch sporadisch (Faulheit).
Die Satelliten-RasPi erstellen Backups via "/usr/local/bin/raspiBackupNfsWrapper.sh" per NFS auf eine am Pi angeschlossene USB-HDD. Der Pi erstellt seine Backups auf der USB-HDD mit "/usr/local/bin/raspiBackup.sh"

Im März 2022 hatten wir eine Diskussion auf: https://forum-raspberrypi.de/forum/thread/43483-raspibackup-sichere-deine-raspberry-regelmae%C3%9Fig-im-laufenden-betrieb/?postID=522794#post522794, dass die Start- Stoppdienste ggf. mit ";" statt mit "&&" verknüpft werden sollten.
Weiterhin hatten wir darüber diskutiert, ob ein Reboot nach dem Backup gut wäre: https://forum-raspberrypi.de/forum/thread/43483-raspibackup-sichere-deine-raspberry-regelmae%C3%9Fig-im-laufenden-betrieb/?postID=522812#post522812.

In der neuen Version von raspiBackup ist die Möglichkeit einen Reboot nach Backup zu machen eingebaut.

Zum härten meine RasPi habe ich die Einstellungen mit ";" anstatt "&&" in der "/usr/local/etc/raspiBackup.conf" sowie des Reboot nach Backup im "/usr/local/bin/raspiBackupNfsWrapper.sh" auf HeizPi und HuhnPi eingebaut. Den Aufruf im im "/usr/local/bin/raspiBackupNfsWrapper.sh"-Skript habe ich auf "raspiBackup.sh -g --timestamps --notifyStart --rebootSystem" umgestellt.

Die Optionen weil:

    -g um bei den Testaufrufen aus der bash den Fortgang beobachten zu können, da ansonsten der Eindruck entsteht die Kiste ist abgestürzt,
    --timestamps um bei den Testaufrufen aus der bash den zeitlichen Ablauf zu verstehen,
    --notifyStart um eine E-Mail zum Startzeitpunkt des Skripts zu sehen, falls es unbeobachtet in der Nacht läuft,
    --rebootSystem um die RasPi nach dem Backup wieder in einen definierte Zustand zu bekommen (Alle Dienste wieder auf Start).

Danach ist mir aufgefallen, dass von raspiBackup zum Teil Backups erstellt werden, diese aber keine Ende-des Backup E-Mail senden und die Anzahl der Backups größer als die Anzahl der zu vorhaltenden Backups (DEFAULT_KEEPBACKUPS="3") sind. Aufgrund eines Absturzes des HeizPi habe ich ein mit raspiBackup erstelltes Backup mit Restore zurück auf die SD-Karte getan. Jedoch waren raspiBackups danach nicht mehr möglich. Letztendlich haben wir vermutet, dass es an NFS liegen könnte und der Eintrag "hard" auf den Satelliten-RasPi zum Herstellen der NFS-Verbindung notwendig wäre.

Um wieder ein lauffähigen HeizPi zu bekommen habe ich eines der selteneren Backups via Win32 Disk Imager vom Februar 2022 wieder zurück auf die SD-Karte getan. Leider sind Änderungen vom Februar bis Juli somit nicht auf dem HeizPi. Ich habe versucht aus der Erinnerung die Änderungen nachzuholen.

Letztendlich ist der HeizPi soweit, dass er bei einem "händischen"-Backup, also dem Start aus der bash mit:

-     mount -t nfs -o hard 192.168.0.20:/media/USBHDD1/public/Backup /backup
-     raspiBackup.sh -g --timestamps --notifyStart --rebootSystem

ein Backup erstellt und die Start und Stopp E-Mails sendet.

Zunächst habe ich mir überlegt Dein "/usr/local/bin/raspiBackupNfsWrapper.sh"-Skript umzugestalten und eine Option **NFSOPTIONS**="-t nfs -o hard " einzubauen sodass der Mount in Deinem Skript zu mount **$NFSOPTIONS** $NFSSERVER:$NFSDIRECTORY $MOUNTPOINT mutiert. Hat funktioniert, dennoch waren die Backups nicht Ok.

Im weiteren habe ich zunächst zur Reduzierung der Fehlermöglichkeiten alles aus Deinem "/usr/local/bin/raspiBackupNfsWrapper.sh"-Skript rausgeworfen um die funktionierenden "händischen"-Backup Befehlen der bash einzusetzen. Also Skript wie folgt:

    #!/bin/bash
    sudo mount -t nfs -o hard 192.168.0.20:/media/USBHDD1/public/Backup /backup
    sudo raspiBackup.sh -g --timestamps --notifyStart --rebootSystem

Dieses Skript aus der bash aufgerufen funktioniert, es wird ein Backup erstellt und die Start und Stopp E-Mails kommen an.

Mach ich das gleiche über einen Cron-Job, wird zwar die Start E-Mail gesendet, aber eine Stopp E-Mail nicht und das Backup enthält nicht raspiBackup.log. Außerdem haben die mit Cron-Job erstellten Backups einen Datei-Zeitstempel des Verzeichnisse des letzten unvollständigen Backups obwohl im Dateinamen Datum und Zeit richtig ist.
Aus dem Cron-Job heraus funktioniert es also nicht. Siehe dazu auch: https://forum-raspberrypi.de/forum/thread/43483-raspibackup-sichere-deine-raspberry-regelmae%C3%9Fig-im-laufenden-betrieb/?postID=534572#post534572

Nun zu Deiner Frage, ob ein neues Backup im Backupverzeichnis erstellt wird aber dort das Logfile fehlt?
Die Antwort ist Ja, siehe den Absatz zu vor.

Im Moment ist die Situation auf dem Backup-Zielverzeichnis der USB-HDD des Pi wie folgt:

    b@pi:/Backup/HeizPi-Alt$ ls -l
    insgesamt 40
    drwxr-xr-x+ 22 root root 4096 Jun  7 17:19 HeizPi-rsync-backup-20220715-005703
[... cut ...]
    drwxr-xr-x+ 22 root root 4096 Jul 20 14:22 HeizPi-rsync-backup-20220720-134550
    drwxr-xr-x+ 22 root root 4096 Jul 20 15:31 HeizPi-rsync-backup-20220720-145400
    drwxr-xr-x+ 22 root root 4096 Jul 18 15:46 HeizPi-rsync-backup-20220720-154102
    drwxr-xr-x+ 22 root root 4096 Jul 18 15:46 HeizPi-rsync-backup-20220720-170803
    drwxr-xr-x+ 22 root root 4096 Jul 18 15:46 HeizPi-rsync-backup-20220720-182003
    drwxr-xr-x+ 22 root root 4096 Jul 18 15:46 HeizPi-rsync-backup-20220720-203503

Gruß Bracew

framps, 2022-07-21:
> vielen Dank für Deine Mühen mit mir.

Kein Problem. Das bekommen wir schon hin. Nur kann es etwas dauern denn ich werde nicht immer sofort auf Deine Antworten reagieren koennen :wink: 

> da sich die Backups so über Nacht und ohne Benutzerinteraktion, ohne Wege, ohne Gehäuse aufschrauben und ohne SD-Karten Transport sehr bequem von alleine erledigen.

Da mir das auch zu umstaendlich bei meinen Raspis war habe ich raspiBackup fuer mich geschrieben.

> Zunächst habe ich mir überlegt Dein "/usr/local/bin/raspiBackupNfsWrapper.sh"-Skript umzugestalten und eine Option **NFSOPTIONS**="-t nfs -o hard " einzubauen sodass der Mount in Deinem Skript zu [...]

Das kannst Du machen. Der eigentlich richtige Weg ist in der `/etc/fstab` einen Eintrag fuer den `/backup` mountpoint zu machen der `hard` enthaelt.

Aber lass uns da jetzt nichts aendern sondern erst einmal rausfinden warum der Backup von der Befehlszeile funktioniert und per Cron failed.

> Nun zu Deiner Frage, ob ein neues Backup im Backupverzeichnis erstellt wird aber dort das Logfile fehlt?
> Die Antwort ist Ja, siehe den Absatz zu vor.

raspiBackup sichert das Debuglog im Homeverzeichnis des aufrufenden Nutzers wenn es nicht im Backupverzeichnis gesichert werden kann. D.h. wenn Du es als User `pi` aufrufst wird es in /home/pi abgelegt. Wenn es per Cron aufgerufen wird wird es in `/root` abgelegt da cronjobs als root laufen.

D.h. ich braeuchte jetzt mal zwei Logs die Du per drag and drop hier anhaengen kannst:

1) Ein Log aus dem Backupverzeichnis wenn Du den Backup manuell anstoesst und er erfolgreich ist
2) Ein Log aus dem Verzeichnis `/root` welches nach einem Cronaufruf bei dem raspiBackup nicht erfolgreich ist abgelegt wird. 

Du musst nicht unbedingt den StandardCronaufruf von raspiBackup abwarten. Ich wuerde z.B. in `/etc/cron.d/raspiBackup` die Zeit mal kurz so aendern dass der Cronjob in 5 Minuten startet. Wenn der Job dann geendet hat hast Du das Log (2) und solltest die Zeit dann wieder so einstellen wie es sein soll. Anschliessend rufst Du das Wrapperscript noch einmal als Nutzer `pi` auf und hast dann das Log (1) :smile:

bracewGitHub, 2022-07-21:
Vor dem Start: 

ein Blick auf die vorhandenen Log-Dateien. Auf dem HeizPi habe ich die letzten Logs je einmal im:

/home/HeizPi/ Verzeichnis: 
-rw-r--r--  1 HeizPi HeizPi  778465 Jul 20 09:25 raspiBackup.log

und einmal im /root Verzeichnis:
-rw-r--r-- 1 root root 54927 Jul 20 07:05 raspiBackup.log

Ich editiere Cron-Jobs gerne mit Hilfe von Webmin [.](url https://www.webmin.com/) Ich komme halt von Windows 1.0 (ca. Mitte der 1980er Jahre) bis Windows 10 (heute) zu Linux. Die Maus ist gewohnt. Mit Webmin ist das Ändern des Cron-Jobs ganz easy.

Trotzdem hier Auszug aus "sudo crontab –e" des HeizPi:
28 20 * * 4 /bin/bash --login /usr/local/bin/raspiBackupNfsWrapper.sh #Backup via NFS auf Pi

Zu Log (2):

Ich habe mit Webmin den Cron-Job mit kurzem Zeitverlauf geändert und ein raspiBackup via Cron gestartet:
•	die E-Mail „HeizPi: Backup gestarted“ ist angekommen, 
07-21-2022 20:28:09 --- RBK0009I: HeizPi: raspiBackup.sh V0.6.7 (2022-06-19/7e37673) Do 21. Jul 20:28:09 CEST 2022 gestartet.
•	das Backup „/Backup/HeizPi/HeizPi-rsync-backup-20220721-202803/“ wurde ohne Log Datei im Backup-Verzeichnis erstellt,
•	es ist keine Ende des Backup E-Mail angekommen,
•	der HeizPi wurde neu gestartet.

Seltsam ist, dass es keine Veränderungen an den am Anfang genannten Log Dateien gibt. Beide des oben genannten Zeitstempel 
/home/HeizPi/ Verzeichnis: 
-rw-r--r--  1 HeizPi HeizPi  778465 Jul 20 09:25 raspiBackup.log

und im /root Verzeichnis:
-rw-r--r-- 1 root root 54927 Jul 20 07:05 raspiBackup.log

Ich kann dir also keine aktuelles Log-2 aus dem /root Verzeichnis anhängen, es wäre nur das alte da.

Zu Log (1):

Nach dem reboot zu vor, habe ich in der bash:
•	sudo /usr/local/bin/raspiBackupNfsWrapper.sh
gestartet 
mit folgendem Inhalt:
•	#!/bin/bash
•	sudo mount -t nfs -o hard 192.168.0.20:/media/USBHDD1/public/Backup /backup
•	sudo raspiBackup.sh -g --timestamps --notifyStart --rebootSystem

Nach dem Start:
•	die E-Mail „HeizPi: Backup gestarted“ ist angekommen 
07-21-2022 21:18:21 --- RBK0009I: HeizPi: raspiBackup.sh V0.6.7 (2022-06-19/7e37673) Do 21. Jul 21:18:20 CEST 2022 gestartet.
•	das Backup „„/Backup/HeizPi/HeizPi-rsync-backup-20220721-211814“ wurde mit Log Datei im Backup-Verzeichnis erstellt,
•	die E-Mail „HeizPi: Backup erfolgreich beendet“ ist angekommen,
07-21-2022 21:18:21 --- RBK0009I: HeizPi: raspiBackup.sh V0.6.7 (2022-06-19/7e37673) Do 21. Jul 21:18:20 CEST 2022 gestartet.
07-21-2022 21:18:28 --- RBK0031I: Prüfe ob eine neue Version von raspiBackup.sh verfügbar ist.
07-21-2022 21:18:35 --- RBK0151I: Backuppfad /backup mit Partitionstyp nfs4 wird benutzt.
07-21-2022 21:18:36 --- RBK0272I: Das System wird am Ende des Backuplaufes neu gestartet.
07-21-2022 21:18:41 --- RBK0044I: Backup der Bootpartition wird in /backup/HeizPi/HeizPi-rsync-backup-20220721-211814/HeizPi-backup.img erstellt.
07-21-2022 21:18:58 --- RBK0158I: rsync Backup "/backup/HeizPi/HeizPi-rsync-backup-20220721-211814" wird erstellt.
07-21-2022 21:18:59 --- RBK0085I: Backuperstellung vom Typ rsync gestartet. Bitte Geduld.
07-21-2022 21:53:33 --- RBK0078I: Backupzeit: 00:34:52.
07-21-2022 21:54:38 --- RBK0033I: Bitte warten bis aufgeräumt wurde.
07-21-2022 21:54:43 --- RBK0159I: 3 Backups werden für den Backuptyp rsync aufbewahrt. Bitte Geduld.
07-21-2022 21:54:43 --- RBK0017I: Backup erfolgreich beendet.
07-21-2022 21:54:44 --- RBK0010I: HeizPi: raspiBackup.sh V0.6.7 (7e37673) Do 21. Jul 21:54:44 CEST 2022 beendet mit Returncode 0.
•	der HeizPi wurde neu gestartet.
•	Im Backup-Ordner auf dem Pi liegt:
Backup/HeizPi/HeizPi-rsync-backup-20220721-211814/raspiBackup.log
welche ich Dir in der Anlage beifüge.

Auf dem Pi sind im Backup-Verzeichnis jetzt 12 Backups enthalten.
drwxr-xr-x+ 22 root root 4096 Jul 18 15:46 HeizPi-rsync-backup-20220720-094003
drwxr-xr-x+ 22 root root 4096 Jul 20 12:06 HeizPi-rsync-backup-20220720-112828
drwxr-xr-x+ 22 root root 4096 Jul 20 12:59 HeizPi-rsync-backup-20220720-122150
drwxr-xr-x+ 22 root root 4096 Jul 18 15:46 HeizPi-rsync-backup-20220720-130502
drwxr-xr-x+ 22 root root 4096 Jul 20 14:22 HeizPi-rsync-backup-20220720-134550
drwxr-xr-x+ 22 root root 4096 Jul 20 15:31 HeizPi-rsync-backup-20220720-145400
drwxr-xr-x+ 22 root root 4096 Jul 18 15:46 HeizPi-rsync-backup-20220720-154102
drwxr-xr-x+ 22 root root 4096 Jul 18 15:46 HeizPi-rsync-backup-20220720-170803
drwxr-xr-x+ 22 root root 4096 Jul 18 15:46 HeizPi-rsync-backup-20220720-182003
drwxr-xr-x+ 22 root root 4096 Jul 18 15:46 HeizPi-rsync-backup-20220720-203503
drwxr-xr-x+ 22 root root 4096 Jul 18 15:46 HeizPi-rsync-backup-20220721-202803
drwxr-xr-x+ 22 root root 4096 Jul 21 21:54 HeizPi-rsync-backup-20220721-211814

[raspiBackup.log](https://github.com/framps/raspiBackup/files/9162456/raspiBackup.log)

framps, 2022-07-22:
Merkwueridg dass 12 Backups existieren obwohl Du nur 3 Backups vorhalten willst :thinking: 

Daraufhin habe ich das Log angesehen und finde

```
20220721-215442 DBG 5570:              *** ls -d /backup/@HOSTNAME@/*
20220721-215442 DBG 5570:                  ls: Zugriff auf '/backup/@HOSTNAME@/*' nicht möglich: Datei oder Verzeichnis nicht gefunden
```

Das ist der Grund warum du 12 Backups hast :cry: . Dummerweise wird in raspiBackup dieser Fehler nicht entdeckt und raspiBackup endet normal. Um das zu fixen habe ich bereits Issue #520 erstellt. Ich bin dabei das genauer zu untersuchen und einen Fix fuer Dich zu bauen.

`/backup` ist wie folgt gemounted:
```
20220721-215335 DBG 5209:                          192.168.0.20:/media/USBHDD1/public/Backup on /backup type nfs4 (rw,relatime,vers=4.1,rsize=131072,wsize=131072,namlen=255,hard,proto=tcp,timeo=600,retrans=2,sec=sys,clientaddr=192.168.0.30,local_lock=none,addr=192.168.0.20)
```
Lass uns das zur Kenntnis nehmen und spaeter nach der Ursache suchen. 

> Ich habe mit Webmin den Cron-Job mit kurzem Zeitverlauf geändert und ein raspiBackup via Cron gestartet:
> • die E-Mail „HeizPi: Backup gestarted“ ist angekommen,
> 07-21-2022 20:28:09 --- RBK0009I: [...]

Das sieht alles von Dir richtig konfiguriert und durchgefuehrt aus :+1: . Dass kein neues Log erstellt wurde ist schon merkwuerdig :thinking: 

> Trotzdem hier Auszug aus "sudo crontab –e" des HeizPi: 28 20 * * 4 /bin/bash --login /usr/local/bin/raspiBackupNfsWrapper.sh #Backup via NFS auf Pi

Das ist eine wichtige Info. Gut dass Du es angegeben hast :+1: . Ich muss gestehen dass ich nicht in vollem Umfang verstehe was die Zeile mit `/binbash --login` genau bewirkt. Das muesste ich testen und genauer untersuchen.

Schreibe da doch bitte mal folgendes rein

```
28 20 * * 4 /usr/local/bin/raspiBackupNfsWrapper.sh #Backup via NFS auf Pi
```
mit entsprechend neuem Startzeitpunkt :wink:

Ich denke deshalb finden wir das Log nicht.

bracewGitHub, 2022-07-22:
Die Zeile mit `/bin/bash --login`  habe ich, siehe: 
`https://forum-raspberrypi.de/forum/thread/43483-raspibackup-sichere-deine-raspberry-regelmae%C3%9Fig-im-laufenden-betrieb/?postID=534572#post534572`
in 
`https://wiki.ubuntuusers.de/Cron/ `
gelesen und am letzten Mittwoch ausprobiert. Vorher war die so wie Du sie angegeben hast:
`59 0 * * 3 /usr/local/bin/raspiBackupNfsWrapper.sh #Backup via NFS auf Pi`

Ich habe (wieder mit Webmin) cron geändert. Auch `path `und `bash `habe ich aus cron entfernt. Der crontab sieht nun in gänze wie folgt aus:
```
sudo crontab -l
5 0,6,12,18 * * * /home/HeizPi/tankmessung.sh #Ultraschallmessung in Datei schreiben und an Pi übertragen
13,33,53 * * * * /home/HeizPi/tempmessung.sh #Temperaturmessung DHT22 Messfühler in Datei schreiben und an Pi übertragen
34 13 * * * /etc/webmin/package-updates/update.pl
50 20 * * 5 /usr/local/bin/raspiBackupNfsWrapper.sh #Backup via NFS auf Pi
```
und Backup sollte gleich starten.

• die E-Mail „HeizPi: Backup gestarted“ ist angekommen
`07-22-2022 20:50:09 --- RBK0009I: HeizPi: raspiBackup.sh V0.6.7 (2022-06-19/7e37673) Fr 22. Jul 20:50:08 CEST 2022 gestartet.`
• das Backup „HeizPi-rsync-backup-20220722-205002“ wurde ohne Log Datei im Backup-Verzeichnis erstellt,
• es ist keine Ende des Backup E-Mail angekommen,
• der HeizPi wurde neu gestartet.
• es gibt wieder keine Veränderungen an den zuvor genannten Log Dateien. 
in /home/HeizPi `-rw-r--r--  1 HeizPi HeizPi  778465 Jul 20 09:25 raspiBackup.log` und 
in /root `-rw-r--r-- 1 root root 54927 Jul 20 07:05 raspiBackup.log` sind beide Logs wie vorher.

framps, 2022-07-22:
Dein Problem ist echt eine harte Nuss ...

Bevor ich mit dem Holzhammer komme der ein bash Debug Log generiert was sehr umfangreich ist und dementsprechend auch fuer mich schwer zu lesen ist:

Aendere die folgende Zeile 

> 0 20 * * 5 /usr/local/bin/raspiBackupNfsWrapper.sh #Backup via NFS auf Pi

in

`0 20 * * 5 /usr/local/bin/raspiBackupNfsWrapper.sh &>/tmp/raspiBackup_bracew.log #Backup via NFS auf Pi`

Vielleicht zeigst Du auch mal den aktuellen Inhalt des Wrapperscripts.

bracewGitHub, 2022-07-22:
Ok,
`20 23 * * 5 /usr/local/bin/raspiBackupNfsWrapper.sh &>/tmp/raspiBackup_bracew.log #Backup via NFS auf Pi`
läuft gleich. Log kommt.

bracewGitHub, 2022-07-22:
• die E-Mail „HeizPi: Backup gestarted“ ist angekommen
07-22-2022 23:20:17 --- RBK0009I: HeizPi: raspiBackup.sh V0.6.7 (2022-06-19/7e37673) Fr 22. Jul 23:20:15 CEST 2022 gestartet.
• das Backup „HeizPi-rsync-backup-20220722-232002“ wurde ohne Log Datei im Backup-Verzeichnis erstellt,
• es ist keine Ende des Backup E-Mail angekommen,
• der HeizPi wurde neu gestartet.
• Kein Log! Nicht in / und auch nicht in /tmp

franjo-G, 2022-07-22:
Ich habe mir mal das o.ben verlinkte Log angesehen.
Da werden 6 Services gestoppt, aber 12 geatartet, davon 6 x "systemctl start" (Nach jedem Service-Start folgt einmal nur "systemctl start" ????  
Wie kommen die da rein?
Sind die manuell in die raspiBackup.conf eingetragen worden?

#DBG 3588:      --- DEFAULT_STOPSERVICES="systemctl stop cpufrequtils ; systemctl stop loadcpufreq ; systemctl stop lm-sensors ; systemctl stop webmin ; systemctl stop nmbd ; systemctl stop smbd ; systemctl stop cron"

#DBG 3588:      --- DEFAULT_STARTSERVICES="systemctl start cron ; systemctl start ; systemctl start smbd ; systemctl start ; systemctl start nmbd ; systemctl start ; systemctl start webmin ; systemctl start ; systemctl start lm-sensors ; systemctl start ; systemctl start loadcpufreq ; systemctl start ; systemctl start cpufrequtils"

Kann das Probleme machen?
Wird beim "Start" von systemctl die NFS-Verbindung.......... getrennt? Das würde evtl. die Nichterreichbarkeit des Backup-Verzeichnisses sowie des fehlenden mailversands  erklären,

bracewGitHub, 2022-07-22:
Ok, ich habe in `/usr/local/etc/raspiBackup.conf` wie folgt geändert:

```
# Durch && getrennte Befehle, die vor dem Starten des Backups auszufÃ¼hren sind
DEFAULT_STOPSERVICES="systemctl stop cpufrequtils ; systemctl stop loadcpufreq ; systemctl stop lm-sensors ; systemctl stop webmin ; systemctl stop nmbd ; systemctl stop smbd ; systemctl stop cron"
# Durch && getrennte Befehle, die nach dem Starten des Backups auszufÃ¼hren sind
DEFAULT_STARTSERVICES="systemctl start cron ; systemctl start smbd ; systemctl start nmbd ; systemctl start webmin ; systemctl start lm-sensors ; systemctl start loadcpufreq ; systemctl start cpufrequtils"
```
und lasse via cron ein neues Backup machen.

franjo-G, 2022-07-22:
> Durch && getrennte Befehle,

Bei mir sieht die .conf folgendermaßen aus (automatisch erstellt duch raspiBackupInstallUI"
DEFAULT_STARTSERVICES="systemctl start ssh && systemctl start rpi-eeprom-update && systemctl start rc-local.............

bracewGitHub, 2022-07-22:
Wir hatten schon besprochen, alle `&&` durch `;` zu ersetzen, siehe:
`https://forum-raspberrypi.de/forum/thread/43483-raspibackup-sichere-deine-raspberry-regelmae%C3%9Fig-im-laufenden-betrieb/?postID=522918#post522918`, da danch sowieso ein reboot gemacht wird und die Startreihenfolge nicht maßgeblich ist.

franjo-G, 2022-07-22:
Aber vielleicht hättest du mit && eine Warnmeldung erhalten...

bracewGitHub, 2022-07-22:
• die E-Mail „HeizPi: Backup gestarted“ ist angekommen
07-23-2022 00:28:09 --- RBK0009I: HeizPi: raspiBackup.sh V0.6.7 (2022-06-19/7e37673) Sa 23. Jul 00:28:09 CEST 2022 gestartet.
• das Backup „HeizPi-rsync-backup-20220723-002802“ wurde ohne Log Datei im Backup-Verzeichnis erstellt,
• es ist keine Ende des Backup E-Mail angekommen,
• der HeizPi wurde neu gestartet.
• Kein Log! Nicht in / und auch nicht in /tmp

Ok, am Vormittag dann ein Versuch mit `&&`.

Kann es sein, dass in /tmp beim Neustart aufgeräumt wird?

bracewGitHub, 2022-07-23:
Guten Tag,

ich habe heute Morgen auf dem HeizPi mit `sudo raspiBackupInstallUI.sh` die Konfiguration geändert. Alle Start/Stopp-Service habe ich entfernt, ausser für cron. Sieht auszugsweise in `/usr/local/etc/raspiBackup.conf` jetzt so aus:
```
# Durch && getrennte Befehle, die vor dem Starten des Backups auszufÃ¼hren sind
DEFAULT_STOPSERVICES="systemctl stop cron"
# Durch && getrennte Befehle, die nach dem Starten des Backups auszufÃ¼hren sind
DEFAULT_STARTSERVICES="systemctl start cron"
```
also weder `;` noch `&&`.
 
Den crontab habe ich wie folgt geändert:
`30 9 * * 6 /usr/local/bin/raspiBackupNfsWrapper.sh &>/raspiBackup_bracew.log #Backup via NFS auf Pi`
damit das Log nicht mehr in /tmp sondern in / gespeichert wird.

Um 9:30 ist raspiBackup aus cron gesteuert gestartet:
• die E-Mail „HeizPi: Backup gestarted“ ist angekommen
07-23-2022 09:30:09 --- RBK0009I: HeizPi: raspiBackup.sh V0.6.7 (2022-06-19/7e37673) Sa 23. Jul 09:30:08 CEST 2022 gestartet.
• es ist keine Ende des Backup E-Mail angekommen,
• der HeizPi wurde neu gestartet.
• es gibt sowohl ein raspiBackup_bracew.log in / des HeizPi sowohl als auch im Backup-Ordner des Pi, aber **beide LEER**.

Auf dem HeizPi in /:
-rw-r--r--   1 root root     **0** Jul 23 09:30 raspiBackup_bracew.log

Auf dem Pi im Backup-Ordner `HeizPi-rsync-backup-20220723-093002`
-rw-r--r--+   1 root root        **0** Jul 23 09:30 raspiBackup_bracew.log

framps, 2022-07-23:
@franjo-G Good catch mit den Startservices. Da habe ich zugegebenermassen nicht drauf geachtet. Das kann aber auch kaum die Ursache fuer kein Log sein. Anyhow ist es keine gute Idee die `&&` durch `;`zu ersetzen.

> Vielleicht zeigst Du auch mal den aktuellen Inhalt des Wrapperscripts.

Wo ist es :question: 

> aber **beide LEER**.

Ok. Dann nehmen wir jetzt mal den Holzhammer :smiling_face_with_tear: 

Setze mal folgendes mit entsprechendem geaenderten Zeitpunkt in Deine crontab:
```
28 0 * * 6 /bin/bash -x /usr/local/bin/raspiBackupNfsWrapper.sh &>/root/raspiBackup_bracew.log 
```
Danach muss ein `/root/raspiBackup_bracew.log` existieren. Aber Vorsicht: Es koennte sein dass sich darin sensitive Infos finden. Wenn Du Bedenken hast das hier zu posten schicke es mir per eMail zu. Ich sehe dann u.U. auch die sensitiven Infos - aber es ist was anderes ob ich alleine sie sehe - oder die ganze Welt ... 

Bzgl. raspiBackup.log brauchst Du Dir keine Sorgen deswegen zu machen. Ich habe extra einen Schritt eingebaut der sensitive Infos maskiert. Nur greift die Maskierung nicht bei der Holzhammermethode.

franjo-G, 2022-07-23:
> Das kann aber auch kaum die Ursache fuer kein Log sein. Anyhow ist es keine gute Idee die `&&` durch `;`zu ersetzen.

Das ist schon richtig. Allerdings sehe ich das Problem eher in der Konfiguration als in raspiBackup. (Wie z.B. die Einträge in den zu startenden services)
Ich nutze raspiBackup ja auch in den verschiedensten Variationen. Manuell, per Cron jeden Sonntag..... Allerdings nicht mit dem Wrapperscript. 
Manuelle Backuos .... erstelle ich mittlerweile ausschliesslich mit meinem Helper-Script. Ich nutze auch keine Optionen wie restart, Timestamp oder ähnliches.
Allerdings wird die "Ende"-email ja erst nach starten der Services gesendet.. Was ist denn, wenn einige Services nicht gestartet werden, Cron diese Fehler (im Gegensatz zu raspiBackup) bemerkt und den gesamten Vorgang abbricht? 
Ich persönklich bekomme dann immer noch eine Mail vom Cron-Daemon, wie letztens bei den ACL's.
Die Fehler wurden ja auch nur von Cron realisiert und nicht von raspiBackup.

framps, 2022-07-23:
> Allerdings sehe ich das Problem eher in der Konfiguration als in raspiBackup.

Vermutlich hast Du Recht. Aber dass kein Log erstellt wird ist eine Sache die ich verstehen und fixen muss. Denn ohne Log ist kein Support moeglich.

> mit dem Wrapperscript

Da habe ich schon zwei Mal nach gefragt wie das aussieht. Bislang leider keine Antwort.

> Ich persönklich bekomme dann immer noch eine Mail vom Cron-Daemon, wie letztens bei den ACL's.

Wenn keine eMail MTA konfiguriert sind bekomt man leider nix. Deshalb habe ich ja den Cronoutput auf /root redirected.

Die Wahrheit werde ich mit der Holzhammermethode sehen. Allerdings muss ich dann vorher tief Luft holen bevor ich da reinsehe denn das Log ist extrem schwer zu lesen und auch elendlich lang :disappointed:

franjo-G, 2022-07-23:
> Die Wahrheit werde ich mit der Holzhammermethode sehen. Allerdings muss ich dann vorher tief Luft holen bevor ich da reinsehe denn das Log ist extrem schwer zu lesen und auch elendlich lang

Das ist dann wohl dein Problem :-)
Wenn es für dich schon schwer zu lesen ist, dann ist es für andere wie auch mich unmöglich zu lesen :-)

> Wenn keine eMail MTA konfiguriert sind bekomt man leider nix. Deshalb habe ich ja den Cronoutput auf /root redirected.
Ich habe auch nur msmtp installiert, (die config einmal im User-verzeichnis und einmal unter /etc) und unter /etc/aliases eine Mailadresse eingegeben. Ansonsten habe ich mir da keine Gedanken drüber gemacht.  Auf jeden Fall funktioniert das einwandfrei.

framps, 2022-07-23:
> Wenn es für dich schon schwer zu lesen ist, dann ist es für andere wie auch mich unmöglich zu lesen :-)

raspiBackup.log kann man durchaus auch als Nutzer von raspiBackup lesen. Ich bekam schon verschiedentlich Bug Reports wo mich ein Nutzer anhand des Debuglogs auf die Ursache hingewiesen hat. 

Die Korrelation der Zeilennummern im Debuglog gelingt wenn man sich parallel den entsprechenden commit Code Stand (der wird in der Welcomemessage ausgegeben) ansieht :wink:

bracewGitHub, 2022-07-23:
>Vielleicht zeigst Du auch mal den aktuellen Inhalt des Wrapperscripts.
>mit dem Wrapperscript
>Da habe ich schon zwei Mal nach gefragt wie das aussieht. Bislang leider keine Antwort.

Immer mit der Ruhe, das hatte ich doch schon, in meinem 3. Post hier, siehe zuvor und hier nochmals:
>Im weiteren habe ich zunächst zur Reduzierung der Fehlermöglichkeiten alles aus Deinem "/usr/local/bin/raspiBackupNfsWrapper.sh"-Skript rausgeworfen um die funktionierenden "händischen"-Backup Befehlen der bash einzusetzen. **Also Skript wie folgt**:
```
#!/bin/bash
sudo mount -t nfs -o hard 192.168.0.20:/media/USBHDD1/public/Backup /backup
sudo raspiBackup.sh -g --timestamps --notifyStart --rebootSystem
```

>Loesche ein vielleicht bereits existierendes /root/raspiBackup_bracew.log erst einmal.

Ok. Ich habe dass auf dem HeizPi in / gelöscht. in /root war keines.

Ich lasse via cron ein neues Backup machen mit cron-Eintrag
`55 22 * * 6 /bin/bash -x /usr/local/bin/raspiBackupNfsWrapper.sh &>>/root/raspiBackup_bracew.log`

Um 22:55 ist raspiBackup aus cron gesteuert gestartet:
• die E-Mail „HeizPi: Backup gestarted“ ist angekommen
07-23-2022 22:55:10 --- RBK0009I: HeizPi: raspiBackup.sh V0.6.7 (2022-06-19/7e37673) Sa 23. Jul 22:55:09 CEST 2022 gestartet.
• es ist keine Ende des Backup E-Mail angekommen,
• der HeizPi wurde neu gestartet.
• es gibt ein **raspiBackup_bracew.log in /root des HeizPi aber LEER** und keines im Backup-Ordner des Pi.
-rw-r--r-- 1 root root    0 Jul 23 22:55 raspiBackup_bracew.log

framps, 2022-07-24:
> Immer mit der Ruhe, das hatte ich doch schon, in meinem 3. Post hier, siehe zuvor und hier nochmals:

Leider kommt es immer sehr genau drauf an wie etwas codiert wurde. Ein Komma an der falschen Stelle oder zuviel kann schon die gesammte Logik aendern. Deshalb moechte ich keine Beschreibung was man getan hat sondern den Code in einer Datei per drag and drop angefuegt sehen :wink: 

Ich habe heute mal Dein Szenario bei mir nachgestellt. Also Wrapperscript mit dem Mount und raspiBackup Aufruf. Es funktioniert bei mir sowohl per cmdline als auch per cronjob. Ergo liegt irgendein Environmentproblem bei Dir vor :thinking: 

Dann habe ich festgestellt dass auch bei mir ein leeres Log erstellt wird wenn ich den Holzhammer nehme. Ich hatte da einen kleinen Denkfehler :cry: Nachdem ich den Code wie folgt geaendert habe bekam ich das Holzhammerlog in `/root/rp.log`:

/etc/crontab.d/raspiBackup
```
30 18 * * *	root	/usr/local/bin/raspiBackupNfsWrapper.sh
```
/usr/local/bin/raspiBackupNfsWrapper.sh
```
#!/bin/bash
rm -rf /root/rp.log
mount -t nfs -o hard 192.168.0.9:/volume1/raspiBackup /backup &>>/root/rpi.log
/bin/bash -x /usr/local/bin/raspiBackup.sh --timestamps --notifyStart --rebootSystem &>>/root/rp.log 
```
PS: Ich habe die Option -G rausgenommen da sie das Debuglog nur unnoetiog aufblaeht. Ausserdem habe ich einen Issue erstellt dass zukuenftig wenn raspiBackup im Background laeuft die Option -G ignoriert wird.

bracewGitHub, 2022-07-24:
Entschuldigung, dass mit dem Code in einer Datei habe ich dann wohl falsch verstanden.
In der Anlage also `/usr/local/bin/raspiBackupNfsWrapper.sh`
Meldung von GitHub: We don't support that file type.
Ich habe die Datei umbenannt in raspiBackupNfsWrapper.sh.txt
[raspiBackupNfsWrapper.sh.txt](https://github.com/framps/raspiBackup/files/9176274/raspiBackupNfsWrapper.sh.txt)

Ist es egal, ob `sudo nano /etc/crontab.d/raspiBackup` oder `sudo crontab -e` ?

`/volume1/raspiBackup` muss ich auch anpssen, oder?

framps, 2022-07-24:
> Entschuldigung, dass mit dem Code in einer Datei habe ich dann wohl falsch verstanden.

Kein Problem. Deshalb habe ich es ja noch mal erklaert :smile: 

> Ich habe die Datei umbenannt in raspiBackupNfsWrapper.sh.txt

Jetzt brauche ich sie nicht mehr. Ich habe es ja durchaus anhand Deiner Beschreibung verstanden und beim Reproduzieren dann gemerkt was nicht passt.

> Ist es egal, ob `sudo nano /etc/crontab.d/raspiBackup` oder `sudo crontab -e` ?

Letzteres ist besser da noch geprueft wird ob alles syntaktisch OK ist. Ich nutze den ersten Weg :rofl: 

> `/volume1/raspiBackup` muss ich auch anpassen, oder?

Du musst alles an Deine lokalen Gegebenheiten anpassen. So sieht mein lokales Environment aus.

bracewGitHub, 2022-07-24:
Um 19:45 ist raspiBackup aus cron gesteuert gestartet:
• die E-Mail „HeizPi: Backup gestarted“ ist angekommen
`07-24-2022 19:45:12 --- RBK0009I: HeizPi: raspiBackup.sh V0.6.7 (2022-06-19/7e37673) So 24. Jul 19:45:11 CEST 2022 gestartet.`
• die E-Mail „HeizPi: Backup nicht erfolgreich !!!.“ ist angekommen
```
07-24-2022 19:45:12 --- RBK0009I: HeizPi: raspiBackup.sh V0.6.7 (2022-06-19/7e37673) So 24. Jul 19:45:11 CEST 2022 gestartet.
07-24-2022 19:45:23 --- RBK0031I: Prüfe ob eine neue Version von raspiBackup.sh verfügbar ist.
07-24-2022 19:45:32 --- RBK0151I: Backuppfad /backup mit Partitionstyp nfs4 wird benutzt.
07-24-2022 19:45:33 --- RBK0272I: Das System wird am Ende des Backuplaufes neu gestartet.
07-24-2022 19:45:39 --- RBK0044I: Backup der Bootpartition wird in /backup/HeizPi/HeizPi-rsync-backup-20220724-194502/HeizPi-backup.img erstellt.
07-24-2022 19:45:55 --- RBK0158I: rsync Backup "/backup/HeizPi/HeizPi-rsync-backup-20220724-194502" wird erstellt.
07-24-2022 19:45:56 --- RBK0085I: Backuperstellung vom Typ rsync gestartet. Bitte Geduld.
07-24-2022 20:22:33 ??? RBK0021E: Backupprogramm des Typs rsync beendete sich mit RC 24.
07-24-2022 20:22:34 --- RBK0033I: Bitte warten bis aufgeräumt wurde.
07-24-2022 20:22:36 --- RBK0043I: Unvollständiges Backup in /backup/HeizPi/HeizPi-rsync-backup-20220724-194502 wird gelöscht. Das kann etwas dauern. Bitte Geduld.
07-24-2022 20:39:16 ??? RBK0005E: Backup fehlerhaft beendet. Siehe vorhergehende Fehlermeldungen.
07-24-2022 20:39:17 --- RBK0010I: HeizPi: raspiBackup.sh V0.6.7 (7e37673) So 24. Jul 20:39:16 CEST 2022 beendet mit Returncode 109.
```
• der HeizPi wurde **nicht** neu gestartet.
• in /root sind folgende Dateien:
```
root@HeizPi:~# ls -l
-rw-r--r-- 1 root root      0 Jul 23 22:55 raspiBackup_bracew.log
-rw-r--r-- 1 root root 872876 Jul 24 20:40 raspiBackup.log
-rw-r--r-- 1 root root   1399 Jul 24 20:40 raspiBackup.msg
-rw-r--r-- 1 root root      0 Jul 24 19:45 rpi.log
-rw-r--r-- 1 root root 987607 Jul 24 20:40 rp.log
```
Welche Datei möchtest Du haben? An welche Adresse kann ich diese Dir per E-Mail senden?
• im Backup-Zielordner sind folgende Dateien:
```
b@pi:/home/public/daten/Backup/HeizPi$ ls -l
insgesamt 68
drwxr-xr-x+ 22 root root 4096 Jul 18 15:46 HeizPi-rsync-backup-20220720-094003
drwxr-xr-x+ 22 root root 4096 Jul 20 12:06 HeizPi-rsync-backup-20220720-112828
drwxr-xr-x+ 22 root root 4096 Jul 20 12:59 HeizPi-rsync-backup-20220720-122150
drwxr-xr-x+ 22 root root 4096 Jul 18 15:46 HeizPi-rsync-backup-20220720-130502
drwxr-xr-x+ 22 root root 4096 Jul 20 14:22 HeizPi-rsync-backup-20220720-134550
drwxr-xr-x+ 22 root root 4096 Jul 20 15:31 HeizPi-rsync-backup-20220720-145400
drwxr-xr-x+ 22 root root 4096 Jul 18 15:46 HeizPi-rsync-backup-20220720-154102
drwxr-xr-x+ 22 root root 4096 Jul 18 15:46 HeizPi-rsync-backup-20220720-170803
drwxr-xr-x+ 22 root root 4096 Jul 18 15:46 HeizPi-rsync-backup-20220720-182003
drwxr-xr-x+ 22 root root 4096 Jul 18 15:46 HeizPi-rsync-backup-20220720-203503
drwxr-xr-x+ 22 root root 4096 Jul 18 15:46 HeizPi-rsync-backup-20220721-202803
drwxr-xr-x+ 22 root root 4096 Jul 21 21:54 HeizPi-rsync-backup-20220721-211814
drwxr-xr-x+ 22 root root 4096 Jul 18 15:46 HeizPi-rsync-backup-20220722-205002
drwxr-xr-x+ 22 root root 4096 Jul 18 15:46 HeizPi-rsync-backup-20220722-232002
drwxr-xr-x+ 22 root root 4096 Jul 18 15:46 HeizPi-rsync-backup-20220723-002802
drwxr-xr-x+ 22 root root 4096 Jul 23 09:30 HeizPi-rsync-backup-20220723-093002
```

P.S.: Ab Morgen Nachmittag fahre ich mit meiner Familie ca. 2 Wochen in Sommerurlaub. Ich melde mich danach wieder.

framps, 2022-07-24:
> Welche Datei möchtest Du haben? An welche Adresse kann ich diese Dir per E-Mail senden?

rp.log

Die eMail findest Du auf meiner Kontakseite :wink:

framps, 2022-07-24:
Ok. Das Log ist angekommen.

Geniesse Deinen Urlaub und ich habe genuegend Zeit Luft zu holen und dann das Log zu analysieren :laughing:

bracewGitHub, 2022-07-24:
Vielen Dank, so machen wir das. Schöne Tage auch Dir!

framps, 2022-07-24:
18000 Zeilen ... da habe ich was zu tun ...

bracewGitHub, 2022-07-24:
Kann es vielleicht auch an dem Update auf 0.6.7 liegen?
Würde ein Reinstall von 0.6.6.1 vielleicht helfen?

Ich möchte die beiden Systeme HeizPi und HuhnPi eh nicht mehr updaten/upgraden, weder raspbian noch FHEM noch sonst. Wenn beide laufen, möchte ich im großen und ganzen ein freeze machen und so laufen lassen.

framps, 2022-07-24:
Ist unwahrscheinlich. Details wirst Du in 2 Wochen wissen sofern ich mich in den Logzeilen nicht verloren habe :laughing:

framps, 2022-08-02:
Anbei was ich so im Log gefunden habe:

```
|2447> 	executeCmd(): rsync --exclude=/backup --exclude=/tmp/raspiBackup.log --exclude=/tmp/raspiBackup.msg --exclude=.gvfs '--exclude=/proc/*' '--exclude=/lost+found/*' '--exclude=/sys/*' '--exclude=/dev/*' '--exclude=/boot/*' '--exclude=/tmp/*' '--exclude=/run/*' --exclude=/HeizPi-backup.img --exclude=/HeizPi-backup.tmg --exclude=/HeizPi-backup.sfdisk --exclude=/HeizPi-backup.mbr --exclude=/raspiBackup.log --exclude=/raspiBackup.msg --link-dest=/backup/HeizPi/HeizPi-rsync-backup-20220723-225503 --numeric-ids -aHAx --delete --info=NAME0 / /backup/HeizPi/HeizPi-rsync-backup-20220724-194502
file has vanished: "/etc/webmin/system-status/info.lock"
rsync warning: some files vanished before they could be transferred (code 24) at main.c(1196) [sender=3.1.2]
|2449> 	executeCmd(): rc=24
```
Es scheint webmin auf dem System zu laufen und es loescht waehrend des Backups die Datei `/etc/webmin/system-status/info.lock`. Das fuehrt zum Backupabbruch. Webmin muss noch gestoppt werden vor dem Backup. Oder `/etc/webmin/system-status/*` muss aus dem Backup escluded werden.

```
||1891> 	getMessageText(): echo '--- RBK0026I: Debug Logdatei wurde in /root/raspiBackup.log gesichert.'
```
Daraus sehe ich dass ein Debuglog geschrieben wurde.

bracewGitHub, 2022-08-02:
Hallo, 

ich habe von unterwegs nicht die Möglichkeit mir die alten Dateien anzusehen, bin mir aber sicher, dass am Anfang unserer Diskussion webmin mit bei den Start/Stopp Jobs war. Wenn ich wieder Zuhause bin sehe ich nach.

Aber, wie ich vor 10 Tagen geschrieben habe: "Alle Start/Stopp-Service habe ich entfernt, ausser für cron. ". Deshalb ist webmin zuletzt nicht bei den Start/Stopp Jobs dabei gewesen.

Könnte webmin wieder rein nehmen, wenn ich Zuhause bin.

P.S. Liebe Grüße aus dem Sommerurlaub

framps, 2022-08-02:
Kein Stress. Ich habe eben mal den Mut gefunden mich durch die knapp 20000 Zeilen zu wuehlen und festgehalten was ich gefunden habe. Dass Du nun im Urlaub Deine eMails checks hatte ich nicht erwartet. Geniesse weiterhin Deinen Urlaub und melde Dich wenn Du wieder @home bist :-)

bracewGitHub, 2022-08-10:
Bin wieder da!

bracewGitHub, 2022-08-14:
Hallo Framp,

ich habe ein neues cron gesteuerte raspiBackup erstellt. 
Diesmal mit Stopp von webmin.

Die Dateien:
/root/rp.log
/usr/local/etc/raspiBackup.conf
/usr/local/bin/raspiBackupNfsWrapper.sh
sende ich Dir in einer ZIP-Datei per E-Mail.

• die E-Mail „HeizPi: Backup gestarted“ ist angekommen
`08-14-2022 12:35:12 --- RBK0009I: HeizPi: raspiBackup.sh V0.6.7 (2022-06-19/7e37673) So 14. Aug 12:35:11 CEST 2022 gestartet.`

• es ist keine Ende des Backup E-Mail angekommen,

• der HeizPi wurde neu gestartet.

• im Zielordner wurde nicht aufgeräumt.

framps, 2022-08-15:
Danke fuer das Debuglog. EM Ende steht

```
||2198> 	writeToConsole(): getMessageText 33
||1850> 	getMessageText(): local msg p i s
||1852> 	getMessageText(): msgVar=MSG_DE
||1854> 	getMessageText(): [[ -n EN ]]
||1855> 	getMessageText(): msgVar='MSG_DE[33]'
||1856> 	getMessageText(): msg='RBK0033I: Bitte warten bis aufgeräumt wurde.'
||1857> 	getMessageText(): [[ -z RBK0033I: Bitte warten bis aufgeräumt wurde. ]]
||1864> 	getMessageText(): [[ -z RBK0033I: Bitte warten bis aufgeräumt wurde. ]]
||1868> 	getMessageText(): shift
||1871> 	getMessageText(): i=1
||1872> 	getMessageText(): [[ RBK0033I: Bitte warten bis aufgeräumt wurde. =~ %s ]]
||1877> 	getMessageText(): (( i = 1 ))
||1877> 	getMessageText(): (( 1 <= 0 ))
|||1884> 	getMessageText(): sed 's/%[0-9]+//g'
||1884> 	getMessageText(): msg='RBK0033I: Bitte warten bis aufgeräumt wurde.'
||1886> 	getMessageText(): local msgPref=RBK
||1887> 	getMessageText(): [[ RBK == \R\B\K ]]
||1888> 	getMessageText(): local severity=I
||1889> 	getMessageText(): [[ I =~ [EWI] ]]
||1890> 	getMessageText(): local msgHeader=---
||1891> 	getMessageText(): echo '--- RBK0033I: Bitte warten bis aufgeräumt wurde.'
|2198> 	writeToConsole(): msg='--- RBK0033I: Bitte warten bis aufgeräumt wurde.'
|2200> 	writeToConsole(): ((  0 <= 0  ))
||2205> 	writeToConsole(): cut -f 2 -d ' '
|2205> 	writeToConsole(): local msgNumber=RBK0033I:
|2206> 	writeToConsole(): local msgSev=I
|2208> 	writeToConsole(): ((  1  ))
```
und dann ist ploetzlich Schluss.  D.h. es sieht fuer mich so aus als haettest Du mir nicht das gesamte Debuglog zugeschickt. Genau danach kommen die interessanten Codepathes die ich sehen muss. Kannst Du das noch mal prufen?

PS: Die config brauchst Du mir nicht zu schicken. Diese werden alle im Debuglog ausgegeben :wink:

bracewGitHub, 2022-08-15:
Ich habe Dir eine E-Mail mit dem ungezippten /root/rp.log geschickt.

bracewGitHub, 2022-08-17:
Hallo Framp,

in aller Kürze: 
**Es hat funktioniert!** mit raspiBackup.sh 0.6.6.1

Lange Version:

Ich habe nun doch auf die "alte" Version raspiBackup.sh 0.6.6.1 mit: `sudo raspiBackup.sh -V ` zurückgesetzt.
und in der Original /usr/local/bin/raspiBackupNfsWrapper.sh habe ich von:
`  raspiBackup.sh -g --timestamps --notifyStart --rebootSystem`
zu
`  raspiBackup.sh --timestamps --notifyStart`
geändert, da die Option "--rebootSystem" nicht in 0.6.6.1 funktioniert.

In der /usr/local/etc/raspiBackup.conf steht:
`DEFAULT_STOPSERVICES="systemctl stop webmin && systemctl stop cron"`
`DEFAULT_STARTSERVICES="systemctl start cron && systemctl start webmin"`

Die E-Mail „HeizPi: Backup gestarted“ ist angekommen:
08-17-2022 09:35:09 --- RBK0009I: HeizPi: raspiBackup.sh V0.6.6.1 (c674d9f) Mi 17. Aug 09:35:08 CEST 2022 gestartet.

Die E-Mail „;-) HeizPi: Backup erfolgreich beendet.“ ist angekommen:
08-17-2022 09:35:43 --- RBK0085I: Backuperstellung vom Typ rsync gestartet. Bitte Geduld.
08-17-2022 10:12:59 --- RBK0078I: Backupzeit: 00:37:32.
08-17-2022 10:13:02 --- RBK0033I: Bitte warten bis aufgeräumt wurde.
08-17-2022 10:13:19 --- RBK0159I: 3 Backups werden für den Backuptyp rsync aufbewahrt. Bitte Geduld.
08-17-2022 15:20:03 --- RBK0017I: Backup erfolgreich beendet.
08-17-2022 15:20:04 --- RBK0010I: HeizPi: raspiBackup.sh V0.6.6.1 (c674d9f) Mi 17. Aug 15:20:03 CEST 2022 beendet mit Returncode 0.

Das Backup hat also etwas mehr als eine halbe Stunde gedauert, das aufräumen jedoch über 5 Stunden.
Im Zielverzeichnis sind jetzt die gewünschten 3 Backups.

**Vielen Dank für Deine umfangreichen Hilfen!!!**

Letztendlich weiß ich nicht an was es liegt. Falls Du weitere "Ursachenforschung" betreiben möchtest, bin ich gerne behilflich.

Für mich ist das die Lösung des Problems :-) Ich hoffe aber, dass ich bei der Version raspiBackup.sh 0.6.6.1 langfristig bleiben kann.

bracewGitHub, 2022-08-17:
P.S.: Wie kann ich nun den RasPi nach dem erfolgreichen Backup ein reboot durchführen lassen?
Wo in /usr/local/bin/raspiBackupNfsWrapper.sh (vermute ich) müsste ich welchen Befehl ergänzen?

framps, 2022-08-17:
> **Es hat funktioniert!** mit raspiBackup.sh 0.6.6.1

:+1: Sehr gut dass es wieder funktioniert :+1: 

> das aufräumen jedoch über 5 Stunden.

Du hattest wohl eine Menge rsync Backups. Das Loeschen dauert bei einer groesseren Menge wie man sieht schon lange. Zukuenftig wird es wieder kuerzer sein da nur maximal ein Backup zu loeschen ist.

> Letztendlich weiß ich nicht an was es liegt. Falls Du weitere "Ursachenforschung" betreiben möchtest, bin ich gerne behilflich.

Vielen Dank fuer das Angebot. Ich auch nicht und es ist fuer mich unbefriedigend den Grund nicht zu kennen. Aber den bekomme ich so wie ich es sehe nur raus wenn ich mich bei Dir vor Ort mit dem System beschaeftigen wuerde. Ich werde noch mal den Code von 0.6.6.1 mit dem 0.6.7 Code vergleichen. Vielleicht bekomme ich noch eine Idee. Ich wuerde den Issue closen wenn ich nichts finde.

> Wo in /usr/local/bin/raspiBackupNfsWrapper.sh (vermute ich) müsste ich welchen Befehl ergänzen?

```
		raspiBackup.sh
		rc=$?
		if (( $rc > 0 )); then
			echo "raspiBackup failed with rc $rc"
			exit $rc
		fi
```

aendern in

```
		raspiBackup.sh
		rc=$?
		if (( $rc > 0 )); then
			echo "raspiBackup failed with rc $rc"
			exit $rc
		fi
                shutdown -r now
```

> DEFAULT_STARTSERVICES="systemctl start cron && systemctl start webmin"

solltest Du auf `""` setzen denn Du rebootest ja und es werden alle Services wieder gestartet.

> Ich hoffe aber, dass ich bei der Version raspiBackup.sh 0.6.6.1 langfristig bleiben kann.

Setze `DEFAULT_NOTIFY_UPDATE=0` damit Du keine Benachrichtigung bekommst dass es neue Releases gibt.

bracewGitHub, 2022-08-18:
So, ich habe nun die gleiche Einrichtung für den HuhnPi übernommen. 
Auch bei diesem hat es heute funktioniert mit raspiBackup.sh 0.6.6.1.

Seltsam ist, dass die Mail:
`  HuhnPi: Backup gestarted. MIME-Version: 1.0 Content-Type: text/html; charset=utf-8`
im Gegensatz zum HeizPi:
`  HeizPi: Backup gestarted`
mit dem "_MIME-Version: 1.0 Content-Type: text/html; charset=utf-8_" Zusatz kommen?

Im Crontab habe ich beide wieder wie vor der Misere für die Nächte zwischen 0 und 1 Uhr aufgenommen. Werde ich mal beobachten.

Kann es sein, dass es daran gelegen hat, dass beide mit Raspbian 9.13 stretch laufen?

framps, 2022-08-20:
> Kann es sein, dass es daran gelegen hat, dass beide mit Raspbian 9.13 stretch laufen?

Ist eher unwahrscheinlich. Es hat eine Menge Aenderungen von 0.6.6.1 auf 0.6.7 gegeben. Warum das Problem nur bei Dir auftritt - 0.6.7 wird mittlerweile von fast allen raspiBackup Nutzer eingesetzt - verstehe ich nicht.

> Seltsam ist, dass die Mail:

Entweder nutzt Du unterschiedliche MUAs oder die eMailConfig ist unterschiedlich (`DEFAULT_EMAIL_COLORING`)

closed 2022-09-04 by github-actions[bot]

bracewGitHub, 2022-10-11:
> Danke fuer das Log. Es ist aber auch nicht laenger 😒 .
>
> Erst danach kommt der Code der aufraeumt und dessen Verlauf ich sehen
> muss um die Ursache zu finden.
>
> Solange Du es nicht schaffst ein vollstaendiges Debuglog zu erstellen
> kann ich Dir nicht helfen.
>
> Was tun?
>
> Du solltest ein normales Debuglog - kein bash debug log erstellen
> lassen. Dort findest Du Timestamps. Ich wuerde mal im Syslog nachsehen
> was in der Zeit des Backuplaufs so alles passiert. Vermutlich
> graetscht irgendwann ein Service rein oder irgendein Fehler tritt auf.
>
> Ich habe schon viele Debuglogs erhalten und damit immer die Ursache
> eines Problems gefunden. Manchmal fehlte an einer wichtigen Stelle
> noch ein Debugstatement was ich dann aber eingefuegt habe um die
> Ursache zu finden. Aber sowas wie bei Dir ist mir noch nie vorgekommen.
>
> Vielleicht bringt es auch den Setup von Deinem HeizPi mit den anderen
> Pis zu vergleichen wo alles funktioniert und Du bekommst eine Idee was
> die ursache sein kann.
>
> Ich bin jedenfalls leider mit meinem Latein am Ende 😫

Ich habe kein anderes Log.
Was tun?

Wie kann ich auf Vers. 0.6.6.1 zurück? Damit lief es zuletzt.
Alles neu installieren?

Der einziges RasPi, welcher mit WrapperScript arbeitet und funktioniert
ist der ePi, welcher aber bereits auf bullseye läuft.
HuhnPi und HeizPi laufen mit stretch und dies soll auch (möglichst) so
bleiben für alle Zukunft. Beide haben die gleichen raspiBackup Probleme.
Pi (der Zentrale RasPi) läuft mit buster und ohne WrapperScript, aber
funktioniert auch.

------------------------------------------------------------------------