framps/raspiBackup #526: Cant figure out why backup is failing

p1r473, 2022-07-24:
Hi,
I just checked and saw my backups have been failing for a long time
I have viewed the log and can't seem to figure out why its failing.
```
20220723-200235 DBG 1993:  --- Logfiles used: /tmp/raspiBackup.log and /tmp/raspiBackup.msg
20220723-200235 DBG 2004:  --- raspiBackup 0.6.7, 2022-06-19/19:37:48 - 7e37673
20220723-200235 DBG 2007:  --- +================================================================================================================================================+
20220723-200235 DBG 2008:  --- | ===> A lot of sensitive information is masqueraded in this log file. Nevertheless please check the log carefully before you distribute it <=== |
20220723-200235 DBG 2009:  --- +================================================================================================================================================+
20220723-200235 DBG 1836:  --> trapWithArg cleanupStartup SIGINT SIGTERM EXIT
20220723-200235 DBG 1841:  <-- trapWithArg 
20220723-200235 DBG 2608:  --> initializeDefaultConfigVariables 
20220723-200235 DBG 2746:  <-- initializeDefaultConfigVariables 
20220723-200235 DBG 2751:  --> copyDefaultConfigVariables 
20220723-200235 DBG 2823:  <-- copyDefaultConfigVariables 
20220723-200235 DBG 3573:  --> readConfigParameters 
20220723-200235 DBG 3588:      --- Read config /usr/local/etc/raspiBackup.conf : 0.1.5
20220723-200235 DBG 3588:      --- VERSION_CONFIG="0.1.5"
20220723-200235 DBG 3588:      --- DEFAULT_BACKUPPATH="/mnt/BlackPearl"
20220723-200235 DBG 3588:      --- DEFAULT_BACKUPTYPE="tar"
20220723-200235 DBG 3588:      --- DEFAULT_ZIP_BACKUP="1"
20220723-200235 DBG 3588:      --- DEFAULT_DD_BACKUP_SAVE_USED_PARTITIONS_ONLY=0
20220723-200235 DBG 3588:      --- DEFAULT_KEEPBACKUPS="3"
20220723-200235 DBG 3588:      --- DEFAULT_KEEPBACKUPS_DD=0
20220723-200235 DBG 3588:      --- DEFAULT_KEEPBACKUPS_DDZ=0
20220723-200235 DBG 3588:      --- DEFAULT_KEEPBACKUPS_TAR=0
20220723-200235 DBG 3588:      --- DEFAULT_KEEPBACKUPS_TGZ=0
20220723-200235 DBG 3588:      --- DEFAULT_KEEPBACKUPS_RSYNC=0
20220723-200235 DBG 3588:      --- DEFAULT_SMART_RECYCLE="1"
20220723-200235 DBG 3588:      --- DEFAULT_SMART_RECYCLE_DRYRUN="0"
20220723-200235 DBG 3588:      --- DEFAULT_SMART_RECYCLE_OPTIONS="7 4 12 3"
20220723-200235 DBG 3588:      --- DEFAULT_STOPSERVICES="systemctl stop log2ram"
20220723-200235 DBG 3588:      --- DEFAULT_STARTSERVICES="systemctl start log2ram"
20220723-200235 DBG 3588:      --- DEFAULT_BEFORE_STOPSERVICES="systemctl stop rsyslog.service && systemctl stop syslog.socket && systemctl stop log2ram && systemctl disable log2ram"
20220723-200235 DBG 3588:      --- DEFAULT_AFTER_STARTSERVICES="systemctl enable log2ram && systemctl start log2ram && systemctl start syslog.socket && systemctl start rsyslog.service"
20220723-200235 DBG 3588:      --- DEFAULT_EMAIL="s@@@@m(21)"
20220723-200235 DBG 3588:      --- DEFAULT_SENDER_EMAIL=""
20220723-200235 DBG 3588:      --- DEFAULT_EMAIL_PARMS=""
20220723-200235 DBG 3588:      --- DEFAULT_MAIL_PROGRAM="mail"
20220723-200235 DBG 3588:      --- DEFAULT_TELEGRAM_TOKEN=""
20220723-200235 DBG 3588:      --- DEFAULT_TELEGRAM_CHATID=""
20220723-200235 DBG 3588:      --- DEFAULT_TELEGRAM_NOTIFICATIONS="F"
20220723-200235 DBG 3588:      --- DEFAULT_NOTIFY_START=0
20220723-200235 DBG 3588:      --- DEFAULT_MAIL_ON_ERROR_ONLY=0
20220723-200235 DBG 3588:      --- DEFAULT_MSG_LEVEL="0"
20220723-200235 DBG 3588:      --- DEFAULT_COLORING="CM"
20220723-200235 DBG 3588:      --- DEFAULT_EMAIL_COLORING="SUBJECT"
20220723-200235 DBG 3588:      --- DEFAULT_RESTORE_DEVICE=""
20220723-200235 DBG 3588:      --- DEFAULT_VERBOSE=0
20220723-200235 DBG 3588:      --- DEFAULT_SKIPLOCALCHECK=0
20220723-200235 DBG 3588:      --- DEFAULT_DD_BLOCKSIZE=1M
20220723-200235 DBG 3588:      --- DEFAULT_DD_PARMS=""
20220723-200235 DBG 3588:      --- DEFAULT_EXCLUDE_LIST="--exclude=/mnt/* --exclude=/var/lib/samba/private/msg.sock/*"
20220723-200235 DBG 3588:      --- DEFAULT_NOTIFY_UPDATE=1
20220723-200235 DBG 3588:      --- DEFAULT_EXTENSIONS=""
20220723-200235 DBG 3588:      --- DEFAULT_LANGUAGE="EN"
20220723-200235 DBG 3588:      --- DEFAULT_DEPLOYMENT_HOSTS=""
20220723-200235 DBG 3588:      --- DEFAULT_YES_NO_RESTORE_DEVICE=""
20220723-200235 DBG 3588:      --- DEFAULT_RESIZE_ROOTFS=1
20220723-200235 DBG 3588:      --- DEFAULT_LINK_BOOTPARTITIONFILES=0
20220723-200235 DBG 3588:      --- DEFAULT_USE_HARDLINKS=1
20220723-200235 DBG 3588:      --- DEFAULT_TAR_BOOT_PARTITION_ENABLED=0
20220723-200235 DBG 3588:      --- DEFAULT_SKIP_DEPRECATED=""
20220723-200235 DBG 3588:      --- DEFAULT_USE_UUID=1
20220723-200235 DBG 3588:      --- DEFAULT_CHECK_FOR_BAD_BLOCKS=0
20220723-200235 DBG 3588:      --- DEFAULT_TIMESTAMPS=0
20220723-200235 DBG 3588:      --- DEFAULT_SYSTEMSTATUS=0
20220723-200235 DBG 3588:      --- DEFAULT_RESTORE_REMINDER_INTERVAL=6
20220723-200235 DBG 3588:      --- DEFAULT_RESTORE_REMINDER_REPEAT=3
20220723-200235 DBG 3588:      --- DEFAULT_PARTITIONBASED_BACKUP="0"
[... 670 lines cut ...]
20220723-202522 DBG 3841:                      <-- colorAnnotation 
!!! RBK0197W: eMail send command mail failed with RC 36.
20220723-202522 DBG 4084:                  <-- sendEMail 
20220723-202522 DBG 4160:                  --- Masquerading eMail
20220723-202522 DBG 4183:                  --- Masquerading some mount options
20220723-202522 DBG 4201:                  --- Masquerading home directory name
20220723-202522 DBG 4206:                  --- Masquerading hostname
20220723-202522 DBG 4211:                  --- Masquerading sensitive non local IPs
20220723-202523 DBG 2015:                  --> logFinish 
20220723-202523 DBG 2026:                      --- 109 2 0
20220723-202523 DBG 2031:                      --- LOG_OUTPUT=3
20220723-202523 DBG 2035:                      --- LOG_OUTPUT: 3
20220723-202523 DBG 2062:                      --- DEST_LOGFILE: /root/raspiBackup.log
20220723-202523 DBG 2063:                      --- DEST_MSGFILE: /root/raspiBackup.msg
20220723-202523 DBG 2068:                      --- Logfiles used: /root/raspiBackup.log and /tmp/raspiBackup.msg
20220723-202523 DBG 2073:                      --- Logfiles used: /root/raspiBackup.log and /root/raspiBackup.msg
20220723-202523 DBG 2080:                          *** cat /tmp/raspiBackup.logf
20220723-202523 DBG 2080:                              
20220723-202523 DBG 3817:                      --> colorAnnotation 1
20220723-202523 DBG 3841:                      <-- colorAnnotation 
20220723-202523 DBG 2095:                  <-- logFinish 
20220723-202523 DBG 4308:                  --> unLockMe 
20220723-202523 DBG 4310:                  <-- unLockMe 
20220723-202523 DBG 4418:              <-- cleanup 
```

framps, 2022-07-24:
```
tar: Removing leading `/' from member names
tar: Removing leading `/' from hard link targets
tar: /etc/pihole/pihole-FTL.db: file changed as we read it
tar: /etc/pihole: file changed as we read it
```

PiHole modifies some files when the backup runs. Either stop PiHole first or exclude `/etc/pihole` directory from backup.

I strongly suggest to fix your mail issue or to use telgram. That's the only way to get notifications when the backup fails.

framps, 2022-07-24:
See also [this issue](https://github.com/framps/raspiBackup/issues/460)

closed 2022-07-24 by p1r473