framps/raspiBackup #530: @eaDir issues - ignore that

shakisha, 2022-08-03:
When making a backup into a Synology nfs mounted partition I have got this error: 
[image]

So, when I switch into that directory I can see that an @eaDir is created by the Synology NAS and *there is no easy way to stop that*.

[image]

so, when I remove that directory, the backup will start to work again:

[image]

So, is it there any way to instruct raspibackup to "ignore" @eaDir directories?

framps, 2022-08-03:
I also use my Synology as backupspace and I don't have the `@eadir` directories. Looks like they appear when you have some special config on your Syno.

I added this check recently because additional files or directories may screw up the backup strategy algorithms.

As of now there is no way to turn this error message off other than to delete the directory or to modify Synology settings and this directory is not created any more.

I see your point. Looks like the check is to restrictive. I will check the code and I'm confident I find a solution for you :wink:

shakisha, 2022-08-03:
hello @framps . 

i have found in the raspiBackup.sh code at line 6563 this: 

`	logCommand "ls -1 ${BACKUPPATH}"
	local nonRaspiGeneratedDirs=$(ls -1 ${BACKUPPATH} | egrep -Ev "$HOSTNAME\-($POSSIBLE_BACKUP_TYPES_REGEX)\-backup\-([0-9]){8}.([0-9]){6}" |wc -l)
	logItem "nonRaspiGeneratedDirs: $nonRaspiGeneratedDirs" `

	if (( $nonRaspiGeneratedDirs > 0 )); then
		writeToConsole $MSG_LEVEL_MINIMAL $MSG_INVALID_BACKUPNAMES_DETECTED $nonRaspiGeneratedDirs $BACKUPPATH
		exitError $RC_BACKUP_DIRNAME_ERROR
	fi
`

I can try to put @eaDirs inside the regex to avoid that and send you a merge request, tell me if you like the idea of if you desire another logic. 
I'll try the code to my machine first :-)

framps, 2022-08-03:
> I can try to put @eaDirs inside the regex to avoid that

Yes. But this is just a workaround for you with `@eadir`. I want to use a more general approach so other files and directories can also  exist in the backup directory. I only have to check whether the directoryname or filename clashes with the naming convention or raspiBackup.  Otherwise they will not screw up the backup strategy code.

framps, 2022-08-03:
I just created a general fix for your issue :wink: . 

Note: I didn't execute any tests. Please grab the fixed version and verify the fix. Please let let me know whether you use the intelligent backup strategy or the normal keep strategy.

shakisha, 2022-08-03:
@framps thank you so much, I'll test with both strategies and I'll let you know :-)

shakisha, 2022-08-03:
I have tried, the fix works PERFECTLY!

framps, 2022-08-03:
Thank you very much you verified the fix :+1: . 

I will create now an official fix for you and attach the file in this issue. Please use this version in your environment. Your current version from git misses some important build stuff and should be used only for fix verification. 

I will publish the fix in next release. When this release is available you will be notified and can upgrade.

framps, 2022-08-03:
[raspiBackup.sh.txt](https://github.com/framps/raspiBackup/files/9253919/raspiBackup.sh.txt)

That's the official build for the fix. You have to delete `.txt` I had to add in order to attach the file :wink:

framps, 2022-08-03:
If you're find now please close the issue.

shakisha, 2022-08-03:
ok, i’ll update soon, meanwhile i have temporally applied the patch to the local code :)

closed 2022-08-03 by shakisha